espressif/esp-idf #15177: lwip recv 出错，errno 值为113

luacerbb, 2025-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF_V5.1.4

### Espressif SoC revision.

ESP8684-WROOM-02C

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom board

### Power Supply used.

USB

### What is the expected behavior?

recv 接口可以正常使用，可用ram 不会变少

### What is the actual behavior?

出现这个错误后，可用的ram 会变少2k附近

### Steps to reproduce.

设备正常使用的mqtt 连接云平台的情况下，偶发会出现该错误，ram 可用资源会变少。mqtt 重连后可以恢复连接，但资源不会恢复。MQTT 使用的不是IDF 的库，但从日志看是在lwip recv 报错后出现的。

luacerbb, 2025-01-10:
其中 lwip 库用的是5.1.2的库

euripedesrocha, 2025-01-14:
Hi @luacerbb could you add more information? It would be useful to have some logs to understand the issue.

luacerbb, 2025-01-16:
> Hi @luacerbb could you add more information? It would be useful to have some logs to understand the issue.
hi,The problem of memory leak, I have solved it, and there is one problem left, which is the error code of 113? What causes it.

if (FD_ISSET(network_handle->fd, &recv_sets)) {
                recv_res = recv(network_handle->fd, buffer + recv_bytes, len - recv_bytes, 0);
                if (recv_res == 0) {
                    printf("_core_sysdep_network_recv, nwk connection closed\n");
                    return STATE_PORT_NETWORK_RECV_CONNECTION_CLOSED;
                } else if (recv_res < 0) {
                    printf("_core_sysdep_network_recv, errno: %d\n", errno);
                    perror("_core_sysdep_network_recv, nwk recv error: ");
                    if (errno == EINTR) {
                        continue;
                    }
                    return STATE_PORT_NETWORK_RECV_FAILED;
                } else {
                    recv_bytes += recv_res;
                    /* printf("recv_bytes: %d, len: %d\n",recv_bytes,len); */
                    if (network_handle->socket_type == CORE_SYSDEP_SOCKET_UDP_CLIENT || recv_bytes == len) {
                        break;
                    }
                }
            }