espressif/esp-idf #15178: [WiFi] Support setting / enforcing 11AX explicitly

h-milz, 2025-01-10:
### Is your feature request related to a problem?

I need to enforce an 11ax connection because the project I'm working on is supposed to work in a congested environment, which is what 11ax was made for. As far as I can see, there is no way to enforce 11ax. I checked with the iperf example that was suggested: 

```
iperf> wifi_mode ap
I (1488536) WIFI: mode: ap
W (1488537) wifi:(itwt)itwt_stop_process!
I (1488537) wifi:mode : softAP (60:55:f9:fb:cf:41)
I (1488539) wifi:Total power save buffer number: 24
I (1488549) wifi:Init max length of beacon: 752/752
I (1488550) wifi:Init max length of beacon: 752/752
I (1488550) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1488550) WIFI: DONE.SET_WIFI_MODE,OK.
iperf> ap_set BLA ************
I (1495539) wifi:Total power save buffer number: 24
I (1495540) esp_netif_lwip: DHCP server started on interface WIFI_AP_DEF with IP: 192.168.4.1
I (1495541) WIFI: DONE.SET_AP_CONFIG,OK.
iperf> wifi_band 5g
I (1503504) WIFI: DONE.SET_WIFI_BAND_MODE,OK.
iperf> wifi_protocol ax
I (1515651) WIFI: DONE.SET_WIFI_PROTOCOL,FAIL.258,ESP_ERR_INVALID_ARG
iperf> wifi_protocol --5g=ax
I (1520442) WIFI: DONE.SET_WIFI_PROTOCOL,FAIL.258,ESP_ERR_INVALID_ARG
iperf> wifi_protocol
E (1525456) WIFI: Can not convert bitmap 0x0 to protocol string.
I (1525456) WIFI: (sta)GET_WIFI_PROTO:unknown, GET_WIFI_PROTO_5G:a/n/ac/ax

```

As soon as you try to

`esp_wifi_set_protocol(WIFI_IF_AP, WIFI_PROTOCOL_11AX);`

you get the error `ESP_ERR_INVALID_ARG`. 

And `esp_wifi_set_protocols()` appears to also set the whole protocol family for 5G, and there seems no way to explicitly select only one of these. `cmd_do_wifi_protocol()` from the iperf example also invokes only one of these two library functions. 

The documentation says "2. When WiFi band mode is 5G only, support 802.11a or 802.11an or 802.11anac or 802.11anacax". 

Apparently, you need to set the whole family A, N, AC and AX, which is the default for 5G. And in this case the Wifi stack seems to scan the lower bits first, and choose the lowest one that succeeds, in this case N. It won't even try to select AX because N succeeds. (I'd really like to try and look this up in the source code of `libnet80211.a` but sadly, there's only the binary library.) This is different with my Wifi router which negotiates AX at least every other connection attempt. 

### Describe the solution you'd like.

It should be possible to enforce 11ax because that is what you want in congested IoT environments. 

### Describe alternatives you've considered.

None. As I mentioned, my Wifi Router negotiates 11ax at least every now and then, but the ESP32 AP does not, and will always negotiate 11an.

QingzhaoYin, 2025-01-10:
@h-milz Hi~,  I think you should use "wifi_protocol --5g=ax -i ap" to set ap's protocol and use "wifi_protocol -i ap" to get ap's protocol

[image]

The wifi_protocol command is used to set or get the protocol. Note the following:
1. If the interface (-i) is not specified, the protocol for the STA interface is set by default.
[image]

3. In the 5G band, only the protocol for the 5G band can be set or get(2G protocol getted is unknown). In the 2G band, only the protocol for the 2G band can be set or get(5G protocol getted is unknown). In the 2G+5G mode (AUTO), both 2G and 5G protocols can be set or get
[image]

h-milz, 2025-01-10:
Oh wow that actually works, at least with the iperf example. I'm getting about 63 MBit/s between my two C5 boards (albeit over just 20 cm ;-) ) . Exactly what I need. Thank you! 

Now extracting what the iperf example actually does under the hood.

closed 2025-01-10 by h-milz