espressif/esp-idf #15179: ESP-NOW authentication and vulnerability

h-milz, 2025-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am concerned about ESP-NOW authentication which by and large seems not to exist, hence opening the door for denial of service. Maybe I misunderstand the concept, but according to my analysis, ESP-NOW supports packet encryption, but the master will always happily accept all packets (also unencrypted ones!) that are sent to him. ESP-NOW does not enforce encryption on the receiving end. One can verify the sender's MAC or IP address but these can easily be spoofed by anyone who knows how to handle a packet sniffer. One can add an HMAC to the payload to perform integrity checksumming and authenticity but the HMAC can only be verified after reception of each full datagram, which means it is fairly easy to saturate the receiver with nonsense data. So - how can this be mitigated? 

Can someone shed some light on this please?

Frostie314159, 2025-01-10:
Without having access to the code, this may be hard to know, but ESP-NOW uses CCMP for Encryption, which generates a message integrity tag during encryption. As long as the key remains secret, both sides can still establish authenticity. The issue with accepting unencrypted frames still remains though.

zhangyanjiaoesp, 2025-01-15:
@h-milz 
The unicast packets are encrypted, while broadcast packets are unencrypted. Can you provide an example or log for the problem you described?

h-milz, 2025-01-15:
@zhangyanjiaoesp the issue is not encryption, but the missing authentication prior to packet processing. It appears that the receiver will accept all packets sent to him, encrypted or not, and he will only be able to determine a legitimate packet after receiving and processing it, which opens the door for denial of service. Even if you add an HMAC to your payload for authentication, all an attacker needs to do is flooding the receiver with packets. The sender's and the receiver's MAC addresses can easily be determined and spoofed. 

So the question is, how is ESP-NOW supposed to prevent this kind of DoS, or is that a known issue that should be documented?

zhangyanjiaoesp, 2025-01-20:
OK I got it, for connectionless ESPNOW, it is difficult to completely defend against DoS attacks. We will add documentation to clarify this situation.