espressif/esp-idf #15180: Performance optimization for Wi-Fi BSS scan.

Frostie314159, 2025-01-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I opened an on the esp32-wifi-lib repository, which suggests an optimization to BSS scans. I'm unsure, whether that repo is actively monitored, so I reopened the issue here.
The issue can be found here https://github.com/espressif/esp32-wifi-lib/issues/42

Greetings

Simon

hansw123, 2025-01-23:
@Frostie314159  I'm sorry I can't answer more on this question due to company confidentiality requirements, but thank you very much for your support of espressif!

Frostie314159, 2025-01-23:
Hi, thanks for the response. I totally get that. If it gets fixed, I'll see it in the decompilation :)

closed 2025-01-23 by Frostie314159