espressif/esp-idf #15183: esp_wifi_start ignores channel in config

kb1sph, 2025-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Admittedly, I am actually using Arduino, but I have gone through the IDF code and tracked this down by modifying the code to print debug messages.

What I'm trying to do is tell my ESP32-S3 to connect to an AP on a specific channel  I know this may be an unusual thing to do, but I want to make a program in which you can not only restrict the BSSID connected to, but the channel as well.  This seems like it should be possible as the Arduino library has an option in the "begin" function to set a channel.  Looking in the code, that information is then passed to esp_wifi_set_config.  After that, it called the esp_wifi_start, which seems to ignore the channel setting and just connect to the SSID on any channel.

The use case for this would be rare, but useful.  With multiple access points set up in a large environment, in order to make the device only work within it's intended area (for example, a device restricted to one office space that will fail to work outside of that office) I would like for the channel to be checked against the config and not connect if it does not match.

One could set up a different SSID for each office, but then the devices that should be able to roam between them would need every access point set up.

So, I'd like to ask, if this is a broken feature, can it be fixed?  If this is not broken, but then channel setting is there and just ignored (not yet used?), can we add a little bit of code to not connect if the channel does not match?

Thank you!

vik-gokhale, 2025-01-20:
Hi @kb1sph ,
This isn't a broken feature, but an non-existent one.
The channel that is passed to the config, is the starting channel for scanning before a connection attempt. If the ESPdevice is programmed to use 'Fast scan' and it finds the AP with intended WIFI profile in the same channel, it would attempt a connection. 

However, there are cases where upon scanning the intended channel the AP either responds late (after device moves to the next channels for scans) or does not respond. In such cases, the device tries attempting a connecting to the AP in the next scanned channel (unintended one) where it finds the AP.

As you've rightly said, the use case of connecting to a particular channel is rare and we do not plan to implement connecting to a channel passed by the application.

Please revert for any more queries.

Regards,
Vikram

hansw123, 2025-01-23:
@kb1sph   sorry idf dont support Specify the channel connection， and will not add this feature ，thanks for your report