espressif/esp-idf #15186: Failed to connect to wifi under certain situation

mushsoooup, 2025-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4 release

### Espressif SoC revision.

~

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32S3-CORE from LuatOS

### Power Supply used.

USB

### What is the expected behavior?

connect to wifi smoothly.

### What is the actual behavior?

It takes a long time(usually a few minutes) or even impossible to connect to wifi from my router while it can smoothly connect to my hotspot or ap from another esp board. 
I wonder what might caused this.

### Steps to reproduce.

```
#include "sdkconfig.h"
#include "freertos/FreeRTOS.h"
#include "nvs_flash.h"
#include "esp_log.h"
#include "./wifi/wifi.h"
#include "esp_wifi.h"
#include <mutex>
#include <sys/socket.h>
#include "driver/gpio.h"
#include "gpio/gpio.h"

static const char *TAG = "main";

enum Status {
    BAD,
    OK,
};

struct State {
    std::mutex lock;
    Status status;
    sockaddr_in addr;
    int socket;
};

static State state;

void wifi_handler(void *arg, esp_event_base_t event_base, int32_t event_id, void *event_data) {
    if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_START) {
        state.lock.lock();
        state.status = BAD;
        state.socket = socket(AF_INET, SOCK_DGRAM, IPPROTO_UDP);
        state.addr.sin_family = AF_INET;
        state.addr.sin_port = htons(12345);
        state.lock.unlock();
        esp_wifi_connect();
    } else if (event_base == WIFI_EVENT && event_id == WIFI_EVENT_STA_DISCONNECTED) {
        state.lock.lock();
        state.status = BAD;
        state.lock.unlock();
        ESP_LOGI(TAG, "connect to the AP fail, retrying");
        esp_wifi_connect();
    } else if (event_base == IP_EVENT && event_id == IP_EVENT_STA_GOT_IP) {
        ip_event_got_ip_t* event = (ip_event_got_ip_t*) event_data;
        ESP_LOGI(TAG, "connect to the AP succeed");
        state.lock.lock();
        state.status = OK;
        state.addr.sin_addr.s_addr = inet_addr("192.168.1.100");
        connect(state.socket, (sockaddr *)&state.addr, sizeof(state.addr));
        state.lock.unlock();
    }
}

extern "C" void app_main(void)
{
        //Initialize NVS
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES || ret == ESP_ERR_NVS_NEW_VERSION_FOUND) {
      ESP_ERROR_CHECK(nvs_flash_erase());
      ret = nvs_flash_init();
    }
    ESP_ERROR_CHECK(ret);

    esp_wifi_set_ps(WIFI_PS_NONE);

    ESP_LOGI(TAG, "ESP_WIFI_MODE_STA");
    wifi_init_sta("HappyFlower", "ilovediana", wifi_handler);
    setup_output(GPIO_NUM_10);
    int n = 0;
    while(1) {
        n++;
        sleep(1);
        gpio_set_level(GPIO_NUM_10, n % 2);
        state.lock.lock();
        if (state.status == BAD) {
            state.lock.unlock();
            continue;
        }
        send(state.socket, "test", 4, 0);
        state.lock.unlock();
        ESP_LOGI(TAG, "send complete");
    }
}
```
#include "sdkconfig.h"
#include "esp_log.h"
#include "esp_event.h"
#include "esp_wifi.h"
#include <string>
#include "wifi.h"
#include <cstring>

static const char *TAG = "wifi station";

void wifi_init_sta(std::string ssid, std::string pass, esp_event_handler_t handler)
{
    ESP_ERROR_CHECK(esp_netif_init());

    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_sta();

    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));

    esp_event_handler_instance_t instance_any_id;
    esp_event_handler_instance_t instance_got_ip;
    ESP_ERROR_CHECK(esp_event_handler_instance_register(WIFI_EVENT,
                                                        ESP_EVENT_ANY_ID,
                                                        handler,
                                                        NULL,
                                                        &instance_any_id));
    ESP_ERROR_CHECK(esp_event_handler_instance_register(IP_EVENT,
                                                        IP_EVENT_STA_GOT_IP,
                                                        handler,
                                                        NULL,
                                                        &instance_got_ip));

    wifi_config_t wifi_config = {};
    memcpy(wifi_config.sta.ssid, ssid.c_str(), ssid.length() + 1);
    memcpy(wifi_config.sta.password, pass.c_str(), pass.length() + 1);
    wifi_config.sta.threshold.authmode = WIFI_AUTH_WPA2_PSK;
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));
    ESP_ERROR_CHECK(esp_wifi_set_config(WIFI_IF_STA, &wifi_config));
    ESP_ERROR_CHECK(esp_wifi_start());

    ESP_LOGI(TAG, "wifi_init_sta finished.");
}
```

### Debug Logs.

```plain
I (586) wifi:state: init -> auth (0xb0)
I (1586) wifi:state: auth -> init (0x200)
I (1606) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (1606) main: connect to the AP fail, retrying
I (4016) main: connect to the AP fail, retrying
I (4066) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (4076) wifi:state: init -> auth (0xb0)
I (4096) wifi:state: auth -> assoc (0x0)
I (4136) wifi:state: assoc -> run (0x10)
I (14146) wifi:state: run -> init (0xcc00)
I (14156) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (14156) main: connect to the AP fail, retrying
I (16566) main: connect to the AP fail, retrying
I (16666) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (16666) wifi:state: init -> auth (0xb0)
I (17676) wifi:state: auth -> init (0x200)
I (17686) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (17686) main: connect to the AP fail, retrying
```
```

AkshatAgrawl, 2025-01-14:
Hi @mushsoooup ,

Could you please share logs with debug-level enabled, sniffer capture, and AP model/configuration to help debug the issue further?

Thanks in advance!

Alvin1Zhang, 2025-03-06:
@mushsoooup Thanks for reporting, would you please help share more details https://github.com/espressif/esp-idf/issues/15186#issuecomment-2589048009 ? Thanks.

closed 2025-03-27 by Alvin1Zhang