espressif/esp-idf #15187: FatFs' VFS `unlink` implementation does not adhere to POSIX

Aric5301, 2025-01-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi.
I've discovered that VFS' implementation of the FatFs file system uses FatFs' implementation of the unlink function as the unlink function implementation.
The problem is that [FatFs' unlink function does not support removing a currently open file](http://elm-chan.org/fsw/ff/doc/unlink.html).
This is contrary to the definition of the POSIX unlink function, which allows it (this is the reason why it's called unlink).

I wasn't able to find any documentation for this discrepancy in ESP-IDF Programming Guide.
I think that it should be either properly documented (like the documentation added for explaining how the link function actually copies a file's content instead of just adding a second link), or it should be resolved in code by writing some special implementation of the unlink function.

Thanks!

haberturdeur, 2025-01-16:
Hi, thank you for reporting the issue!

Strictly speaking the wording of the standard allows for this in description of `EBUSY` errno (but this was likely not meant to allow such usecase).

Implementing this on devices where power loss is common and resources limited would be challenging, and likely would require separate implementation specific flags stored in flash, but this would still not solve all the problems.

For now, we will try to better document this issue.

The divergent behaviour of `link` is documented [in the FAT documentation](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/fatfs.html#using-fatfs-with-vfs).

Aric5301, 2025-01-18:
`EBUSY` will only be returned if `FF_FS_LOCK` is enabled. If not, the unlink will actually "work", but in practice that's of course venturing into the undefined behaviour teritory.
Adding this to the documentation like you've added the divergent behaviour of `link` is a great solution!

Thanks you Tomáš!

closed 2025-02-28 by espressif-bot