espressif/esp-idf #15190: ASSERT_PARAM(8192 0), in rwbt.c at line 360

vortex314, 2025-01-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

espidf 5.3.0

### Espressif SoC revision.

ESP32 devkit v1

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

Platformio with espidf

### Power Supply used.

USB

### What is the expected behavior?

The device works as a bluetooth to udp  bridge. The data from a PS4 controller is streamed to Zenoh middleware.

### What is the actual behavior?

It behaves correctly during a long time ( 8 min -> 56 min ) , no stack overflow or memory leak that drains the heap. 
Then it crashes in rwbt.c with an assertion error. I suspect that the Wifi and Bluetooth conflict somewhere and create a racing conditions

### Steps to reproduce.

1. Start the chip without any Bluetooth connection 
2. Connect PS4 controller to ESP32
3. After 8 to 55 min it crashes 

### Debug Logs.

```plain
I 00:08:28.913 |        main.cpp: 184 |  free heap size: 34144
I 00:08:29.921 |        main.cpp: 184 |  free heap size: 34396
I 00:08:30.945 |        main.cpp: 184 |  free heap size: 34380
I 00:08:31.953 |        main.cpp: 184 |  free heap size: 35104
ASSERT_PARAM(8192 0), in rwbt.c at line 360
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008ebfc: f01d020c 00004136 f01d0000
Core  0 register dump:
PC      : 0x4008ec03  PS      : 0x00060034  A0      : 0x800884da  A1      : 0x3ffc2108  
A2      : 0x00000000  A3      : 0x00002000  A4      : 0x00000000  A5      : 0x3ff9e7d2  
A6      : 0x00000168  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc2078  
A10     : 0x00000000  A11     : 0x3ffc209b  A12     : 0x3ffc2047  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x3ffc204c  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008ec00:0x3ffc2108 |<-CORRUPTED

ELF file SHA256: 77db749cd
[... 329 lines cut ...]
.coredump.tasks.data 0x3ffda7e0 0x240 RW 
.coredump.tasks.data 0x3ffbd6bc 0x154 RW 
.coredump.tasks.data 0x3ffdb810 0x210 RW 
.coredump.tasks.data 0x3fff83f8 0x154 RW 
.coredump.tasks.data 0x3fff8210 0x1e0 RW 
.coredump.tasks.data 0x3ffea734 0x154 RW 
.coredump.tasks.data 0x3ffea4c0 0x260 RW 
.coredump.tasks.data 0x3ffeb890 0x154 RW 
.coredump.tasks.data 0x3ffeb670 0x210 RW 
.coredump.tasks.data 0x3ffedc70 0x154 RW 
.coredump.tasks.data 0x3ffeda80 0x1e0 RW 
.coredump.tasks.data 0x3ffb7a50 0x154 RW 
.coredump.tasks.data 0x3ffb7890 0x1b0 RW 
.coredump.tasks.data 0x3ffb7318 0x154 RW 
.coredump.tasks.data 0x3ffb7160 0x1b0 RW 
.coredump.tasks.data 0x3ffddac4 0x154 RW 
.coredump.tasks.data 0x3ffdd890 0x220 RW 
.coredump.tasks.data 0x3fff6e9c 0x154 RW 
.coredump.tasks.data 0x3fff6a20 0x470 RW 
.coredump.tasks.data 0x3fff2f80 0x154 RW 
.coredump.tasks.data 0x3fff2d70 0x200 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!

vortex314, 2025-01-12:
See also : https://github.com/ricardoquesada/bluepad32/issues/137

ESP-YTGerd, 2025-01-13:
Hi, @vortex314,
Could you kindly provide the specific ESP-IDF version or the commit hash of the Bluetooth library you are using? The printing information you shared for v5.3.0 doesn't match our expected output, so it's possible that you're not using the exact v5.3.0 version.

vortex314, 2025-01-13:
@ESP-YTGerd  in the above screenshot you can see that at boot, it gives this line 
```
I (31) boot: ESP-IDF 5.3.0 2nd stage bootloader
```
This looks to confirm the 5.3.0 version, I use platformio to compile , see here : https://github.com/vortex314/zenoh-projects/tree/main/zenoh-esp32 .  So this defines the version I guess.

The crash is located in source rwbt.c based on an assertion. I suspect that the ASSERT ( 8192 0 ) indicates that it is running out of memory, when trying to allocate 8K block. This code is part of the espressif BT library, blob only.

Do you have access to this source code ? Can you validate the assumption ?

Any other way I can check the version ?

vortex314, 2025-01-13:
@ESP-YTGerd  I'll print the version  in the code . 
Another observation as I am using the [Bluepad32 library](https://github.com/ricardoquesada/bluepad32) to talk to the PS4 controller The bluepad32 [doc](https://bluepad32.readthedocs.io/en/latest/plat_esp32/) :  specifies to execute a script from 'btstack' which replaces the btstack of espressif with another one .  Don't know if this changes the blob concerned.
The [script](https://github.com/bluekitchen/btstack/blob/5d4d8cc7b1d35a90bbd6d5ffd2d3050b2bfc861c/port/esp32/integrate_btstack.py) from btstack seems to insert itself in the espressif provided one , the script is 3 years old.

ESP-YTGerd, 2025-01-13:
@vortex314 Maybe you could check the version of the ESP-IDF on your local setup? It seems that the script does not explicitly switch to a specific version but rather depends on the ESP-IDF version installed locally.

vortex314, 2025-01-13:
Version information : 
I (817) coexist: coex firmware version: dab85ae96
I (1073) BTDM_INIT: BT controller compile version [f021fb7]

Does this clarify something more ? 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080638
I (31) boot: ESP-IDF 5.3.0 2nd stage bootloader
I (31) boot: compile time Jan 12 2025 20:35:20
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (38) boot.esp32: SPI Speed      : 40MHz
I (43) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 2MB
I (52) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (61) boot: ## Label            Usage          Type ST Offset   Length
I (68) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (76) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (83) boot:  2 app0             OTA app          00 10 00010000 00170000
I (91) boot:  3 coredump         Unknown data     01 03 00180000 00020000
I (98) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=3321ch (209436) map
I (183) esp_image: segment 1: paddr=00043244 vaddr=3ffbdb60 size=061d0h ( 25040) load
I (192) esp_image: segment 2: paddr=0004941c vaddr=40080000 size=06bfch ( 27644) load
I (203) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=f20c4h (991428) map
I (543) esp_image: segment 4: paddr=001420ec vaddr=40086bfc size=15db4h ( 89524) load
I (593) boot: Loaded app from partition at offset 0x10000
I (593) boot: Disabling RNG early entropy source...
I (605) cpu_start: Multicore app
I (614) cpu_start: Pro cpu start user code
I (614) cpu_start: cpu freq: 160000000 Hz
I (614) app_init: Application information:
I (617) app_init: Project name:     zenoh-lib
I (622) app_init: App version:      609f9e2-dirty
I (628) app_init: Compile time:     Jan 13 2025 18:38:10
I (634) app_init: ELF file SHA256:  e4bc66dc6...
I (639) app_init: ESP-IDF:          5.3.0
I (644) efuse_init: Min chip rev:     v0.0
I (648) efuse_init: Max chip rev:     v3.99 
I (653) efuse_init: Chip rev:         v1.0
I (658) heap_init: Initializing. RAM available for dynamic allocation:
I (665) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (671) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (677) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (684) heap_init: At 3FFD89E0 len 00007620 (29 KiB): DRAM
I (690) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (696) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (703) heap_init: At 4009C9B0 len 00003650 (13 KiB): IRAM
I (711) spi_flash: detected chip: generic
I (713) spi_flash: flash io: dio
W (717) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I 00:00:00.022 |         actor.h:  23 | Channel created [10][4]  
I 00:00:00.027 |  wifi_actor.cpp:  23 | Starting WiFi actor sizeof(WifiCmd ) : 44 
I 00:00:00.035 |         actor.h:  23 | Channel created [6][4]  
I 00:00:00.040 | zenoh_actor.cpp:  16 | Starting WiFi actor sizeof(ZenohCmd ) : 48 
I 00:00:00.048 |         actor.h:  23 | Channel created [10][4]  
I 00:00:00.053 |   sys_actor.cpp:   6 | Starting Sys actor sizeof(SysCmd ) : 48 
I 00:00:00.060 |         actor.h:  23 | Channel created [10][4]  
I 00:00:00.066 |   ps4_actor.cpp:  20 | Starting PS4 actor sizeof(Ps4Cmd ) : 36 
[... 30 lines cut ...]
I (957) wifi_init: WiFi IRAM OP enabled
I (957) wifi_init: WiFi RX IRAM OP enabled
I (957) uart: queue free spaces: 10
I (969) phy_init: phy_version 4830,54550f7,Jun 20 2024,14:22:08
I 00:00:00.269 |   ps4_actor.cpp:  27 | 
btdm_controller_get_compile_version()=f021fb7
I 00:00:00.285 |   ps4_actor.cpp:  76 | custom: init()
I (1059) wifi:mode : sta (98:f4:ab:6d:2d:74)
I (1064) wifi:enable tsf
I (1073) BTDM_INIT: BT controller compile version [f021fb7]
I (1088) BTDM_INIT: Bluetooth MAC: 98:f4:ab:6d:2d:76
I 00:00:00.461 |  wifi_actor.cpp: 432 | Max AP number ap_info can hold = 10
I (1368) wifi:new:<1,0>, old:<1,0>, ap:<255,255>, sta:<1,0>, prof:1, snd_ch_cfg:0x0
I (1382) wifi:state: init -> auth (0xb0)
I 00:00:00.683 |  wifi_actor.cpp: 435 | Total APs scanned = 0, actual AP number ap_info holds = 0
I 00:00:00.695 |  wifi_actor.cpp: 455 | Highest RSSI = -128, AP index = 0
I 00:00:00.703 |  wifi_actor.cpp:  41 | Scanned SSID: I (1395) wifi:
state: auth -> assoc (0x0)
I 00:00:00.735 |  wifi_actor.cpp: 148 | Connecting to WiFi network...
I (1450) wifi:state: assoc -> run (0x10)
I (1469) wifi:<ba-add>idx:0 (ifx:0, 00:23:cd:19:fc:f5), tid:0, ssn:0, winSize:64
I (1498) wifi:connected with Merckx2, aid = 3, channel 1, BW20, bssid = 00:23:cd:19:fc:f5
I (1512) wifi:security: WPA2-PSK, phy: bgn, rssi: -33
I (1542) wifi:pm start, type: 1
```

vortex314, 2025-01-13:
Not sure where platformio seeks for espidf , because it looks that my normal install points to version 5.4 and not 5.3 like shown in the logs.
```
lieven@pcdell:~/esp/esp-idf$ . export.sh
Checking "python3" ...
Python 3.10.12
"python3" has been detected
Activating ESP-IDF 5.4
* Checking python version ... 3.10.12
* Checking python dependencies ... OK
* Deactivating the current ESP-IDF environment (if any) ... OK
* Establishing a new ESP-IDF environment ... OK
* Identifying shell ... bash
* Detecting outdated tools in system ... Found tools that are not used by active ESP-IDF version.
For removing old versions of esp32ulp-elf, xtensa-esp-elf-gdb, openocd-esp32, xtensa-esp32s2-elf, xtensa-esp32-elf, xtensa-esp-elf, riscv32-esp-elf-gdb, riscv32-esp-elf, xtensa-esp32s3-elf, esp-rom-elfs use command 'python 
/home/lieven/esp/esp-idf/tools/idf_tools.py uninstall'
To free up even more space, remove installation packages of those tools.
Use option python /home/lieven/esp/esp-idf/tools/idf_tools.py uninstall --remove-archives.
* Shell completion ... Autocompletion code generated

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
lieven@pcdell:~/esp/esp-idf$ idf.py --version
ESP-IDF v5.4
```

vortex314, 2025-01-13:
AFter upgrading the espressif framework in platformio :+1: 
I 00:00:00.253 |   ps4_actor.cpp:  27 | btdm_controller_get_compile_version()=b022216
I (1061) BTDM_INIT: BT controller compile version [b022216]
The ESPIDF version mentioned is still 5.3

ESP-YTGerd, 2025-01-14:
@vortex314  The commit  `f021fb7` is from June last year, and it should already include the fix for the issue you mentioned. If you can confirm that the Bluetooth library you’re using is from this commit, could you try running the tests again to see if the problem still occurs?

vortex314, 2025-01-14:
@ESP-YTGerd as I upgraded platformio build system, it brought in also a newer version of ESPIDF 5.3.1 and a new version of the BT controller version : b022216. I guess b022216 is newer than f021fb7 ? 
The crashes still occur ( after some time ) , I'll check if the coredump is also the same.

ESP-YTGerd, 2025-01-14:
> @ESP-YTGerd as I upgraded [platformio](https://github.com/platformio/platform-espressif32/releases) to 6.9.0 build system, it brought in also a newer version of ESPIDF 5.3.1 and a new version of the BT controller version : b022216. I guess b022216 is newer than f021fb7 ? The crashes still occur ( after some time ) , I'll check if the coredump is also the same.

@vortex314  Yes, Is the assert information still like `ASSERT_PARAM(8192 0)`?

vortex314, 2025-01-14:
@ESP-YTGerd  it's still the same issue , it runs fine without connecting to the PS4, up to 2 hours.
After connecting with the PS4 controller , it dies after 5 min. 
As you see in the logs below it's with ESP-IDF 5.3.1 and BT version btdm_controller_get_compile_version()=b022216
```
```
I 02:03:47.756 |        main.cpp: 194 |  free heap size: 19212 biggest block : 18432 
I 02:03:48.769 |        main.cpp: 194 |  free heap size: 17416 biggest block : 16384 
I 02:03:49.788 |        main.cpp: 194 |  free heap size: 17304 biggest block : 16384 
I 02:03:50.822 |        main.cpp: 194 |  free heap size: 21000 biggest block : 19456 
ASSERT_PARAM(8192 0), in rwbt.c at line 360
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008eeac: f01d020c 00004136 f01d0000
Core  0 register dump:
PC      : 0x4008eeb3  PS      : 0x00060134  A0      : 0x80088376  A1      : 0x3ffc2670  
A2      : 0x00000000  A3      : 0x00002000  A4      : 0x00000000  A5      : 0x3ff9e7d2  
A6      : 0x00000168  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc25e0  
A10     : 0x00000000  A11     : 0x3ffc2603  A12     : 0x3ffc25af  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x3ffc25b4  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008eeb0:0x3ffc2670 0x40088373:0x3ffc2690 0x40055749:0x3ffc26b0 0x4008c03b:0x3ffc26d0 0x40081255:0x3ffc26f0 0x400811c1:0x3ffc2710 0x400811a3:0x00000000 |<-CORRUPTED

ELF file SHA256: e45ca4550

I (23005) esp_core_dump_flash: Save core dump to flash...
I (23013) esp_core_dump_flash: Erase flash 28672 bytes @ 0x180000
I (23416) esp_core_dump_flash: Write end offset 0x6e04, check sum length 4
I (23416) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:7176
load:0x40078000,len:15564
ho 0 tail 12 room 4
load:0x40080400,len:4
load:0x40080404,len:3904
entry 0x40080638
I (31) boot: ESP-IDF 5.3.1 2nd stage bootloader
I (31) boot: compile time Jan 14 2025 01:25:13
I (31) boot: Multicore bootloader
I (35) boot: chip revision: v1.0
I (39) boot.esp32: SPI Speed      : 40MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (76) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (84) boot:  2 app0             OTA app          00 10 00010000 00170000
I (91) boot:  3 coredump         Unknown data     01 03 00180000 00020000
I (99) boot: End of partition table
I (103) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=33a90h (211600) map
I (184) esp_image: segment 1: paddr=00043ab8 vaddr=3ffbdb60 size=06270h ( 25200) load
I (194) esp_image: segment 2: paddr=00049d30 vaddr=40080000 size=062e8h ( 25320) load
I (204) esp_image: segment 3: paddr=00050020 vaddr=400d0020 size=f0cc4h (986308) map
I (542) esp_image: segment 4: paddr=00140cec vaddr=400862e8 size=1695ch ( 92508) load
I (593) boot: Loaded app from partition at offset 0x10000
I (593) boot: Disabling RNG early entropy source...
I (605) cpu_start: Multicore app
I (614) cpu_start: Pro cpu start user code
I (614) cpu_start: cpu freq: 160000000 Hz
I (615) app_init: Application information:
I (617) app_init: Project name:     zenoh-lib
I (622) app_init: App version:      609f9e2-dirty
I (628) app_init: Compile time:     Jan 14 2025 00:27:08
[... 78 lines cut ...]
I 00:00:00.816 |   ps4_actor.cpp:  99 | custom: on_init_complete()
I 00:00:00.824 |   ps4_actor.cpp: 231 | custom: Bluetooth enabled: 1
A9 45 EA CB 5A EF 53 2F E9 3F 7F 90 54 8D 38 75 
I (1727) wifi:<ba-add>idx:1 (ifx:0, 00:23:cd:19:fc:f5), tid:6, ssn:0, winSize:64
I 00:00:01.187 |        main.cpp: 194 |  free heap size: 54812 biggest block : 53248 
I 00:00:02.193 |        main.cpp: 194 |  free heap size: 54812 biggest block : 53248 
I (2945) esp_netif_handlers: sta ip: 192.168.0.231, mask: 255.255.255.0, gw: 192.168.0.1
I 00:00:02.312 | zenoh_actor.cpp:  52 | Connected to Zenoh.
I 00:00:02.316 | zenoh_actor.cpp: 156 | Declaring subscriber for 'dst/lm1/**'...
I 00:00:02.326 | zenoh_actor.cpp: 165 | OK
I 00:00:03.204 |        main.cpp: 194 |  free heap size: 33084 biggest block : 30720 
I 00:00:04.222 |        main.cpp: 194 |  free heap size: 34616 biggest block : 30720 
I 00:00:05.233 |        main.cpp: 194 |  free heap size: 32672 biggest block : 30720 
I 00:00:06.244 |        main.cpp: 194 |  free heap size: 32672 biggest block : 30720 
I 00:00:07.255 |        main.cpp: 194 |  free heap size: 32672 biggest block : 30720 
I 00:00:08.266 |        main.cpp: 194 |  free heap size: 32904 biggest block : 30720 
I 00:00:09.280 |        main.cpp: 194 |  free heap size: 32904 biggest block : 30720 
I 00:00:10.290 |        main.cpp: 194 |  free heap size: 32904 biggest block : 30720 
I 00:00:11.301 |        main.cpp: 194 |  free heap size: 32896 biggest block : 30720 
I 00:00:12.317 |        main.cpp: 194 |  free heap size: 34452 biggest block : 30720 
I 00:00:13.329 |        main.cpp: 194 |  free heap size: 34460 biggest block : 30720 
I 00:00:14.345 |        main.cpp: 194 |  free heap size: 32672 biggest block : 30720 
I 00:00:15.356 |        main.cpp: 194 |  free heap size: 32668 biggest block : 30720 
I 00:00:16.366 |        main.cpp: 194 |  free heap size: 32672 biggest block : 30720 
```

vortex314, 2025-01-14:
The last time it run was from 17:00 -> 22:16. So about 5 hours. I don't have the message that appeared. The core dump is below and looks the same as previous cases
```
Checking "python3" ...
Python 3.10.12
"python3" has been detected

Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build
===============================================================
==================== ESP32 CORE DUMP START ====================
The ROM ELF file won't load automatically since it was not found for the provided chip type.

Crashed task handle: 0x3ffdfa9c, name: 'wifi', GDB name: 'process 1073609372'
Crashed task is not in the interrupt context

================== CURRENT THREAD REGISTERS ===================
exccause       0xffff (InvalidCauseRegister)
excvaddr       0x0
pc             0x20000000          0x20000000
lbeg           0x4000c2e0          1073791712
lend           0x4000c2f6          1073791734
lcount         0xffffffff          4294967295
sar            0x17                23
ps             0x30                48
threadptr      <unavailable>
br             <unavailable>
scompare1      <unavailable>
acclo          <unavailable>
acchi          <unavailable>
m0             <unavailable>
m1             <unavailable>
m2             <unavailable>
m3             <unavailable>
expstate       <unavailable>
f64r_lo        <unavailable>
f64r_hi        <unavailable>
f64s           <unavailable>
fcr            <unavailable>
fsr            <unavailable>
a0             0x20000070          536871024
a1             0x0                 0
a2             0x0                 0
a3             0x0                 0
a4             0x60820             395296
a5             0x60823             395299
a6             0x12d025b           19726939
a7             0x0                 0
a8             0x3ff000dc          1072693468
a9             0x1                 1
a10            0x3ffbde40          1073471040
a11            0xffffffff          -1
a12            0x3ffbba64          1073461860
a13            0x60823             395299
a14            0x1                 1
a15            0xcdcd              52685

==================== CURRENT THREAD STACK =====================
#0  0x20000000 in ?? ()

======================== THREADS INFO =========================
  Id   Target Id          Frame 
* 1    process 1073609372 0x20000000 in ?? ()
  2    process 1073461856 0x4000bff0 in ?? ()
  3    process 1073581428 0x4008f086 in esp_cpu_wait_for_intr () at /home/lieven/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:64
  4    process 1073583568 0x4008f086 in esp_cpu_wait_for_intr () at /home/lieven/.platformio/packages/framework-espidf/components/esp_hw_support/cpu.c:64
  5    process 1073689376 0x4000bff0 in ?? ()
  6    process 1073594512 0x4000bff0 in ?? ()
  7    process 1073689720 0x40081d0c in esp_crosscore_int_send_yield (core_id=0) at /home/lieven/.platformio/packages/framework-espidf/components/esp_system/crosscore_int.c:121
[... 240 lines cut ...]
.coredump.tasks.data 0x3ffe2370 0x210 RW 
.coredump.tasks.data 0x3ffbd2c0 0x154 RW 
.coredump.tasks.data 0x3ffbd070 0x240 RW 
.coredump.tasks.data 0x3ffdfbf4 0x154 RW 
.coredump.tasks.data 0x3ffe1370 0x210 RW 
.coredump.tasks.data 0x3ffe3d78 0x154 RW 
.coredump.tasks.data 0x3ffe3b10 0x260 RW 
.coredump.tasks.data 0x3ffe4ec0 0x154 RW 
.coredump.tasks.data 0x3ffe4ca0 0x210 RW 
.coredump.tasks.data 0x3ffe8260 0x154 RW 
.coredump.tasks.data 0x3ffe8070 0x1e0 RW 
.coredump.tasks.data 0x3ffb7a50 0x154 RW 
.coredump.tasks.data 0x3ffb7890 0x1b0 RW 
.coredump.tasks.data 0x3ffdd340 0x154 RW 
.coredump.tasks.data 0x3ffdd110 0x220 RW 
.coredump.tasks.data 0x3ffb7318 0x154 RW 
.coredump.tasks.data 0x3ffb7160 0x1b0 RW 
.coredump.tasks.data 0x3ffed658 0x154 RW 
.coredump.tasks.data 0x3ffed450 0x200 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!

```

vortex314, 2025-01-15:
@ESP-YTGerd  can you share the  piece of code and what this assert is testing ? Maybe it can give me a clue where to look for.

vortex314, 2025-01-18:
@ESP-YTGerd  I have tried all possible combinations of threading and memory allocation , it always ends in a crash at line 360 for rwbt.c. Any guess on what I could try further ?

vortex314, 2025-01-18:
Could this be caused by the fact that the process is not tight to a single CPU ? 
  8    process 1073462580 0x40081d48 in esp_crosscore_int_send_yield (core_id=0) at /home/lieven/.platformio/packages/framework-espidf/components/esp_system/crosscore_int.c:121

As I see that there is a crosscore yield. So I'm testing now to push all threads on 1 core.

vortex314, 2025-01-19:
https://community.platformio.org/t/espressif-framework-update-replace-components-with-esp-idf-recent-ones/45300/3

platformio upgrade let's check

Looks indeed bringing a new version of the bluetooth software :
btdm_controller_get_compile_version()=20b459d
was previously
btdm_controller_get_compile_version()=b022216
Could solve some stability issues.

vortex314, 2025-01-19:
And after 1h 8min back to square one , despite new version
```
I 01:08:45.205 | zenoh_actor.cpp:  27 | Timer publish : Publishing Zenoh properties
I 01:08:45.208 |   sys_actor.cpp:  30 | Timer 1 : Publishing Sys properties
ASSERT_PARAM(8192 0), in rwbt.c at line 360
Guru Meditation Error: Core  0 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x4008f424: f01d020c 00004136 f01d0000
Core  0 register dump:
PC      : 0x4008f42b  PS      : 0x00060334  A0      : 0x8008856e  A1      : 0x3ffc270c  
A2      : 0x00000000  A3      : 0x00002000  A4      : 0x00000000  A5      : 0x3ff9e7d2  
A6      : 0x00000168  A7      : 0xfffffffc  A8      : 0x8000814b  A9      : 0x3ffc267c  
A10     : 0x00000000  A11     : 0x3ffc269f  A12     : 0x3ffc264b  A13     : 0x00000030  
A14     : 0x00000000  A15     : 0x3ffc2650  SAR     : 0x00000004  EXCCAUSE: 0x00000000  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0xffffffff  

Backtrace: 0x4008f428:0x3ffc270c |<-CORRUPTED

ELF file SHA256: 44d117f90

I (19825) esp_core_dump_flash: Save core dump to flash...
I (19832) esp_core_dump_flash: Erase flash 24576 bytes @ 0x280000
I (20205) esp_core_dump_flash: Write end offset 0x5b04, check sum length 4
I (20205) esp_core_dump_flash: Core dump has been saved to flash.
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6276
load:0x40078000,len:15716
load:0x40080400,len:4
ho 8 tail 4 room 4
load:0x40080404,len:3860
entry 0x40080634
I (30) boot: ESP-IDF 5.4.0 2nd stage bootloader
I (31) boot: compile time Jan 19 2025 18:34:10
I (31) boot: Multicore bootloader
I (32) boot: chip revision: v1.0
I (35) boot.esp32: SPI Speed      : 40MHz
I (39) boot.esp32: SPI Mode       : DIO
I (42) boot.esp32: SPI Flash Size : 4MB
I (46) boot: Enabling RNG early entropy source...
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (59) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (66) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (72) boot:  2 app0             OTA app          00 10 00010000 00270000
I (79) boot:  3 coredump         Unknown data     01 03 00280000 00020000
I (85) boot: End of partition table
I (89) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=337c8h (210888) map
I (168) esp_image: segment 1: paddr=000437f0 vaddr=3ff80000 size=0001ch (    28) load
I (168) esp_image: segment 2: paddr=00043814 vaddr=3ffbdb60 size=0630ch ( 25356) load
I (181) esp_image: segment 3: paddr=00049b28 vaddr=40080000 size=064f0h ( 25840) load
I (192) esp_image: segment 4: paddr=00050020 vaddr=400d0020 size=ee5fch (976380) map
I (525) esp_image: segment 5: paddr=0013e624 vaddr=400864f0 size=16ec4h ( 93892) load
I (576) boot: Loaded app from partition at offset 0x10000
I (576) boot: Disabling RNG early entropy source...
I (587) cpu_start: Multicore app
I (596) cpu_start: Pro cpu start user code
I (596) cpu_start: cpu freq: 160000000 Hz
I (596) app_init: Application information:
I (599) app_init: Project name:     zenoh-lib
I (604) app_init: App version:      1dedd31-dirty
I (610) app_init: Compile time:     Jan 19 2025 18:31:53
I (616) app_init: ELF file SHA256:  44d117f90...
I (621) app_init: ESP-IDF:          5.4.0
I (626) efuse_init: Min chip rev:     v0.0
I (630) efuse_init: Max chip rev:     v3.99 
[... 62 lines cut ...]

I (1084) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (1091) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I 00:00:00.410 |   ps4_actor.cpp:  86 | custom: init()
I (1105) BTDM_INIT: BT controller compile version [20b459d]
I (1109) BTDM_INIT: Bluetooth MAC: 30:ae:a4:ff:22:82
I 00:00:00.432 |  wifi_actor.cpp: 423 | Max AP number ap_info can hold = 10
I 00:00:00.436 |  wifi_actor.cpp: 426 | Total APs scanned = 0, actual AP number ap_info holds = 0
I 00:00:00.439 |  wifi_actor.cpp: 446 | Highest RSSI = -128, AP index = 0
I 00:00:00.447 |  wifi_actor.cpp:  42 | Scanned SSID: 
W (1143) wifi:Password length matches WPA2 standards, authmode threshold changes from OPEN to WPA2
I 00:00:00.464 |  wifi_actor.cpp: 147 | Connecting to WiFi network...
I (1393) wifi:<ba-add>idx:1 (ifx:0, 00:23:cd:19:fc:f5), tid:6, ssn:0, winSize:64
BTstack up and running at 30:AE:A4:FF:22:82
I 00:00:00.787 |   ps4_actor.cpp: 109 | custom: on_init_complete()
I 00:00:00.788 |   ps4_actor.cpp: 244 | custom: Bluetooth enabled: 1
DC CB 06 19 63 46 BD 60 72 7D 78 07 52 FC 4E CB 
I 00:00:01.014 |  wifi_actor.cpp:  72 | Timer 1 : Publishing WiFi properties
I 00:00:01.019 | zenoh_actor.cpp:  27 | Timer publish : Publishing Zenoh properties
I 00:00:01.025 |   sys_actor.cpp:  30 | Timer 1 : Publishing Sys properties
I 00:00:02.015 |  wifi_actor.cpp:  72 | Timer 1 : Publishing WiFi properties
I 00:00:02.020 | zenoh_actor.cpp:  27 | Timer publish : Publishing Zenoh properties
I 00:00:02.028 |   sys_actor.cpp:  30 | Timer 1 : Publishing Sys properties
I (2812) esp_netif_handlers: sta ip: 192.168.0.231, mask: 255.255.255.0, gw: 192.168.0.1
```

ESP-YTGerd, 2025-01-20:
Hi, @vortex314 
Maybe I have found the root cause. Can you check the version of the `bt.c`  (in path `esp-idf/components/bt/controller/esp32/` )file you're currently using? If it's the latest version, it should include calls to the following functions.

[image]

The Bluetooth library needs to be updated, and the corresponding bt.c file should also be updated.

vortex314, 2025-01-20:
Hi @ESP-YTGerd  , I'm guessing that platformio from within .vscode is using this directory for the files mentioned 
~/.platformio/packages/framework-espidf/components/bt/controller/esp32

[image]

[image]

ESP-YTGerd, 2025-01-20:
> Hi [@ESP-YTGerd](https://github.com/ESP-YTGerd) , I'm guessing that platformio from within .vscode is using this directory for the files mentioned ~/.platformio/packages/framework-espidf/components/bt/controller/esp32
> 
> [image]
> 
> [image]
> 
> ```
> I (596) app_init: Application information:
> I (599) app_init: Project name:     zenoh-lib
> I (604) app_init: App version:      1dedd31-dirty
> I (610) app_init: Compile time:     Jan 19 2025 18:31:53
> I (616) app_init: ELF file SHA256:  44d117f90...
> I (621) app_init: ESP-IDF:          5.4.0
> I (626) efuse_init: Min chip rev:     v0.0
> I (630) efuse_init: Max chip rev:     v3.99 
> I (635) efuse_init: Chip rev:         v1.0
> ```
> 
> [image]

@vortex314  Aha,the  sdkconfig config is `config_btdm_ctrl_mode_ble_only`. We should select `config_btdm_ctrl_mode_btdm` or `config_btdm_ctrl_mode_bredr_only`

vortex314, 2025-01-20:
ok, will try. keep you posted.

closed 2025-01-20 by vortex314