espressif/esp-idf #15191: [Feature] in esp-idf, is there a way to log all USB communication?

chipweinberger, 2025-01-13:
### Is your feature request related to a problem?

For Windows, there is a [USB Wireshark sniffer](https://wiki.wireshark.org/CaptureSetup/USB). 

Is there any way to log communications in esp-idf?

### Describe the solution you'd like.

A KCONFIG to increase USB logging

chipweinberger, 2025-01-13:
@Dazza0 
@roma-jam 
@peter-marcisovsky

peter-marcisovsky, 2025-01-13:
Hi @chipweinberger,

The only esp-idf setting, which would allow you to log USB communication within the esp-idf, during the code execution is just increasing log level from Info to Debug, or Verbose in menuconfig.

`Component Config -> Log -> Log Level -> Default log verbosity`

Do you have some specific request for more detailed logging for USB applications in esp-idf?

chipweinberger, 2025-01-13:
Thanks for your reply Peter!

Yes, I'm looking for a way to just log USB traffic, so I can debug why some USB devices work with Windows but not with my ESP32-S3 code.

Dazza0, 2025-01-20:
@peter-marcisovsky @chipweinberger Logging USB traffic might be tricky depending on what granularity of data we want to log.

- **Bit/symbol level** (similar to what you get from a logic analyzer): This probably isn't possible. The DWC OTG hardware handles all bit/symbol encoding/decoding, thus there isn't a way for the SW to get the raw bits sent.
- **Packet level**: Similarly, the hardware handles all USB packet composition/parsing. The endpoint number and type are simply set in a register (HCCHAR) and the hardware is responsible for composing/parsing packets according the the EP information.
- **Transaction level**: Feasible if non-dma (i.e., slave) mode. In non-dma mode, the SW is responsible for triggering each individual transaction and handling their completion. In theory, the handler can just dump the data of a just-completed transaction into some logging buffer, along with the EP information from the HCCHAR register (i.e., EPnum and type).
- **Transfer level**:  Feasible if buffer-dma mode is supported. In buffer-dma mode, the SW is responsible for triggering each transfer, and the HW will handle all individual transactions. Thus, the SW can just dump the entire transfer's data buffer into a logging buffer from the completion interrupt. However, the details of each transaction/packets are not recoverable.
- **Multi-transfer level**: Doable in the current scatter-gather DMA mode implementation. Similarly, the details of each transfer are lost.

Another thing to consider is whether the copying of data to the log buffer is handled on the "driver-side" or "user-side", as this may have timing implications.

- **Driver-side**: The driver/host-stack will copy all data buffers to the log buffer before generating any completion event to the user-side. However, I would imagine this could impose a short delay which could in theory lead to race conditions.
- **User-side**: Provide the user an API (e.g., `usb_host_dump(usb_transfer_t *transfer, some_log_buf_handle_t handle)`) to dump the data to the log buffer at their leisure (e.g., deferring it to the IDLE task such that no delay is imposed on USB driver operations). In the current  `usb_host.h` API, users will need to pass the entire `usb_transfer_t` object as the `data_buffer` field is `const`. Allowing the `data_buffer` to be dynamically assigned would probably allow users to recycle `usb_transfer_t` objects.

I'm not too sure about data capture formats, but I presume there's some sort of prefix we need to add in front of each transfer that gets dumped to indicate the EP/length of each transfer.

The interface used to push the capture data from the log buffer to host may also impose limitations. For example, using UART throughput will probably the outpaced by USB throughput leading to log buffer overflows.

TLDR: Having a logging feature is doable, but will come with some limitations and will take some work in the Host stack to support it.

chipweinberger, 2025-01-20:
@Dazza0 , thanks for your well considered reply. Transfer level would still be very useful. I hope it gets implemented.