espressif/esp-idf #15195: About memory overflow

learnerzcz, 2025-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My issue is when i build  a project, the output log prompt the region overflow. it's expect behavior is to display a peciece of audio. And the .pcm file sizeof is about 4MB.   

I have attemp to use custom partition table  to enlarge the factory region size. But it seem not valid to slove the problem . Because the overflowed  space is not reduse. 

The log like this. If need more information,please propose.

```
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: i2s-es8311-example.elf section `.flash.rodata' will not fit in region `drom0_0_seg'
 C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: region `drom0_0_seg' overflowed by 362616 bytes
```

L-KAYA, 2025-01-14:
Hi @learnerzcz, seems you are using ESP32-S3 right? And the issue is that your customized audio is too large to fit your Flash?

Please check the Flash size of your S3 first, you can find the info in the product introduction [website](https://www.espressif.com.cn/en/products/modules). For example, you're using ESP32-S3-WROOM-1, it might carry ESP32-S3, ESP32-S3R2 or ESP32-S3R8 chip, then you can find the actual Flash size it carries. For example it carries a 8MB Flash:

Then you can run `idf.py menuconfig`, and choose `Serial flasher config -> Flash size` to 8MB.

[image]

And then use your customized partition table for this large binary. Or just use the default large app partition in `Partition Table -> Partition Table`

[image]

You can save it as the default configurations by the command `idf.py save-defconfig`.

However, if your Flash only sizes `4MB`, you can never download a image that `4714433 bytes` large.

learnerzcz, 2025-01-14:
@L-KAYA I have used a customed par.csv file,and set configed the flash size. But it not able to solve the problem.

L-KAYA, 2025-01-14:
@learnerzcz  Could you take a try with default large app partition, and provide your partition if possible, thanks!

learnerzcz, 2025-01-14:
sorry too  later reply @L-KAYA   I build my project success.  
But i have some confusions , have to trouble you. I set  target board to esp32s3 and enlarge flash size, It is ok . however when i set the target to esp32 . and nothing to change . the project is not run. I want to know why?
This is my partition table
# Espressif ESP32 Partition Table,,,,
# Name,     Type,   SubType,    Offset,     Size
nvs,        data,   nvs,        0x9000,     0x4000
otadata,    data,   ota,        0xd000,     0x2000
phy_init,   data,   phy,        0xf000,     0x1000
factory,    app,  factory,                   , 7M,

And i will refer the file list
```

#include <stdint.h>
#include <stdlib.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/i2s_std.h"
#include "driver/gpio.h"
#include "esp_check.h"

```

learnerzcz, 2025-01-14:
I think the audio develop is compatible for esp32 and esp32s3

L-KAYA, 2025-01-14:
@learnerzcz  So your ESP32 also carries a 8MB Flash?

learnerzcz, 2025-01-14:
I want to say is not that , becase i only set the serial flash size. and not download to board . I think it is not important the real size of my esp32 

> @learnerzcz So your ESP32 also carries a 8MB Flash?

learnerzcz, 2025-01-14:
yes i still facing the issue 

> Yes, the example is compatible to ESP32 and ESP32-S3, are you still facing the insufficient memory error?

L-KAYA, 2025-01-14:
Hi @KonstantinKondrashov, could you help with this partition issue?

L-KAYA, 2025-01-14:
Suggestion from @KonstantinKondrashov:

Maybe you can try to remove all the `Offset` in your partition table.

learnerzcz, 2025-01-14:
Thank you for your suggestion @L-KAYA  @KonstantinKondrashov , but  it had no effect .

> Maybe you can try to remove all the `Offset` in your partition table. Because there is partition table offset that has to be used todefine

And i set target to esp32c3, it also can build ok

learnerzcz, 2025-01-14:
@igrr  Thank you  . But i have set the app size to 4MB . there is no reduce of the overflowed size . That is normal ?

`C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: region `drom0_0_seg' overflowed by **362616** bytes`

suda-morris, 2025-08-11:
> And the .pcm file sizeof is about 4MB.

Can we save the pcm file in the filesystem, rather than embedding it in your firmware binary?

ginkgm, 2025-09-24:
Hi @learnerzcz ,

THe check is here: https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp_system/ld/esp32/memory.ld.in#L86 , independent of the configured flash size or partition size.

And it's a restriction of hardware, we can't fix it from SW.

Issue closed due to no reply

closed 2025-09-24 by ginkgm