espressif/esp-idf #15196: install.sh doesn't install to the correct directory

inuitviking, 2025-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

That the files and directories ended up in their proper directories, and looks something like so in `~./espressif/python_env/idf5.4_py3.10_env`:

```
drwxr-xr-x 3 crt@hamag-2011.local domain users@hamag-2011.local 4096 Jan 13 15:59 bin
-rw-r--r-- 1 crt@hamag-2011.local domain users@hamag-2011.local    3 Jan 13 15:58 idf_version.txt
drwxr-xr-x 2 crt@hamag-2011.local domain users@hamag-2011.local 4096 Jan 13 15:58 include
drwxr-xr-x 3 crt@hamag-2011.local domain users@hamag-2011.local 4096 Jan 13 15:58 lib
lrwxrwxrwx 1 crt@hamag-2011.local domain users@hamag-2011.local    3 Jan 13 15:58 lib64 -> lib
-rw-r--r-- 1 crt@hamag-2011.local domain users@hamag-2011.local   71 Jan 13 15:58 pyvenv.cfg
```

### What is the actual behavior?

Everything seems to go into `~/.espressif_old/python_env/idf5.4_py3.10_env/bin` instead of just `~/.espressif_old/python_env/idf5.4_py3.10_env`.

### Steps to reproduce.

1. Step: `mkdir -p ~/esp && cd ~/esp`
2. Step: `git clone -b v5.4 --recursive https://github.com/espressif/esp-idf.git esp-idf && esp-idf`
3. Step: `./install.sh all` ← This likely causes the error
4. Step: `. ./export.sh` ← Error becomes apparent here
5. Step: Go to your project
6. Step: `idf.py build` ← Error is also apparent here

### More Information.

I can work around this by going to `~/.espressif/python_env/idf5.4_py3.10_env/bin` and run the following commands:

```bash
 mv idf_version.txt ../
mv include ../
mv lib* ../
mv pyvenv.cfg ../
mv bin/* .
rm -rf bin
```

After that, it works.

This seems like it just installs the files in the wrong directory, or a level too deep. :smile:

I'm not certain where the error occurs, though.

fhrbata, 2025-01-13:
Hello @inuitviking , 

isn't it possible that the  `IDF_PYTHON_ENV_PATH`  environment variable is actually set to `~/.espressif_old/python_env/idf5.4_py3.10_env/bin`? Can you please check its value with `echo $IDF_PYTHON_ENV_PATH` before running `./install.sh`?

Thank you

inuitviking, 2025-01-14:
> Hello @inuitviking ,
> 
> isn't it possible that the `IDF_PYTHON_ENV_PATH` environment variable is actually set to `~/.espressif_old/python_env/idf5.4_py3.10_env/bin`? Can you please check its value with `echo $IDF_PYTHON_ENV_PATH` before running `./install.sh`?
> 
> Thank you

```bash
$ echo $IDF_PYTHON_ENV_PATH
~/.espressif/python_env/idf5.4_py3.10_env/bin/
```

Sure does! I changed it to `~/.espressif/python_env/idf5.4_py3.10_env/` and it works. Sorry for any inconvenience for this user error. :sweat_smile:

closed 2025-01-14 by inuitviking