espressif/esp-idf #15197: Can't send / sniff NDF packet

roger-, 2025-01-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.5

### Espressif SoC revision.

ESP32-S2 (revision v0.0) 

### Operating System used.

Linux

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S2-Saola-1

### Power Supply used.

USB

### What is the expected behavior?

I'm generating a null data frame and transmitting it with `esp_wifi_80211_tx()`. The destination MAC is a random device and I'm setting the source MAC in the frame to a unique MAC (which I filter later).

I'm then sniffing and trying to find the ACK from the the destination device, but I'm not receiving a response. 

It worked in the past with 4.x.

### What is the actual behavior?

I don't observe the expected packet.

### Steps to reproduce.

My transmit task looks like this:

```c
    uint8 MAC_ADDR_SRC[6] = {0x00, 0x00, 0x88, 0x88, 0x88, 0x88}; // meant to be invalid
    ieee80211::frame_data_null_t pkt;
    memset(&pkt, 0, sizeof(pkt));
    pkt.frame_control.type = ieee80211::WIFI_FC_DATA;
    pkt.frame_control.subtype = ieee80211::WIFI_NULL;
    pkt.frame_control.to_ds = 1;

    pkt.dest  = MAC_ADDR_DST;
    pkt.src   = MAC_ADDR_SRC;
    pkt.bssid = MAC_ADDR_DST;

    ESP_ERROR_CHECK(esp_wifi_config_80211_tx_rate(WIFI_IF_STA, WIFI_PHY_RATE_24M));

    ESP_LOGE(TAG, "starting NDP ping: %s -> %s", format_macaddr(pkt.src).c_str(),
                                                 format_macaddr(pkt.dest).c_str());

    while(true) {
        ESP_LOGE(TAG, "sending 802.11 NDP");
        ESP_ERROR_CHECK(esp_wifi_80211_tx(WIFI_IF_STA, (void *)&pkt, sizeof(pkt), true));

        vTaskDelay(this_obj->ping_interval_ms / portTICK_PERIOD_MS);
    }
```

And my promiscuous setup code:

```c
    wifi_promiscuous_filter_t filter = {
        .filter_mask = WIFI_PROMIS_FILTER_MASK_CTRL 
    };

    wifi_promiscuous_filter_t filter_ctrl = {
        .filter_mask = WIFI_PROMIS_CTRL_FILTER_MASK_ACK
    };

    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_filter(&filter));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_ctrl_filter(&filter_ctrl));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous_rx_cb(&callback_wifi_promiscuous));
    ESP_ERROR_CHECK(esp_wifi_set_promiscuous(true));
```

And my promiscuous callback:

```c
    wifi_promiscuous_pkt_t *p = (wifi_promiscuous_pkt_t *)buf;
    ieee80211::frame_ctrl_ack_t *frame_ack = (ieee80211::frame_ctrl_ack_t *)(p->payload);

    // check if actually ACK
    if(frame_ack->frame_control.type != ieee80211::WIFI_FC_CTRL || frame_ack->frame_control.subtype != ieee80211::WIFI_ACK) {
        return;
    }

    if(memcmp((void *)&frame_ack->dest, (void *)&MAC_ADDR_SRC, sizeof(ieee80211::mac_address_t)) == 0) {
        ESP_LOGE(TAG, "FOUND ACK");
    }
```

zhangyanjiaoesp, 2025-01-16:
@roger- 
Have you used Wireshark to capture packets? Can you capture null data packets and ACK packets?

roger-, 2025-01-17:
@zhangyanjiaoesp I checked in Wireshark and noticed a flood of RTS messages and an inconsistent ACK. Things were better when I sent the packets to another ESP32 and when I used the undocumented API [here](https://github.com/espressif/esp-idf/issues/10668#issuecomment-2574559862) to disable RTS. 

Does the ESP32 look for CTS with a destination address matching its own (true) MAC or the provided source in the buffer passed to esp_wifi_80211_tx?

I still have an issue where no packets are sent when I enable promiscuous and CSI modes. I will debug more and update.

zhangyanjiaoesp, 2025-01-20:
> Does the ESP32 look for CTS with a destination address matching its own (true) MAC or the provided source in the buffer passed to esp_wifi_80211_tx?

the provided source in the buffer passed to `esp_wifi_80211_tx`

roger-, 2025-02-02:
Thanks! 

Was able to get everything working.

closed 2025-02-02 by roger-