espressif/esp-idf #15198: bleprph_wifi_coex has outdated reference to BLE_GAP_EVENT_LINK_ESTAB

nskalsky, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.1

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Expected it to compile. I had to change BLE_GAP_EVENT_LINK_ESTAB to BLE_GAP_EVENT_CONNECT

### What is the actual behavior?

Does not compile

### Steps to reproduce.

1. Extract Source
2. Build

rahult-github, 2025-01-14:
Hi @nskalsky ,

The event BLE_GAP_EVENT_LINK_ESTAB  was introduced later on and is not part of release tag v5.3.1 .  May i know if you are switching branches / codebase and hence you have stale code with reference to the event left around ?

Also can please share your console log depicting what compilation failure are you observing on release tag v5.3.1 ?  Currently, no event has been removed , so there shouldn't be any compilation failure.

rahult-github, 2025-02-03:
Closing this issue. Feel free to reopen in case of any updates.

closed 2025-02-03 by rahult-github