espressif/esp-idf #15199: Installing esp-idf v5.4 installs the wrong Python..?

Nightsd01, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.4

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

After installing ESP-IDF v5.4 (via the `git clone` method mentioned in the [releases page](https://github.com/espressif/esp-idf/releases/tag/v5.4)), I am having a nasty issue where in my `~/.espressif` path, it _does_ install python after running `./install.sh` in the esp-idf directory....BUT it is the wrong python! It's installing `idf5.1_py3.11_env`. But this fails, because when I try to run `. ./export.sh`, it expects it to be `idf5.4_py3.11_env`?

### What is the actual behavior?

The primary symptom is that when I try to run `. ./export.sh`, it produces the following error:

```
ERROR: ESP-IDF Python virtual environment not found. Please run the install script to set it up before proceeding.
```

### Steps to reproduce.

1. On macOS, follow the installation steps via the git clone method on the [releases page](https://github.com/espressif/esp-idf/releases/tag/v5.4)
2. Try to run `. ./export.sh`
3. Get error
...

### Build or installation Logs.

```plain
➜  esp-idf-v5.4 git:(v5.4) ./install.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.3
"python3" has been detected
Checking Python compatibility
Installing ESP-IDF tools
Updating /Users/bradhesse/.espressif/idf-env.json
Selected targets are: esp32c6, esp32p4, esp32, esp32h2, esp32c61, esp32s2, esp32s3, esp32c3, esp32c2, esp32c5
Current system platform: macos-arm64
Installing tools: xtensa-esp-elf-gdb, riscv32-esp-elf-gdb, xtensa-esp-elf, riscv32-esp-elf, esp32ulp-elf, openocd-esp32, esp-rom-elfs
Installing xtensa-esp-elf-gdb@14.2_20240403
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz
Destination: /Users/bradhesse/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
Done
Extracting /Users/bradhesse/.espressif/dist/xtensa-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz to /Users/bradhesse/.espressif/tools/xtensa-esp-elf-gdb/14.2_20240403
Installing riscv32-esp-elf-gdb@14.2_20240403
Downloading https://github.com/espressif/binutils-gdb/releases/download/esp-gdb-v14.2_20240403/riscv32-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz
Destination: /Users/bradhesse/.espressif/dist/riscv32-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz.tmp
Done
Extracting /Users/bradhesse/.espressif/dist/riscv32-esp-elf-gdb-14.2_20240403-aarch64-apple-darwin21.1.tar.gz to /Users/bradhesse/.espressif/tools/riscv32-esp-elf-gdb/14.2_20240403
Installing xtensa-esp-elf@esp-14.2.0_20241119
Downloading https://github.com/espressif/crosstool-NG/releases/download/esp-14.2.0_20241119/xtensa-esp-elf-14.2.0_20241119-aarch64-apple-darwin.tar.xz
Destination: /Users/bradhesse/.espressif/dist/xtensa-esp-elf-14.2.0_20241119-aarch64-apple-darwin.tar.xz.tmp
Done
Extracting /Users/bradhesse/.espressif/dist/xtensa-esp-elf-14.2.0_20241119-aarch64-apple-darwin.tar.xz to /Users/bradhesse/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119
[... 162 lines cut ...]
Using cached pydantic_settings-2.7.1-py3-none-any.whl (29 kB)
Using cached ruamel.yaml-0.18.10-py3-none-any.whl (117 kB)
Using cached bitarray-3.0.0-cp311-cp311-macosx_11_0_arm64.whl (121 kB)
Using cached certifi-2024.12.14-py3-none-any.whl (164 kB)
Using cached charset_normalizer-3.4.1-cp311-cp311-macosx_10_9_universal2.whl (194 kB)
Using cached ruamel.yaml.clib-0.2.12-cp311-cp311-macosx_13_0_arm64.whl (132 kB)
Using cached six-1.17.0-py2.py3-none-any.whl (11 kB)
Installing collected packages: reedsolo, pyserial, pygdbmi, pyelftools, intelhex, bitarray, urllib3, typing-extensions, tqdm, six, ruamel.yaml.clib, pyyaml, python-dotenv, pyparsing, pygments, pycparser, pyclang, psutil, packaging, mdurl, jsonref, idna, freertos_gdb, esp-idf-panic-decoder, esp-idf-kconfig, construct, colorama, click, charset-normalizer, certifi, bitstring, argcomplete, annotated-types, ruamel.yaml, requests, pydantic-core, markdown-it-py, ecdsa, cffi, rich, requests-toolbelt, requests-file, pydantic, cryptography, pydantic-settings, esptool, esp-idf-size, esp-idf-nvs-partition-gen, idf-component-manager, esp-coredump, esp-idf-monitor
Successfully installed annotated-types-0.7.0 argcomplete-3.5.3 bitarray-3.0.0 bitstring-4.3.0 certifi-2024.12.14 cffi-1.17.1 charset-normalizer-3.4.1 click-8.1.8 colorama-0.4.6 construct-2.10.70 cryptography-43.0.3 ecdsa-0.19.0 esp-coredump-1.12.0 esp-idf-kconfig-2.4.1 esp-idf-monitor-1.5.0 esp-idf-nvs-partition-gen-0.1.6 esp-idf-panic-decoder-1.2.1 esp-idf-size-1.6.1 esptool-4.8.1 freertos_gdb-1.0.3 idf-component-manager-2.1.2 idna-3.10 intelhex-2.3.0 jsonref-1.1.0 markdown-it-py-3.0.0 mdurl-0.1.2 packaging-24.2 psutil-6.1.1 pyclang-0.6.0 pycparser-2.22 pydantic-2.10.5 pydantic-core-2.27.2 pydantic-settings-2.7.1 pyelftools-0.31 pygdbmi-0.11.0.0 pygments-2.19.1 pyparsing-3.2.1 pyserial-3.5 python-dotenv-1.0.1 pyyaml-6.0.2 reedsolo-1.7.0 requests-2.32.3 requests-file-2.1.0 requests-toolbelt-1.0.0 rich-13.9.4 ruamel.yaml-0.18.10 ruamel.yaml.clib-0.2.12 six-1.17.0 tqdm-4.67.1 typing-extensions-4.12.2 urllib3-1.26.20
All done! You can now run:

  . ./export.sh

➜  esp-idf-v5.4 git:(v5.4) . ./export.sh
Checking "python3" ...
Python 3.11.3
"python3" has been detected

ERROR: ESP-IDF Python virtual environment not found. Please run the install script to set it up before proceeding.
```
```

### More Information.

I've tried looking this same error up elsewhere but it's mostly people on windows, or people with seemingly different problems than the one I am facing.

Nightsd01, 2025-01-14:
Note: I found a fix by simply renaming the python package installed in `/.espressif/python_package` from `idf5.1_py3.11_env` to `idf5.4_py3.11_env`, assuming that not much has changed in this python package, but it still seems like a bad idea

riwalker, 2025-01-30:
why is this closed?  i have the exact issue today with 5.4 on win 11 "ESP-IDF Python virtual environment not found"

fhrbata, 2025-01-31:
Hello @riwalker ,

I believe we addressed this issue in https://github.com/espressif/esp-idf/commit/c0954cf0b280332ef4d0c1f17a68196d17da5e05. To prevent these kinds of problems, it's always best to start the installation from a clean environment where esp-idf isn't already activated, as suggested by the warning. 

Thank you.