espressif/esp-idf #15201: [USB] Korg B2 Piano URB Transfer IN never finishes

chipweinberger, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

custom pcb

### Power Supply used.

USB

### What is the expected behavior?

I am the creator of [Jamcorder](http://jamcorder.com). 

I have over 300 devices out in the wild, and the large majority work fine. They can be sent MIDI OUT (data sent to usb piano) no problem. However, a few devices fail to receive MIDI IN (receiving data from usb piano).

* Korg B2
* Kawai MP 7 

These devices work fine when plugged into a Windows or MacOS computer. So the problem is something related to the ESP32-S3.

Im calling `usb_host_transfer_submit`, but it just never calls the callback. 

### What is the actual behavior?

Im calling `usb_host_transfer_submit`, but it just never calls the callback. 

### Steps to reproduce.

1. claim interface
2. call `usb_host_transfer_submit` on USB IN endpoint

### Debug Logs.

```plain
//
// These logs are not that useful
//

 [0;32mI (43627) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (43755) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (43879) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (44003) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (44127) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (44251) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (44375) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
[... 775 lines cut ...]
 [0;32mI (57611) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (57739) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (57863) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (57987) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58111) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58235) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58359) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58483) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58607) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58735) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58859) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (58983) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (59107) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (59231) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (59355) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (60727) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (60855) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (60979) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61103) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61227) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61351) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61475) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61599) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
 [0;32mI (61727) pd usbh: [client task] usb_host_client_handle_events(client:0x3c3b5da0) [0m
```

chipweinberger, 2025-01-14:
The diff which broke Korg B2 support was this change.

This change was made to support a OYAYO Folding piano, which did not like having its configuration descriptor downloaded multiple times. Again, all these pianos work fine with Windows & MacOS, and they all have a single Configuration Descriptor.

**Before:**
```
// submits control transfer
xesp_usb_config_descriptor_t* pd_usbh_device_getConfigDescriptor(pd_usbh_device_t usbhDevice, uint8_t configIdx, pd_error_t *error)
{
    pd_err_unset(error);

    xesp_usb_config_descriptor_t* parsed = NULL;

    // allocations
    pd_usb_transfer_t* transfer = pd_usbh_service_takeTransfer();
    {
        ESP_LOGI(TAG, "getConfigDescriptor(idx: %" PRIu8 ")", configIdx);

        USB_SETUP_PACKET_INIT_GET_CONFIG_DESC((usb_setup_packet_t *)transfer->val->data_buffer,
                                                configIdx,
                                                CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE);

        transfer->val->num_bytes = CONFIG_USB_HOST_CONTROL_TRANSFER_MAX_SIZE + sizeof(usb_setup_packet_t);
        transfer->val->flags = 0;
        transfer->val->device_handle = usbhDevice.deviceHandleB;
        transfer->val->bEndpointAddress = 0;
        transfer->val->status = USB_TRANSFER_STATUS_ERROR;
        transfer->val->timeout_ms = 500;
        transfer->val->context = 0;
        
        eTry(,pd_usbh_service_submitTransferAndWait,(transfer, error));

        uint32_t length = transfer->val->actual_num_bytes - sizeof(usb_setup_packet_t);
        uint8_t* configDesc = transfer->val->data_buffer + sizeof(usb_setup_packet_t);

        // parse
        parsed = xesp_usbh_parse_config(configDesc, length, error);
        if (!parsed) {
            eFail("failed to parse config descriptor");
            goto end;
        }

        pd_err_success(error);
    }
end:
    pd_usbh_service_giveTransfer(transfer);
    eOnFail(return NULL);
    return parsed;
}
```

**After:**
```
// submits control transfer
xesp_usb_config_descriptor_t* pd_usbh_device_getConfigDescriptor(pd_usbh_device_t usbhDevice, uint8_t configIdx, pd_error_t *error)
{
    pd_err_unset(error);

    // allocations
    xesp_usb_config_descriptor_t* parsed = NULL;
    {
        ESP_LOGI(TAG, "getConfigDescriptor(idx: %" PRIu8 ")", configIdx);

        // read
        const usb_config_desc_t* desc;
        sTry(esp_err_t err =,usb_host_get_active_config_descriptor,(usbhDevice.deviceHandleB, &desc));

        // parse
        eTry(parsed =,xesp_usbh_parse_config,((uint8_t*) desc, sizeof(usb_config_desc_t), error));

        pd_err_success(error);
    }
end:
    eOnFail(return NULL);
    return parsed;
}
```

roma-jam, 2025-01-14:
Hi @chipweinberger, 

> Im calling usb_host_transfer_submit, but it just never calls the callback. 

When you schedule the IN transfer, the callback should be called after the data is received by the hardware. Does that mean that literally, you pressed any piano keys and no data appears on the usb? And the same logic works without problem (you can get the callback) on other MIDI-devices, am I right? 

Is there any specific class-related configuration should be done with some class requests, to setup the device regarding sending back the data? 

Maybe, Wireshark log from attaching device to Win could help (especially, class specific requests sequence during initialization of the MIDI class driver on the Host side).

chipweinberger, 2025-01-14:
> Does the USB MIDI Device work on Linux?

I'll check.

> Can you please enable debug logs in menuconfig Component Config -> Log -> Log Level -> Default log verbosity, run [usb_host_lib example](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/usb_host_lib) with your MIDI Device connected to the esp32s3 and share the output log? This way we can see if the device was enumerated correctly by the USB Host.

Will do.

> When you schedule the IN transfer, the callback should be called after the data is received by the hardware. Does that mean that literally, you pressed any piano keys and no data appears on the usb? [...]

Correct.

> Maybe, Wireshark log from attaching device to Win could help (especially, class specific requests sequence during initialization of the MIDI class driver on the Host side).

Sure, I can provide that.

> UPD: Could you also specify what does mean OYAYO Folding piano, which did not like having its configuration descriptor downloaded multiple times?

I'm not sure exactly, but the diff I posted fixed it. The descriptors were grabbed (a) automatically by esp-idf & (b) again by my code, and (b) was not working. I expected maybe their firmware was bugged, so I played with it.

> Apparently, it is worth it to check the Wireshark log, specifically SetConfguration() request.
During the USB Host Library it is a part of enumeration process by default (with possibility to change the index via enum_filter_cb() ), so set Verbosity level is enough.

Ah I did not realize this. My code also calls SetConfguration. So it's getting called twice.

chipweinberger, 2025-01-14:
So I've found the actual issue! 🎉🎉🎉

In my latest firmware update, I added additional print statement *during* usb parsing -- ironically to aide in debugging USB issues. I use custom parsing code, since my code was written before esp-idf's.

```
xesp_usb_config_descriptor_t* xesp_usbh_parse_config(uint8_t *data, uint32_t length, pd_error_t* error) {
     ...
    usb_util_print_intf(intf); // causes an issue
    ...
}

xesp_usb_intf_descriptor_t* xesp_usbh_parse_intf(uint8_t *data, uint32_t length, pd_error_t* error) {
     ...
    usb_util_print_intf(intf); // causes an issue
    ...
}

xesp_usb_ep_descriptor_t* xesp_usbh_parse_ep(uint8_t *data, uint32_t length, pd_error_t* error) {
     ...
    usb_util_print_ep(intf); // causes an issue
    ...
}
```

For whatever reason, perhaps timing related, a few USB devices do not like this. It's a bit surprising, but also extremely reproducible.

I will close this issue. 

@peter-marcisovsky @roma-jam, I'm you'd like to understand the underlying mechanism of failure more. Unfortunately, I don't have time or a need to dig deeper right now. But the offending device is a Korg B2. [This one.](https://www.amazon.com/gp/product/B07VV6GDVN/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1)

chipweinberger, 2025-01-15:
So I spoke too soon, this issue is back on a more 'production' device with Bluetooth and Wifi enabled. 

I'm not sure if it's related, but since this issue seems like a timing issue, it seems like anything could affect timing.

chipweinberger, 2025-01-15:
In summary, Im not sure if there is anything in esp-idf that needs to change.

roma-jam, 2025-01-15:
Hi @chipweinberger 

USB_HOST_CLIENT_EVENT_NEW_DEV and USB_HOST_CLIENT_EVENT_DEV_GONE are called within `usbh_process()`, which is in its own turn is inside the `usb_host_lib_handle_events()` call.  

Calling any other things such as `pd_alerts_push(PD_ALERT_USB_STATE);` means literally that you call that logic within the task, which handles the USB (where you call `usb_host_lib_handle_events()`). 

If there will be anything, that requires time, the USB will be waiting for these logic to end to proceed handle the USB events. 

So, if you don't want any other tasks affect USB, you should keep them out of the task, which handles USB.

chipweinberger, 2025-01-15:
Yes, it makes sense. 

However, the `usb_util_print_cfg` call I had before was not in the `usbh_process()`. It was just in my `PD_PRIORITY_MIDI_USB_OPEN_TASK`, which is responsible for opening the device, parsing the configuration, and calling SET_CONFIGURATION.

Notably, I also have `PD_PRIORITY_USB_CLIENT_EVENTS_TASK` >  `PD_PRIORITY_USB_HOST_TASK`. Is this expected? 

```
#define PD_PRIORITY_USB_CLIENT_EVENTS_TASK 21
#define PD_PRIORITY_USB_HOST_TASK 20
#define PD_PRIORITY_MIDI_USB_OPEN_TASK 10
```

roma-jam, 2025-01-31:
Hi @chipweinberger ,

Sorry for the delay in reply. 

> Notably, I also have PD_PRIORITY_USB_CLIENT_EVENTS_TASK > PD_PRIORITY_USB_HOST_TASK. Is this expected?

Yes, this is the right approach. 

Is there any other questions regarding calling the SetConfgiuration() or we can close this issue? 
Thanks.

closed 2025-01-31 by chipweinberger