espressif/esp-idf #15203: IP Information Synchronization

gl-chenxiaosheng, 2025-01-14:
### Is your feature request related to a problem?

No.

### Describe the solution you'd like.

When my esp32 device connects to the up-stream network via ethernet, the dhcp is assigned to the IP of 192.168.26.156. When the router IP address of my up-stream network is changed to 192.168.16.1, there is no relevant event to notify me that the IP changed. IP information can only be synchronized when the device is rebooted.
I would like to have a way to synchronize the IP change information while the device is running.

### Describe alternatives you've considered.

Is it possible to provide an IP change event out here?

espressif-abhikroy, 2025-01-21:
@gl-chenxiaosheng 
In your scenario, the DHCP client will automatically re-synchronize its configuration upon the expiration of the T2 timer (Rebinding Time). However, this mechanism relies on the DHCP lease lifecycle and may not detect upstream network changes immediately. If you need real-time detection of network configuration changes (e.g., a router IP change), you can consider the following options:

1. Restarting the DHCP Client: Restarting the DHCP client forces it to request a new lease from the DHCP server. You can achieve this with the following calls:
``` c
esp_netif_dhcpc_stop(netif_sta);
esp_netif_dhcpc_start(netif_sta);
```
This is useful when you detect significant network changes but note that it doesn't automatically trigger based on network conditions.

2. ARP Monitoring: Periodically monitor the gateway's IP address using ARP to detect changes. You can use the 
```c
etharp_find_addr()
``` 
function to retrieve the gateway's MAC address. If the address changes, it may indicate an upstream network reconfiguration.

3. Ping-Based Monitoring: Regularly ping the gateway IP address to ensure it is reachable and verify that the network configuration is still valid. A failure to reach the gateway may indicate a network change requiring a re-synchronization.

Limitations: There is no built-in, straightforward mechanism in DHCP for instant notification of configuration changes. DHCP clients operate on a lease-based model, so proactive detection of changes often requires additional monitoring strategies like ARP or ping.

espressif-abhikroy, 2025-07-04:
Closing this issue as the explanation has been provided and there haven't been any further queries.
Feel free to reopen or comment if you have any additional questions.