espressif/esp-idf #15204: nvs_* getting slower updage from 5.2.1 to 5.3 or 5.4

GuGu927, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi. I'm using NVS to save device data, and have speed problem when update to latest verion of IDF.

When IDF version 5.2.1, nvs_* works very fast.

nvs_set_*  over 10 times each create
I (71606) device: Child created
I (71656) device: Child created
I (71746) device: Child created
I (71796) device: Child created

nvs_erase_*  over 10 times each delete
I (37796) device: Child deleted
I (37816) device: Child deleted
I (37826) device: Child deleted
I (37846) device: Child deleted

When IDF version to 5.3.2 and 5.4, nvs_* works very slow

nvs_set*  over 10 times each create
I (192097) device: Child created 
I (192507) device: Child created 

nvs_erase_*  over 10 times each delete
I (101024) device: Child deleted
I (101864) device: Child deleted
I (102704) device: Child deleted
I (103554) device: Child deleted
I (104404) device: Child deleted

Enable/disable NVS_ALLOCATE_CACHE_IN_SPIRAM  not affect.

Everythings is same but not IDF version.

Can you help me with this?
Thanks.

rrtandler, 2025-01-15:
Hi @GuGu927 ,
Could you please provide the code snippet showing how is the NVS called and how are you measuring the time elapsed in the operation. And, if possible, also the actual number of entries already stored in NVS during the test ?

GuGu927, 2025-01-15:
Hi @rrtandler  ,
I'm using ESP32S3 N8R8.
Here's code flow of mine.

`esp_err_t set_storage_bool(nvs_handle_t *handle, const char *key, bool value)
{
    esp_err_t err = nvs_set_u8(*handle, key, (uint8_t)value);
    if (err != ESP_OK)
    {
    }
    else
        my_commit(handle);
    return err;
}

esp_err_t get_storage_bool(nvs_handle_t *handle, const char *key, bool *value)
{
    uint8_t nvs_value;
    esp_err_t err = nvs_get_u8(*handle, key, &nvs_value);
    if (err != ESP_OK)
    {
        return err;
    }
    *value = (bool)nvs_value;
    return err;
}

void my_commit(nvs_handle_t *handle)
{
// Loop the handle list, and if it exists, try `vTaskDelete`. After that, call `xTaskCreate`. 
// Commit the task after some seconds. This prevents `nvs_commit` from happening multiple times at the same time.

}`

`
esp_err_t save_device()
{
  set_some_nvs_blob()
  my_commit()
}

my_device_t *create_device()
{
  realloc device list
  get_some_nvs_value() != ESP_OK then set_some_nvs_value()
  get_some_nvs_value() != ESP_OK then set_some_nvs_value()
  get_some_nvs_value() != ESP_OK then set_some_nvs_value()
  ...
  printf("Device created");
  save_device(my_device);
  return my_device;
}
`

1. Data comes in.
2. Check for loop and try to create my virtual device.
3. Reallocate memory for the device list.
4. Attempt `nvs_get_*` for many (5–20) properties of the device being added. If a value is missing, set it using `nvs_set_*`.
   (When first booting, devices are loaded from NVS, so the flow always tries both `nvs_get_*` and `nvs_set_*` simultaneously.)
5. At the very end, use `printf` to output "device created," then save the device.
6. Check data.

After every `nvs_set`, a `my_commit` is attempted.  
In IDF version 5.2.1, the interval shown by `printf` in the console time is in the tens of milliseconds.  
In IDF version 5.3.X and 5.4, the interval shown by `printf` in the console time is in the hundreds of milliseconds.

If you need any additional information, please don't hesitate to ask. I can provide detailed reproduction steps or logs if necessary. Thank you for reviewing this issue, and I look forward to your feedback.

rrtandler, 2025-01-23:
Hi @GuGu927,
Between the IDF versions 5.2.1 and 5.3 the internal code of NVS library was modified to allow searching for metadata using incomplete key information. For some special search cases, the hash map (which speeds up the internal metadata search for regular nvs_get / nvs_set operation) usage had to be omitted. For time being, this change slows down the NVS initialisation (as it searches for orphan metadata records without knowing the complete key information needed for hash map) . 
Does your code call nvs_init (or it's variants) with each attempt to emit the message like:
I (71796) device: Child created
respectively
I (37796) device: Child deleted

OR nvs_init is called once, before the sequence of create / delete calls ?

GuGu927, 2025-01-23:
Hi @rrtandler
First of all, thank you for your response.

In my code, nvs_flash_secure_init_partition is called during the initial setup.
After that, I use nvs_open_from_partition to obtain an nvs_handle.

Then, I call nvs_set and nvs_get, and then call nvs_commit.
Thus, nvs_open_from_partition is only called during the initialization phase.

To prevent multiple nvs_commit calls in a short period, I use a task that ensures nvs_commit is executed once every ~2 seconds.(Task created after nvs_set)

So, to answer your question:
Yes, my code follows the behavior you described—"nvs_init is called once, before the sequence of create/delete calls."

Thanks for your help! Let me know if you need any more details.

rrtandler, 2025-03-24:
Hi @GuGu927 ,
Thanks for clarification of the progam flow.
If you do not mix different data types under the same namespace/key combinations, you may try to set kconfig option NVS_LEGACY_DUP_KEYS_COMPATIBILITY. This skips some of the (slow) extra checks performed during nvs_set_* operations.

To do so, in your project run` idf.py menuconfig` then press / and type the NVS_LEGACY_DUP_KEYS_COMPATIBILITY and set is to checked. Save the configuration, bulid and flash the application.

KaeLL, 2025-03-25:
That's basically knowingly introducing an opt-out pessimization. If old behavior is to be broken, that it's done so for the better, not worse. How about adding a note about this on the help section of the Kconfig option?

rrtandler, 2025-03-25:
Hi @KaeLL,

The primary intent of the change was fixing the issue leaving old value for the namespace-key pair in the NVS if it's data type was different than the data type used in actual nvs_set_* operation. The performance penalty is sadly the side effect., not the intent. 
The kconfig option was introduced for applications relying on the old behaviour (i.e. it was possible to keep the not deleted value in the NVS under the same namespace-key combination if it's data type was different and retrieve it by deleting the actual one - aka one level deep stack). Omitting the correct (and unfortunately costly) lookup of the old value was beneficial for performance.

AxelLin, 2025-03-26:
> Hi [@KaeLL](https://github.com/KaeLL),
> 
>  Omitting the correct (and unfortunately costly) lookup of the old value was beneficial for performance.

@rrtandler 

What's the big O of the lookup algorithm?
If it's linear time O(N), this needs improvement.

rrtandler, 2025-03-26:
@AxelLin ,
The original lookup for the key was assuming the potentially existing namespace-key will have the same data type as the one to be written as a new value. In that case, the hash map in RAM was used to lookup the record. All maps of NVS pages pages were visited until either the end of page list was reached or match happened. So O(page_count)*log2(126)).
The actual version in master has O(page_count)(126) as the search using ANY data type cannot use the hash table.

We have a bug fix in the pipeline dealing with the imperfect delete of existing BLOBs. As a part of small refactor, it also improves the O following way:
- if backward compatibility kconfig symbol is set the potential existing record is searched the original way - no performance impact
- if compatibility symbol is not set, the implementation first looks up the potentially existing value using the same data type of newly written value. If record is found then there is no impact to the performance.
- If record is not found, the search is performed using the ANY data type - complexity is increased to O(N).

Applications rewriting the existing keys with values of the same data types will (by default) see the performance impact only during the write for namespace-key combinations not used yet. All subsequent updates will find the previous values quickly. If application developer doesn't plan to mix data types under the same namespace-key combinations, the compatibility define can be set and then NVS doesn't guarantee deleting of old values (if data types differ) but no performance drop should be observed even during the initial use of namespace-key pair.

GuGu927, 2025-03-26:
@rrtandler Thank you for your help! 👍

GuGu927, 2025-05-11:
@rrtandler Hi, today I tried this "NVS_LEGACY_DUP_KEYS_COMPATIBILITY" checked.
But, problem not solved.
ESP-IDF version is 5.4.1
So I rollback to 5.2.1.

rrtandler, 2025-05-12:
Hi @GuGu927 ,

Indeed, I came to the same conclusion after performing a validation of the recent code update in the NVS. I have observed non linear rise of flash read accesses when performing write operations. I have found a solution and am about to merge into the master code during this week.