espressif/esp-idf #15207: FAILED: esp-idf/mbedtls/x509_crt_bundle

colsceg, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.5_py3.1

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

idf.py build -> should end wit build Ok or something like this

### What is the actual behavior?

[445/495] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle /home/ncb/esp/esp-idf/blink/build/esp-idf/mbedtls/x509_crt_bundle 
cd /home/ncb/esp/esp-idf/blink/build/esp-idf/mbedtls && /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
Illegal instruction
ninja: build stopped: subcommand failed.

### Steps to reproduce.

I have made new installation of esp-idf on Linux Debian 12. (esp-idf v5.4)
I have copied from esp-idf/examples the blink directory
I started idf.py build in the blink directory
and got the error illegal instruction

Have no idea what I did wrong

...

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running cmake in directory /home/ncb/esp/sample_project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/ncb/esp/sample_project"...
-- IDF_TARGET is not set, guessed 'esp32s2' from sdkconfig '/home/ncb/esp/sample_project/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- Minimal build - OFF
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s2-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s2-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s2-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- git rev-parse returned 'Schwerwiegend: Kein Git-Repository (oder irgendeines der Elternverzeichnisse): .git'
-- Could not use 'git describe' to determine PROJECT_VER.
[... 72 lines cut ...]
  is not a full path to an existing compiler tool.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.
Call Stack (most recent call first):
  CMakeLists.txt:72 (project)

-- Configuring incomplete, errors occurred!
See also "/home/ncb/esp/sample_project/build/bootloader/CMakeFiles/CMakeOutput.log".
[160/1003] Building C object esp-idf/bootloader_...ootloader_support.dir/src/bootloader_efuse.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ncb/esp/sample_project/build/log/idf_py_stderr_output_8169 and /home/ncb/esp/sample_project/build/log/idf_py_stdout_output_8169

Bootloader binary size 0x6600 bytes. 0xa00 bytes (9%) free.
[445/495] Generating x509_crt_bundle

FAILED: esp-idf/mbedtls/x509_crt_bundle

 /home/ncb/esp/esp-idf/blink/build/esp-idf/mbedtls/x509_crt_bundle 
cd /home/ncb/esp/esp-idf/blink/build/esp-idf/mbedtls && /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
Illegal instruction
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ncb/esp/esp-idf/blink/build/log/idf_py_stderr_output_10024 and /home/ncb/esp/esp-idf/blink/build/log/idf_py_stdout_output_10024
```

dobairoland, 2025-01-15:
Hi @colsceg. Can you please describe how to install ESP-IDF? Did you follow https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/linux-macos-setup.html or some other way?

If you used the above guide then have you also sourced `export.sh` as explained in the guide?

colsceg, 2025-01-15:
yes, i just used the commands from this site

git clone --recursive https://github.com/espressif/esp-idf.git
./install.sh all
. ./export.sh

and then in the blink directory 
idf.py build

I think otherwise it wouldn't have completed the the esp32 image without errors.
I am wondering what is the 'Generating x509_crt_bundle' for? 
Why do we need a Base64-coded certificate?

If I do the same on a mac it works fine and I can not see, it tries to create a x509 certificate

dobairoland, 2025-01-15:
I'm a little bit confused by your logs because you are talking about the blink example but I see a mixture of `/home/ncb/esp/sample_project` and `/home/ncb/esp/esp-idf/blink/` directories in the same log. Moreover, the sample_project has been modified for sure because the target was changed to ESP32-S2 from the default ESP32. 

Please try the real blink example. The path `/home/ncb/esp/esp-idf/blink/` doesn't seem right. The original, unmodified project should be in the `examples/get-started/blink` sub-directory of ESP-IDF. If you did modify it some way or built it on a different computer and moved to your Linux machine then these all can explain similar issues. Are you able to build the unmodified blink example?

By the way, if you move around the project then it is best to run `idf.py fullclean` before attempting to build again.

Are you running `. export.sh` in a clean terminal or where some other version of ESP-IDF is present? You can check this by running `printenv` before executing `. export.sh`. Are there anything ESP-IDF related? By the way, are you sure you are not forgetting about the dot and space before `export.sh`?

Can you examine the content of your PATH environment variable after  `. export.sh`? ESP-IDF paths should be there. For example `/home/ncb/esp/esp-idf/tools/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin` - because according to the logs, this is one place from where it tries to run the toolchain. This path, by the way, also seem to be wrong. Did you specify ESP-IDF-related environment variables before running the install script? It seems that the tools installation path is directed into the ESP-IDF tools directory. This is wrong for many reasons.

All in all, I think if you unset all ESP-IDF variables you did set before and re-run the install and export scripts then everything will be alright. If you set them with some purpose then I'd suggest to tell what you wanted to achieve or see the related documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/tools/idf-tools.html. Please be advised that these are for advanced users and use-cases and you don't need them to set-up a working ESP-IDF environment. I believe your issue are caused because they were set incorrectly prior running the install and export scripts.

colsceg, 2025-01-15:
sorry about mixing several tests but at least all ends in the same error. 
I tried setting up from the beginning several times.

the printenv in a new Terminal window shows no ESP-IDF variables at all, and no I did not forget the .  followed by ./export.sh

here are all the variables after   .  ./export.sh
IDF_PYTHON_ENV_PATH=/home/ncb/.espressif/python_env/idf5.5_py3.11_env
IDF_PATH=/home/ncb/esp/esp-idf
ESP_IDF_VERSION=5.5
IDF_DEACTIVATE_FILE_PATH=/tmp/tmpfw57bjadidf_4282

PATH=/home/ncb/esp/esp-idf/components/espcoredump:
/home/ncb/esp/esp-idf/components/partition_table:
/home/ncb/esp/esp-idf/components/app_update:
/home/ncb/.espressif/tools/xtensa-esp-elf-gdb/15.2_20241112/xtensa-esp-elf-gdb/bin:
/home/ncb/.espressif/tools/riscv32-esp-elf-gdb/15.2_20241112/riscv32-esp-elf-gdb/bin:
/home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin:
/home/ncb/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin:
/home/ncb/.espressif/tools/esp32ulp-elf/2.38_20240113/esp32ulp-elf/bin:
/home/ncb/.espressif/tools/openocd-esp32/v0.12.0-esp32-20241016/openocd-esp32/bin:
/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin:
/home/ncb/esp/esp-idf/tools:
/usr/local/bin:/usr/bin:/bin:/usr/local/games:/usr/games:/sbin:/usr/sbin
IDF_TOOLS_EXPORT_CMD=/home/ncb/esp/esp-idf/export.sh
IDF_TOOLS_INSTALL_CMD=/home/ncb/esp/esp-idf/install.sh
ESP_ROM_ELF_DIR=/home/ncb/.espressif/tools/esp-rom-elfs/20241011/

seems to be the same as I get on the MAC

colsceg, 2025-01-16:
OK, this is the output, I think it is the same I get every time I tried:

ncb@mx:~/esp/esp-idf/examples/blink
$ idf.py fullclean

Executing action: fullclean
Build directory '/home/ncb/esp/esp-idf/examples/blink/build' not found. Nothing to clean.
Executing action: remove_managed_components
Done
ncb@mx:~/esp/esp-idf/examples/blink

$ idf.py build

Executing action: all (aliases: build)
Running cmake in directory /home/ncb/esp/esp-idf/examples/blink/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/ncb/esp/esp-idf/examples/blink"...
-- IDF_TARGET not set, using default target: esp32
-- Found Git: /usr/bin/git (found version "2.39.5") 
-- Minimal build - ON
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/ncb/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
NOTICE: Dependencies lock doesn't exist, solving dependencies.
...NOTICE: Updating lock file at /home/ncb/esp/esp-idf/examples/blink/dependencies.lock
NOTICE: Processing 2 dependencies:
NOTICE: [1/2] espressif/led_strip (2.5.5)
NOTICE: [2/2] idf (5.5.0)
-- Project sdkconfig file /home/ncb/esp/esp-idf/examples/blink/sdkconfig
Loading defaults file /home/ncb/esp/esp-idf/examples/blink/sdkconfig.defaults...
Loading defaults file /home/ncb/esp/esp-idf/examples/blink/sdkconfig.defaults.esp32...
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python (found version "3.11.2") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "blink" version: v5.5-dev-1346-gcc9fb5bd5e-dirty
-- Adding linker script /home/ncb/esp/esp-idf/examples/blink/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/ncb/esp/esp-idf/examples/blink/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libc-funcs.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-reent-funcs.ld
-- Adding linker script /home/ncb/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Components: app_update bootloader bootloader_support cxx efuse esp_app_format esp_bootloader_format esp_common esp_driver_gpio esp_driver_rmt esp_driver_spi esp_hw_support esp_mm esp_partition esp_pm esp_ringbuf esp_rom esp_security esp_system esp_timer espressif__led_strip esptool_py freertos hal heap log main mbedtls newlib partition_table pthread soc spi_flash xtensa
-- Component paths: /home/ncb/esp/esp-idf/components/app_update /home/ncb/esp/esp-idf/components/bootloader /home/ncb/esp/esp-idf/components/bootloader_support /home/ncb/esp/esp-idf/components/cxx /home/ncb/esp/esp-idf/components/efuse /home/ncb/esp/esp-idf/components/esp_app_format /home/ncb/esp/esp-idf/components/esp_bootloader_format /home/ncb/esp/esp-idf/components/esp_common /home/ncb/esp/esp-idf/components/esp_driver_gpio /home/ncb/esp/esp-idf/components/esp_driver_rmt /home/ncb/esp/esp-idf/components/esp_driver_spi /home/ncb/esp/esp-idf/components/esp_hw_support /home/ncb/esp/esp-idf/components/esp_mm /home/ncb/esp/esp-idf/components/esp_partition /home/ncb/esp/esp-idf/components/esp_pm /home/ncb/esp/esp-idf/components/esp_ringbuf /home/ncb/esp/esp-idf/components/esp_rom /home/ncb/esp/esp-idf/components/esp_security /home/ncb/esp/esp-idf/components/esp_system /home/ncb/esp/esp-idf/components/esp_timer /home/ncb/esp/esp-idf/examples/blink/managed_components/espressif__led_strip /home/ncb/esp/esp-idf/components/esptool_py /home/ncb/esp/esp-idf/components/freertos /home/ncb/esp/esp-idf/components/hal /home/ncb/esp/esp-idf/components/heap /home/ncb/esp/esp-idf/components/log /home/ncb/esp/esp-idf/examples/blink/main /home/ncb/esp/esp-idf/components/mbedtls /home/ncb/esp/esp-idf/components/newlib /home/ncb/esp/esp-idf/components/partition_table /home/ncb/esp/esp-idf/components/pthread /home/ncb/esp/esp-idf/components/soc /home/ncb/esp/esp-idf/components/spi_flash /home/ncb/esp [... cut ...]
cd /home/ncb/esp/esp-idf/examples/blink/build/esp-idf/mbedtls && /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
Illegal instruction
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/ncb/esp/esp-idf/examples/blink/build/log/idf_py_stderr_output_3775 and /home/ncb/esp/esp-idf/examples/blink/build/log/idf_py_stdout_output_3775

dobairoland, 2025-01-16:
Now it looks better. The toolchain is found. Previously it was missing. 

What do you get if you run the following command manually:
```sh
python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
```

colsceg, 2025-01-17:
this is what I get:
$ python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
Ungültiger Maschinenbefehl

dobairoland, 2025-01-17:
Thank you for helping debug this. Could you please try to run the following two commands and share the output?

```sh
/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python --version
```

```sh
/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -m pdb /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
```

I hope this second one will reveal more information because I have no idea what is causing that `Illegal instruction`. Probably it won't exit and will go into command line debugger. You can exit by `q` + hitting <ENTER>.

Thank you again!

colsceg, 2025-01-17:
here are the results:

ncb@mx:~/esp/esp-idf/examples/blink
$ /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python --version
Python 3.11.2
ncb@mx:~/esp/esp-idf/examples/blink
ncb@mx:~/esp/esp-idf/examples/blink
$ /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -m pdb /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
> /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py(13)<module>()
-> import argparse
(Pdb) continue
Invalid --input=['/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem', '/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem'], is neither file nor folder
The program exited via sys.exit(). Exit status: 2
> /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py(13)<module>()
-> import argparse
(Pdb) continue
Illegal instruction
ncb@mx:~/esp/esp-idf/examples/blink

 I’m not so familiar with Python, but the file  cacrt_local.pem  does not exist in this directory

dobairoland, 2025-01-18:
Does the following also produces the `Illegal instruction` error?
```sh
/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -c "import argparse"
```

My guess now is that environment is broken. Could you try to remove `/home/ncb/.espressif/python_env/idf5.5_py3.11_env` and then re-run the install and export scripts?

colsceg, 2025-01-18:
no, no _illegal instruction_ with this command:

ncb@mx:~/esp/esp-idf/examples/blink
_$ /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python -c "import argparse"_
ncb@mx:~/esp/esp-idf/examples/blink

then 
_rm -r -f /home/ncb/.espressif/python_env/idf5.5_py3.11_env_

then the complete new installation

mkdir -p ~/esp
cd ~/esp
git clone --recursive https://github.com/espressif/esp-idf.git

cd ~/esp/esp-idf
./install.sh esp32,esp32s2

. ./export.sh

cd ~/esp/esp-idf/examples/get-started/blink
idf.py build

ending with the well known result

_....Successfully created esp32 image.
Generated /home/ncb/esp/esp-idf/examples/get-started/blink/build/bootloader/bootloader.bin
[117/117] cd /home/ncb/esp/esp-idf/examples/get-...et-started/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x6600 bytes. 0xa00 bytes (9%) free.
[450/500] Generating x509_crt_bundle
FAILED: esp-idf/mbedtls/x509_crt_bundle /home/ncb/esp/esp-idf/examples/get-started/blink/build/esp-idf/mbedtls/x509_crt_bundle 
cd /home/ncb/esp/esp-idf/examples/get-started/blink/build/esp-idf/mbedtls && /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
Illegal instruction
ninja: build stopped: subcommand failed.
ninja failed with exit code 1_

dobairoland, 2025-01-19:
> after deleting certicicates in cacrt_all.pem down to 16 certificates the idf.py build runs without any errors.

Can you tell us which one have you deleted?

So now it completes without `Illegal instruction`?

I will ask colleagues who are more familiar with `gen_crt_bundle.py` to take a look it as well.

AdityaHPatwardhan, 2025-01-20:
Hi @colsceg,
It is strange that the cert bundle is failing to build.

Could you try following 
`cd $IDF_PATH/components/mbedtls/esp_crt_bundle`
`./gen_crt_bundle.py --input cacert_all.pem cacrt_local.pem --max-certs 200`

It's the same command that is failing, it's just we are executing the command directly rather than in the build process. ( Before executing the command, please revert any changes that you may have done in order to fix the build failure)

colsceg, 2025-01-20:
I deleted the certifcates just from the end of the file up to the top so that at least there are only 16 certificates in the cacrt_all.pem file. And then the build runs to the end.

_Successfully created esp32 image.
Generated /home/ncb/esp/esp-idf/examples/get-started/blink/build/blink.bin
[54/54] cd /home/ncb/esp/esp-idf/examples/get-st...p-idf/examples/get-started/blink/build/blink.bin
blink.bin binary size 0x27440 bytes. Smallest app partition is 0x100000 bytes. 0xd8bc0 bytes (85%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blink.bin
or from the "/home/ncb/esp/esp-idf/examples/get-started/blink/build" directory
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"_

while running this command I get the following error:
cd $IDF_PATH/components/mbedtls/esp_crt_bundle
./gen_crt_bundle.py --input cacert_all.pem cacrt_local.pem --max-certs 200

_ncb@mx:~/esp/esp-idf/examples/get-started/blink
$ cd $IDF_PATH/components/mbedtls/esp_crt_bundle
ncb@mx:~/esp/esp-idf/components/mbedtls/esp_crt_bundle
$ ./gen_crt_bundle.py --input cacert_all.pem cacrt_local.pem --max-certs 200
Traceback (most recent call last):
  File "/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/./gen_crt_bundle.py", line 229, in <module>
    main()
  File "/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/./gen_crt_bundle.py", line 217, in main
    bundle.add_from_path(path)
  File "/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/./gen_crt_bundle.py", line 59, in add_from_path
    for file_path in os.listdir(crts_path):
                     ^^^^^^^^^^^^^^^^^^^^^
NotADirectoryError: [Errno 20] Not a directory: '/home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem'_

but the idf.build  runs again  without errors.

colsceg, 2025-01-21:
After searching for at least two hours, I found, that if I put a space at the end of _-----END CERTIFICATE-----  _  of each entry in the _cacrt_all.pem_ file this command  
_/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q --max-certs 200_ 
works without giving an _illegal instruction_ and even a build on the blink  or hello_world directory gives no error any more. 
Maybe one of your Python experts can find out what happens.

AdityaHPatwardhan, 2025-01-21:
Hi @colsceg 

Could you please share following 3 files from your local copy of esp-idf 

`cacrt_all.pem`
`gen_crt_bundle.py`
`cacert_local.pem`

Maybe I would know more if I can reproduce the issue on my side.

colsceg, 2025-01-21:
as .pem files are not allowed, I have packed them.
[esp_crt_bundle.tar.gz](https://github.com/user-attachments/files/18491942/esp_crt_bundle.tar.gz)

AdityaHPatwardhan, 2025-01-21:
@colsceg I tried the same command 
``` ./gen_crt_bundle.py --input cacrt_all.pem cacrt_local.pem --max-certs 200``` on the files that you shared
and I got successful result. Note: I was using the ESP-IDF environment.

Could you setup esp-idf environment in your terminal 
```
$ cd $IDF_PATH
$ ./install.sh
$ . ./export.sh
$ cd /* location_of_your_esp_crt_bundle */
$ ./gen_crt_bundle.py --input cacrt_all.pem cacrt_local.pem --max-certs 200
```

colsceg, 2025-01-21:
I did as you said  and here is the result with the original cacrt_all.pem
_ncb@mx:home/esp/esp-idf/components/mbedtls/esp_crt_bundle
$ ./gen_crt_bundle.py --input cacrt_all.pem cacrt_local.pem --max-certs 200
gen_crt_bundle.py: Parsing certificates from cacrt_all.pem
gen_crt_bundle.py: Successfully added 149 certificates
gen_crt_bundle.py: Parsing certificates from cacrt_local.pem
gen_crt_bundle.py: No certificate found
gen_crt_bundle.py: Successfully added 149 certificates in total
illegal instruction_

Then I tried this with my modified file - adding spaces at the end of each certificate - and I figured out, that putting a space at the end of a certificate makes the program to ignore this certificate.
so I have deleted the spaces one certificate after the other from the beginning and with the 19th certificate I get the error 

_ncb@mx:home/esp/esp-idf/components/mbedtls/esp_crt_bundle
$ ./gen_crt_bundle.py --input cacrt_all.pem cacrt_local.pem --max-certs 200
gen_crt_bundle.py: Parsing certificates from cacrt_all.pem
gen_crt_bundle.py: Successfully added 18 certificates
gen_crt_bundle.py: Parsing certificates from cacrt_local.pem
gen_crt_bundle.py: No certificate found
gen_crt_bundle.py: Successfully added 18 certificates in total

ncb@mx:home/esp/esp-idf/components/mbedtls/esp_crt_bundle
$ ./gen_crt_bundle.py --input cacrt_all.pem cacrt_local.pem --max-certs 200
gen_crt_bundle.py: Parsing certificates from cacrt_all.pem
gen_crt_bundle.py: Successfully added 19 certificates
gen_crt_bundle.py: Parsing certificates from cacrt_local.pem
gen_crt_bundle.py: No certificate found
gen_crt_bundle.py: Successfully added 19 certificates in total
illegal instruction_

it seems as to me as if there is some kind of a memory overflow with a variable or so, but I do not know much about Python.

AdityaHPatwardhan, 2025-01-22:
Hi @colsceg, I tried the same command on multiple systems (linux, macOS) inside the esp-idf environment, and I am unable to reproduce the issue. This seems to be related to python interpreter itself, not sure. @dobairoland would you know more?

I assume you are using the esp-idf environment itself ( Based on the instructions I shared earlier)
but could you please share the output of `$ which python`

Alternatively, There is also an option to disable the certificate bundle feature which will not use this script at all. You could do that by disabling `Component config → mbedTLS → Certificate Bundle -> Enable trusted root certificate bundle`.
This should allow you to build successfully without making any changes to the components directory

dobairoland, 2025-01-22:
What Linux distribution are you using? Did you install the bundled `python3` package or you've obtained it some other way?

We have already tried to re-create the ESP-IDF virtual environment. So probably there is no issue there. But it is possible that the Python package version provided by the Linux distribution has a related bug. It might help to try some other version. I will help you with that if you answer the above questions.

dobairoland, 2025-01-22:
@AdityaHPatwardhan The issue is pretty weird because if it would be a Python issue then we would've receive a traceback. The given issue suggest that the error is coming from the unmanaged part of the code (openssl or similar under the cryptography package). But we have re-created the ESP-IDF environment, so the cryptography package was also reinstalled (and I assume that the latest version). The issue could be coming from the openssl library.

fhrbata, 2025-01-22:
Hello @colsceg , just an idea, maybe worth trying. The `SIGILL` might originate from the cryptography shared library. Maybe the wheel is compiled with some extensions that your cpu is not supporting.

I guess you can try to install cryptography from sources instead of using wheel. In the activated(after running . ./export.sh) esp-idf environment uninstall the cryptography package with `pip uninstall cryptography` and install it from sources with `pip install --no-cache-dir --no-binary cryptography cryptography`. Maybe it will help.

You can also try to use gdb and maybe see what instruction is causing this.
```
gdb --args /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
(gdb) run
(gdb) x/i $pc
```
Anyway I might be completely wrong.

Thank you

colsceg, 2025-01-22:
to run the non-binary installation I first had to instal re Rust package.

After this I got the following Error maybe you know what I have to do:
pip install --no-cache-dir --no-binary cryptography cryptography
Collecting cryptography
  Downloading cryptography-44.0.0.tar.gz (710 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 710.7/710.7 kB 7.5 MB/s eta 0:00:00
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Requirement already satisfied: cffi>=1.12 in /home/ncb/.espressif/python_env/idf5.5_py3.11_env/lib/python3.11/site-packages (from cryptography) (1.17.1)
Requirement already satisfied: pycparser in /home/ncb/.espressif/python_env/idf5.5_py3.11_env/lib/python3.11/site-packages (from cffi>=1.12->cryptography) (2.22)
Building wheels for collected packages: cryptography
  Building wheel for cryptography (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for cryptography (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [170 lines of output]
      Running `maturin pep517 build-wheel -i /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python --compatibility off`
      📦 Including license file "/tmp/pip-install-yc9g636f/cryptography_b44dfa4d825a43a288ca6faff767fa63/LICENSE"
      📦 Including license file "/tmp/pip-install-yc9g636f/cryptography_b44dfa4d825a43a288ca6faff767fa63/LICENSE.APACHE"
      📦 Including license file "/tmp/pip-install-yc9g636f/cryptography_b44dfa4d825a43a288ca6faff767fa63/LICENSE.BSD"
      🍹 Building a mixed python/rust project
      🔗 Found pyo3 bindings with abi3 support for Python ≥ 3.7
      🐍 Not using a specific python interpreter
      📡 Using build options features, locked from pyproject.toml
         Compiling target-lexicon v0.12.16
         Compiling proc-macro2 v1.0.92
         Compiling unicode-ident v1.0.14
         Compiling shlex v1.3.0
         Compiling cc v1.2.1
         Compiling quote v1.0.37
         Compiling pyo3-build-config v0.23.2
         Compiling syn v2.0.89
         Compiling pkg-config v0.3.31
         Compiling vcpkg v0.2.15
         Compiling once_cell v1.20.2
         Compiling openssl-sys v0.9.104
      warning: openssl-sys@0.9.104: Could not find directory of OpenSSL installation, and this `-sys` crate cannot proceed without this knowledge. If OpenSSL is installed and this crate had trouble finding it,  you can set the `OPENSSL_DIR` environment variable for the compilation process. See stderr section below for further information.
      error: failed to run custom build command for `openssl-sys v0.9.104`
      
      Caused by:
        process didn't exit successfully: `/tmp/pip-install-yc9g636f/cryptography_b44dfa4d825a43a288ca6faff767fa63/target/release/build/openssl-sys-158bf67291c9ec4f/build-script-main` (exit status: 101)
        --- stdout
        cargo:rustc-check-cfg=cfg(osslconf, values("OPENSSL_NO_OCB", "OPENSSL_NO_SM4", "OPENSSL_NO_SEED", "OPENSSL_NO_CHACHA", "OPENSSL_NO_CAST", "OPENSSL_NO_IDEA", "OPENSSL_NO_CAMELLIA", "OPENSSL_NO_RC4", "OPENSSL_NO_BF", "OPENSSL_NO_PSK", "OPENSSL_NO_DEPRECATED_3_0", "OPENSSL_NO_SCRYPT", "OPENSSL_NO_SM3", "OPENSSL_NO_RMD160", "OPENSSL_NO_EC2M", "OPENSSL_NO_OCSP", "OPENSSL_NO_CMS", "OPENSSL_NO_COMP", "OPENSSL_NO_SOCK", "OPENSSL_NO_STDIO", "OPENSSL_NO_EC", "OPENSSL_NO_SSL3_METHOD", "OPENSSL_NO_KRB5", "OPENSSL_NO_TLSEXT", "OPENSSL_NO_SRP", "OPENSSL_NO_RFC3779", "OPENSSL_NO_SHA", "OPENSSL_NO_NEXTPROTONEG", "OPENSSL_NO_ENGINE", "OPENSSL_NO_BUF_FREELISTS"))
        cargo:rustc-check-cfg=cfg(openssl)
        cargo:rustc-check-cfg=cfg(libressl)
        cargo:rustc-check-cfg=cfg(boringssl)
        cargo:rustc-check-cfg=cfg(libressl250)
        cargo:rustc-check-cfg=cfg(libressl251)
        cargo:rustc-check-cfg=cfg(libressl252)
        cargo:rustc-check-cfg=cfg(libressl261)
        cargo:rustc-check-cfg=cfg(libressl270)
        cargo:rustc-check-cfg=cfg(libressl271)
        cargo:rustc-check-cfg=cfg(libressl273)
        cargo:rustc-check-cfg=cfg(libressl280)
        cargo:rustc-check-cfg=cfg(libressl281)
        cargo:rustc-check-cfg=cfg(libressl291)
        cargo:rustc-check-cfg=cfg(libressl310)
        cargo:rustc-check-cfg=cfg(libressl321)
        cargo:rustc-check-cfg=cfg(libressl332)
        cargo:rustc-check-cfg=cfg(libressl340)
        cargo:rustc-check-cfg=cfg(libressl350)
        cargo:rustc-check-cfg=cfg(libressl360)
        cargo:rustc-check-cfg=cfg(libressl361)
        cargo:rustc-check-cfg=cfg(libressl370)
        cargo:rustc-check-cfg=cfg(libressl380)
        cargo:rustc-check-cfg=cfg(libressl381)
        cargo:rustc-check-cfg=cfg(libressl382)
[... 100 lines cut ...]
        automatically, please open a bug at https://github.com/sfackler/rust-openssl
        and include information about your system as well as this message.
      
        $HOST = x86_64-unknown-linux-gnu
        $TARGET = x86_64-unknown-linux-gnu
        openssl-sys = 0.9.104
      
      
        It looks like you're compiling on Linux and also targeting Linux. Currently this
        requires the `pkg-config` utility to find OpenSSL but unfortunately `pkg-config`
        could not be found. If you have OpenSSL installed you can likely fix this by
        installing `pkg-config`.
      
      
      warning: build failed, waiting for other jobs to finish...
      💥 maturin failed
        Caused by: Failed to build a native library through cargo
[... cut ...] " "--release" "--lib"`
      Error: command ['maturin', 'pep517', 'build-wheel', '-i', '/home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python', '--compatibility', 'off'] returned non-zero exit status 1
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for cryptography
Failed to build cryptography
ERROR: ERROR: Failed to build installable wheels for some pyproject.toml based projects (cryptography)

fhrbata, 2025-01-22:
Hello @colsceg , thank you so much for responding. I just tried in debian 12 docker image in venv and the following worked for me. You should install the openssl libraries.

```
$ docker run  --rm -it debian:12
# apt-get update
# apt-get install -y curl
# curl --proto '=https' --tlsv1.2 -sSf https://sh.rustup.rs | sh
# . $HOME/.cargo/env
# apt install -y python3 python3-venv
# python3 -m venv venv
# . venv/bin/activate
# apt install libssl-dev pkg-config python3-dev gcc
# pip install --no-cache-dir --no-binary cryptography cryptography
```
Since you have the rust already installed I think only the last two commands should be required.
```
# apt install libssl-dev pkg-config python3-dev gcc
# pip install --no-cache-dir --no-binary cryptography cryptography
```

Thank you

colsceg, 2025-01-22:
you are right, I can install it _--no-cache-dir --no-binary_ in the container but with the outside the container I get an error

$ pip install --no-cache-dir --no-binary cryptography cryptography
error: externally-managed-environment

This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

the debugger shows a SIGILL

_(gdb) run
Starting program: /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_local.pem -q --max-certs 200
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGILL, Illegal instruction.
0x00007ffff2bc7379 in felem_contract ()
   from /home/ncb/.espressif/python_env/idf5.5_py3.11_env/lib/python3.11/site-packages/cryptography/hazmat/bindings/_rust.abi3.so
(gdb) x/i
Argument required (starting display address).
(gdb) x/i $pc
=> 0x7ffff2bc7379 <felem_contract+393>:	pcmpgtq %xmm2,%xmm3
(gdb)_ 

so I think you are right, I use the MX Linux on a MAC with an Intel Core2 Duo CPU.

fhrbata, 2025-01-22:
Hello @colsceg ,

>  I can install it --no-cache-dir --no-binary in the container but with the outside the container I get an error

This is expected. The distribution can mark its python env. as externally managed to inform pip not to modify it. The goal is to prevent conflicts or system instability that might arise from mixing package manager-managed(apt) dependencies with pip-installed ones. More information can be found here https://peps.python.org/pep-0668/

Anyway, as I wrote, you should try to run `pip install --no-cache-dir --no-binary cryptography cryptography` from within the esp-idf environment activated, meaning after `. ./export.sh` when you are in python virtual environment. Please see the steps below.

> => 0x7ffff2bc7379 <felem_contract+393>: pcmpgtq %xmm2,%xmm3
> (gdb)
> 
> so I think you are right, I use the MX Linux on a MAC with an Intel Core2 Duo CPU.

The `pcmpgtq` instr is supported in SSE4.2 since `Nehalem` microarchitecture, but not in `Core`  microarchitecture. You can verify supported extensions in the `cat /proc/cpuinfo` output. The problem seems to be that the wheel is build on newer microarchitecture and we should probably fix this.

I think you can work around it with the following steps.

1. From clean environment, where the esp-idf is not activated(e.g. new terminal) run `./install.sh`
2. Activate esp-idf python environment with `. ./export.sh`
3. Replace the pre-build cryptography wheel compiled with SSE4.2 support:
```
$ pip uninstall cryptography
$ pip install --no-cache-dir --no-binary cryptography cryptography
```

IIUC you have all the requirements needed to build cryptography already installed, so hopefully it should be fast and  hassle-free.

Once again, thank you so much for all the effort and time you put into helping figure this out!

fhrbata, 2025-01-23:
Hi @colsceg,

I think I've figured out the issue. Building the cryptography package as mentioned earlier should work, but it might not be necessary. You just need to remove the wheel downloaded from the https://dl.espressif.com/pypi repository during installation and replace it with the upstream version.

1. Run `. ./export.sh` to activate the ESP-IDF python environment.
2. Use `pip uninstall cryptography` to remove the problematic cryptography-44.0.0-cp39-abi3-manylinux_2_34_x86_64.whl package that was downloaded from the https://dl.espressif.com/pyp registry.
3. Then, `pip install cryptography` to install the upstream cryptography-44.0.0-cp39-abi3-manylinux_2_28_x86_64.whl.

Here's what happened: Espressif manages a PyPI repository at `https://dl.espressif.com/pypi`, which is used during the installation of the ESP-IDF Python virtual environment. This repository includes necessary wheels for ESP-IDF, including wheels that aren't available on PyPI. Unfortunately, it contains the `cryptography-44.0.0-cp39-abi3-manylinux_2_34_x86_64.whl` package, which has an OpenSSL build with SSE4.2 extensions (`pcmpgtq`) causing the issue you're experiencing. This package was originally downloaded from the official cryptography PyPI repository but was removed from there due to this problem. More details can be found here: https://github.com/pyca/cryptography/issues/12069.

> We may need to file an issue with manylinux regarding this. Meanwhile, the manylinux_2_34_x86_64 wheels for 44.0.0 have been removed.

The problem is that `https://dl.espressif.com/pypi` hasn't updated to reflect this change and still includes the problematic package. @dobairoland, I suggest we remove `cryptography-44.0.0-cp39-abi3-manylinux_2_34_x86_64.whl` and replace it with `cryptography-44.0.0-cp39-abi3-manylinux_2_28_x86_64.whl`. That's IMHO all is needed to fix this.

dobairoland, 2025-01-24:
@colsceg Could you please confirm if @fhrbata's solution works for you?

colsceg, 2025-01-24:
Ok great, works fine, no errors when running the Original cacrt_all.pem file.

 /home/ncb/.espressif/python_env/idf5.5_py3.11_env/bin/python /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/gen_crt_bundle.py --input /home/ncb/esp/esp-idf/components/mbedtls/esp_crt_bundle/cacrt_all.pem -q --max-certs 200
ncb@mx:home
The idf.py build at /blink directory works after a idf.py fullclean without an error.
Thanks  for your work

fhrbata, 2025-02-13:
The problematic wheel `cryptography-44.0.0-cp39-abi3-manylinux_2_34_x86_64.whl` was removed at `https://dl.espressif.com/pypi/cryptography/` and replaced with `cryptography-44.0.0-cp39-abi3-manylinux_2_28_x86_64.whl`. Moreover there is a new `44.0.1` release which is compiled with correct flags and should be used by default during `./install.sh`. I believe this problem is resolved now. Thank you all.

closed 2025-02-13 by fhrbata