espressif/esp-idf #15209: Unable to use SD card with TWAI with patched esp_rom_gpio_connect_out_signal for esp32

louisthetall, 2025-01-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2-283-g983653de9a

### Espressif SoC revision.

ESP32-D0WD-V3 (revision v3.0)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom board

### Power Supply used.

External 5V

### What is the expected behavior?

Program should be able read/write to the sd card after the twai has been initialized

### What is the actual behavior?

Program fails to read/write to the sd card after the twai has been initialized.

### Steps to reproduce.

```
/* SD card and FAT filesystem example.
This example code is in the Public Domain (or CC0 licensed, at your option.)

Unless required by applicable law or agreed to in writing, this
software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND, either express or implied.
*/

// This example uses SDMMC peripheral to communicate with SD card.

#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_vfs_fat.h"
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
#include "sd_test_io.h"

#include "driver/twai.h"
#include "esp_idf_version.h"

#define EXAMPLE_MAX_CHAR_SIZE    64

static const char *TAG_SDMMC = "SDMMC";
static const char *TAG_TWAI = "TWAI";
[... 352 lines cut ...]
Name: SD32G
Type: SDHC/SDXC
Speed: 20.00 MHz (limit: 20.00 MHz)
Size: 29580MB
CSD: ver=2, sector_size=512, capacity=60579840 read_bl_len=9
SSR: bus_width=4
I (383) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (393) gpio: GPIO[32]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
I (403) TWAI: Driver installed
I (403) TWAI: Driver started
I (403) SDMMC: Opening file /sdcard/hello.txt
E (413) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0xffffffff
E (413) diskio_sdmmc: sdmmc_read_blocks failed (0xffffffff)
E (423) SDMMC: Failed to open file for writing
I (423) TWAI: Driver stopped
I (423) TWAI: Driver uninstalled
I (433) SDMMC: Card unmounted
I (433) main_task: Returned from app_main()
```

### More Information.

Was working mainly with esp-idf v5.3.2 and decided to update to esp-idf v5.4, when the sd card on our boards using the esp32 stopped working. I narrowed the issue down to the patched esp_rom_gpio_connect_out_signal for esp32 and esp32s2. 

Reverting [fix(gpio): patched esp_rom_gpio_connect_out_signal for esp32 and esp32s2](https://github.com/espressif/esp-idf/commit/ab4157b6cf45af10afe2ceba792af99afe0740c5) resolves the issue

suda-morris, 2025-01-15:
> I (383) gpio: GPIO[39]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0

This log shows you're not using the final version of release v5.4, which should contain a fix of https://github.com/espressif/esp-idf/commit/928b7f6efcf7baf2bcd6e2fd3894f8a40c7c6dfd Not sure if this commit is related to your problem, but please try to upgrade your esp-idf to use the v5.4 release from https://github.com/espressif/esp-idf/releases/tag/v5.4

louisthetall, 2025-01-15:
> This log shows you're not using the final version of release v5.4, which should contain a fix of [928b7f6](https://github.com/espressif/esp-idf/commit/928b7f6efcf7baf2bcd6e2fd3894f8a40c7c6dfd)

Ah, sorry of the confusion. The esp-idf version I listed in the the issue is the location where [ab4157b](https://github.com/espressif/esp-idf/commit/ab4157b6cf45af10afe2ceba792af99afe0740c5) is merged into esp-idf `release/v5.3`. The issue is still present at the head of esp-idf `release/v5.4` ([c8bb532](https://github.com/espressif/esp-idf/commit/c8bb53292d08d6449a09823cf554e62ac839cd8c)) and `release/v5.3` ([adf5319](https://github.com/espressif/esp-idf/commit/adf5319639b7e66e4f80868b14fee1acb73672c7))

I did try reverting [928b7f6](https://github.com/espressif/esp-idf/commit/928b7f6efcf7baf2bcd6e2fd3894f8a40c7c6dfd) when I came across this issue, but it had no effect.

songruo, 2025-01-16:
Hi @louisthetall, please help us narrow down the problem. If you don't install TWAI driver, does the SD card work as expected? If it turns that the problem only happens with TWAI, please see whether `twai_driver_install()` or `twai_start()` causes the issue.

louisthetall, 2025-01-22:
> Hi [@louisthetall](https://github.com/louisthetall), please help us narrow down the problem. If you don't install TWAI driver, does the SD card work as expected? If it turns that the problem only happens with TWAI, please see whether `twai_driver_install()` or `twai_start()` causes the issue.

The problem occurs when `twai_driver_install()` is used. If the TWAI driver is not installed the SD card works as expected.

songruo, 2025-02-05:
[0001-fix-gpio-fix-esp_rom_gpio_connect_out_signal-for-gpi.patch](https://github.com/user-attachments/files/18673506/0001-fix-gpio-fix-esp_rom_gpio_connect_out_signal-for-gpi.patch)

@louisthetall Could you help to check if this patch fixes the issue?

closed 2025-02-13 by espressif-bot