espressif/esp-idf #15212: Smart config - SC_EVENT_SEND_ACK_DONE never received

leenowell, 2025-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

ESP-IDF v5.5-dev-1050-gb5ac4fbdf9

### Espressif SoC revision.

ESP32-D0WDQ6 (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP WROOM 32

### Power Supply used.

USB

### What is the expected behavior?

Having initiated Smart Config and received a SC_EVENT_GOT_SSID_PSWD event I expect to subsequently receive a SC_EVENT_SEND_ACK_DONE event.

### What is the actual behavior?

I never receive the SC_EVENT_SEND_ACK_DONE event

### Steps to reproduce.

Run the smart config example and send the smart config data using the ESP Touch app (I used Android version - App version 2.3.3 ESP Touch version 1.1.1) and used regular ESP Touch as opposed to v2.

### Debug Logs.

```plain
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6328
load:0x40078000,len:15828
load:0x40080400,len:3860
entry 0x40080638
I (27) boot: ESP-IDF v5.5-dev-1050-gb5ac4fbdf9 2nd stage bootloader
I (27) boot: compile time Jan  7 2025 18:10:32
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.0
I (33) boot.esp32: SPI Speed      : 40MHz
I (36) boot.esp32: SPI Mode       : DIO
I (40) boot.esp32: SPI Flash Size : 2MB
I (44) boot: Enabling RNG early entropy source...
I (48) boot: Partition Table:
I (51) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (63) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (70) boot:  2 factory          factory app      00 00 00010000 00100000
I (77) boot: End of partition table
I (80) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=19bb4h (105396) map
I (123) esp_image: segment 1: paddr=00029bdc vaddr=3ff80000 size=0001ch (    28) load
I (124) esp_image: segment 2: paddr=00029c00 vaddr=3ffb0000 size=03f1ch ( 16156) load
I (133) esp_image: segment 3: paddr=0002db24 vaddr=40080000 size=024f4h (  9460) load
I (138) esp_image: segment 4: paddr=00030020 vaddr=400d0020 size=8df7ch (581500) map
I (341) esp_image: segment 5: paddr=000bdfa4 vaddr=400824f4 size=1531ch ( 86812) load
I (386) boot: Loaded app from partition at offset 0x10000
I (386) boot: Disabling RNG early entropy source...
I (397) cpu_start: Multicore app
I (405) cpu_start: Pro cpu start user code
I (405) cpu_start: cpu freq: 160000000 Hz
I (405) app_init: Application information:
I (405) app_init: Project name:     smart_config
I (410) app_init: App version:      1
I (413) app_init: Compile time:     Jan  7 2025 18:09:46
I (418) app_init: ELF file SHA256:  7595b2674...
I (423) app_init: ESP-IDF:          v5.5-dev-1050-gb5ac4fbdf9
I (428) efuse_init: Min chip rev:     v0.0
I (432) efuse_init: Max chip rev:     v3.99 
I (436) efuse_init: Chip rev:         v0.0
I (440) heap_init: Initializing. RAM available for dynamic allocation:
I (446) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (451) heap_init: At 3FFB83B8 len 00027C48 (159 KiB): DRAM
I (456) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (462) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (467) heap_init: At 40097810 len 000087F0 (33 KiB): IRAM
I (474) spi_flash: detected chip: gd
I (476) spi_flash: flash io: dio
W (479) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (492) main_task: Started on CPU0
I (502) main_task: Calling app_main()
I (552) wifi:wifi driver task: 3ffc047c, prio:23, stack:6656, core=0
I (572) wifi:wifi firmware version: 34d97ea27
I (572) wifi:wifi certification version: v7.0
I (572) wifi:config NVS flash: enabled
I (572) wifi:config nano formatting: disabled
I (572) wifi:Init data frame dynamic rx buffer num: 32
I (582) wifi:Init static rx mgmt buffer num: 5
I (582) wifi:Init management short buffer num: 32
I (592) wifi:Init dynamic tx buffer num: 32
I (592) wifi:Init static rx buffer size: 1600
I (592) wifi:Init static rx buffer num: 10
I (602) wifi:Init dynamic rx buffer num: 32
I (602) wifi_init: rx ba win: 6
I (602) wifi_init: accept mbox: 6
I (612) wifi_init: tcpip mbox: 32
I (612) wifi_init: udp mbox: 6
I (612) wifi_init: tcp mbox: 6
I (622) wifi_init: tcp tx win: 5760
I (622) wifi_init: tcp rx win: 5760
I (622) wifi_init: tcp mss: 1440
I (622) wifi_init: WiFi IRAM OP enabled
I (632) wifi_init: WiFi RX IRAM OP enabled
I (632) phy_init: phy_version 4840,02e0d70,Sep  2 2024,19:39:07
I (722) wifi:mode : sta (24:0a:c4:82:6f:1c)
I (722) wifi:enable tsf
I (722) main_task: Returned from app_main()
I (772) smartconfig: SC version: V3.0.2
I (5602) wifi:ic_enable_sniffer
I (5612) smartconfig: Start to find channel...
I (5612) smartconfig_example: Scan done
I (5792) smartconfig: TYPE: ESPTOUCH
I (5792) smartconfig: T|AP MAC: f0:86:20:13:b4:33
I (5792) smartconfig: Found channel on 11-0. Start to get ssid and password...
I (5802) smartconfig_example: Found channel
I (8342) smartconfig: T|pswd: wivbke
I (8342) smartconfig: T|ssid: Lounge
I (8342) wifi:ic_disable_sniffer
I (8342) smartconfig_example: Got SSID and password
I (8352) smartconfig_example: SSID:Lounge
I (8352) smartconfig_example: PASSWORD:wivbke
I (8692) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (8692) wifi:state: init -> auth (0xb0)
I (8692) wifi:state: auth -> assoc (0x0)
I (8702) wifi:state: assoc -> run (0x10)
I (11702) wifi:state: run -> init (0xf00)
I (11712) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (14132) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (14132) wifi:state: init -> auth (0xb0)
I (14152) wifi:state: auth -> assoc (0x0)
I (14162) wifi:state: assoc -> run (0x10)
I (17152) wifi:state: run -> init (0xf00)
I (17172) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (19602) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (19602) wifi:state: init -> auth (0xb0)
I (19612) wifi:state: auth -> assoc (0x0)
I (19622) wifi:state: assoc -> run (0x10)
I (22622) wifi:state: run -> init (0xf00)
I (22632) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (25062) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (25062) wifi:state: init -> auth (0xb0)
I (25072) wifi:state: auth -> assoc (0x0)
I (25082) wifi:state: assoc -> run (0x10)
I (28102) wifi:state: run -> init (0xf00)
I (28112) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (30542) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (30542) wifi:state: init -> auth (0xb0)
I (30552) wifi:state: auth -> assoc (0x0)
I (30552) wifi:state: assoc -> run (0x10)
I (32582) wifi:<ba-add>idx:0 (ifx:0, f0:86:20:13:b4:33), tid:6, ssn:0, winSize:64
I (33552) wifi:state: run -> init (0xf00)
I (33572) wifi:<ba-del>idx:0, tid:6
I (33572) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (35992) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
I (35992) wifi:state: init -> auth (0xb0)
I (36012) wifi:state: auth -> assoc (0x0)
I (36012) wifi:state: assoc -> run (0x10)
I (39012) wifi:state: run -> init (0xf00)
I (39022) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:11, snd_ch_cfg:0x0
```

### More Information.

I modified the code to log when the SC_EVENT_SEND_ACK_DONE was received as follows

    } else if (event_base == SC_EVENT && event_id == SC_EVENT_SEND_ACK_DONE) {
        ESP_LOGI(TAG, "Smart Config ACK done");
        xEventGroupSetBits(s_wifi_event_group, ESPTOUCH_DONE_BIT);
    }

leenowell, 2025-01-21:
Hi @MaxwellAlan  I was wondering if you have any updates on this?

zhangyanjiaoesp, 2025-01-22:
@leenowell 
It seems your wifi connection fail, is the password correct? Can you capture the wifi packets using Wireshark?

leenowell, 2025-01-23:
Hi @zhangyanjiaoesp 

Thanks for getting back to me.  This is related to the other issue we are working on related to getting smart config and the mesh to work together -  https://github.com/espressif/esp-idf/issues/15213.  

I think this one can now be closed as it was covered on the other issue

zhangyanjiaoesp, 2025-01-23:
@leenowell Please close this on your side, thanks

closed 2025-01-23 by leenowell