espressif/esp-idf #15214: [ESP-WIFI-MESH] When the root node is turn ON/OFF random time, Mesh switch will go into a hung state after some time, triggering the watchdog or permanently hanging.

andy-danieal, 2025-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2-1-g9a6aba7401

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

```

[0;33mW (02:03:31.931) [MESH-EVENT]: <MESH_EVENT_ROUTING_TABLE_ADD>add 1, new:4, layer:4[0m
W (7414911) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:0
W (7416712) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:1
W (7418513) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:2
W (7420314) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:3
W (7422115) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:4
W (7423916) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:5
W (7425717) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:6
W (7427520) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:7
W (7429321) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:8
W (7431122) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:9
W (7432923) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:10
W (7434724) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:11
W (7436525) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:12
W (7438326) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:13
W (7440127) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:14
W (7441928) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:15
W (7443729) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:16
W (7445530) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:17
W (7447331) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:18
W (7449132) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:19
W (7450933) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:20
W (7452734) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:21
W (7454535) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:225, xseqno:187, no_wnd_count:0, timeout_count:22
[0;33mW (02:04:36.389) [MESH-EVENT]: <MESH_EVENT_ROUTING_TABLE_REMOVE>remove 1, new:3, layer:4[0m
[0;33mW (02:05:31.447) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
Touch Invalid
[0;33mW (02:05:31.998) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:32.162) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:32.476) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
Touch Invalid
Touch Invalid
Touch Invalid
[0;33mW (02:05:33.193) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
Touch Invalid
[0;33mW (02:05:34.984) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:35.693) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:35.882) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:36.227) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:36.434) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:36.925) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:37.295) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:37.531) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[0;33mW (02:05:37.797) [PACKET]: Send command to gateway :- a0:76:4e:8c:e9:33 [0m
[... 188 lines cut ...]
W (7789709) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:133
W (7791510) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:134
W (7793311) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:135
W (7795112) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:136
W (7796913) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:137
W (7798714) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:138
W (7800515) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:139
W (7802316) mesh: [mesh_schedule.c,3130] [WND-RX]max_wnd:7, 1800 ms timeout, seqno:923, xseqno:202, no_wnd_count:0, timeout_count:140
[0;31mE (7802832) task_wdt: Task watchdog got triggered. The following tasks/users did not reset the watchdog in time:[0m
[0;31mE (7802832) task_wdt:  - main (CPU 0)[0m
[0;31mE (7802832) task_wdt: Tasks currently running:[0m
[0;31mE (7802832) task_wdt: CPU 0: IDLE[0m
[0;31mE (7802832) task_wdt: Aborting.[0m
[0;31mE (7802832) task_wdt: Print CPU 0 (current core) backtrace[0m

Backtrace: 0x4002b80b:0x3ffdab10 0x4009bd4b:0x3ffdab30 0x4002ebbd:0x3ffdab50 0x4002dc42:0x3ffdab70
``

### What is the actual behavior?

If we set block time 2 sec then why is triggering happening in watchdog state.

### Steps to reproduce.

Mesh root node power on/off sometime

zhangyanjiaoesp, 2025-01-16:
@andy-danieal 

> If we set block time 2 sec then why is triggering happening in watchdog state.

Please take a look at this line:
https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/include/esp_mesh.h#L690

Please increase the block time and try again.

zhangyanjiaoesp, 2025-01-16:
> @zhangyanjiaoesp, Watchdog timer set at one minute and esp block time is only 2 seconds. then why does WDT timeout trigger ?

It seems that `esp_mesh_send()` did not return in time, causing the watchdog to trigger.
> If increased timing Block time then other application process was impacted. Can you other alternative way non-blocking way to send packet on root node.

If you want to set the mesh send to non-blocking, you can set the `data.tos = MESH_TOS_DEF`

andy-danieal, 2025-01-17:
@zhangyanjiaoesp, I have replaced *MESH_TOS_P2P* to *MESH_TOS_DEF*. but getting **MESH ERROR:- 0x4008**

```
    mesh_data_t data_out;

    data_out.data  = (uint8_t *)data;
    data_out.size  = data_length;
    data_out.proto = MESH_PROTO_BIN;
    data_out.tos   = MESH_TOS_DEF; //MESH_TOS_P2P

   return esp_mesh_send(NULL, &data_out, MESH_DATA_P2P | MESH_DATA_NONBLOCK, NULL, 0);
```

zhangyanjiaoesp, 2025-01-17:
1. If you set the `to` address to NULL, then `data.tos` must be MESH_TOS_P2P. Please read the description section of the API carefully.
https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/include/esp_mesh.h#L638
https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/esp_wifi/include/esp_mesh.h#L646

2.  > if block time is two seconds, the esp_mesh_send() function doesn't return?

I don’t know why there was no return in your case. We need to find the root cause of the issue.

andy-danieal, 2025-01-17:
@zhangyanjiaoesp,  At the end, we can't send ROOT device Non-block packets via **esp_mesh_send** function.

On field device was freezed state. We need your help to debug this scenario to identify main issue?

zhangyanjiaoesp, 2025-01-17:
@andy-danieal 
1. What does your mesh network structure look like ? 
2. How is your mesh configured ？
3. When do you call `esp_mesh_send` ? 
4. Can you describe the scenario when the issue occurs ?

It would be even better if you could provide a demo to reproduce the issue.

andy-danieal, 2025-01-20:
@zhangyanjiaoesp 

``1. What does your mesh network structure look like ?``

[image]

``2. How is your mesh configured ？``
Fixed Root Node and No Router settings

``3. When do you call esp_mesh_send ?``
When touch switch then send esp_mesh_send package

``4.Can you describe the scenario when the issue occurs ?``
Random time ON/OFF gateway and switch touch then observe that issue.

zhangyanjiaoesp, 2025-01-20:
@andy-danieal can you provide a demo to reproduce the issue?

zhangyanjiaoesp, 2025-02-25:
@andy-danieal 
Do you still have this issue? If no, can we close it?

Alvin1Zhang, 2025-03-06:
Thanks for reporting, will close due to inactivity, feel free to reopen if the issue still happens. Thanks.

closed 2025-03-06 by Alvin1Zhang