espressif/esp-idf #15216: ESP32-C6 "wifi:FTM Initiator mode not supported!"

thielemans, 2025-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32-C6 (QFN40) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C6-DevKitC-1 V1.2

### Power Supply used.

USB

### What is the expected behavior?

WiFi FTM example should run on ESP32-C6 as it is listed as supported in the example and C6 datasheet.

### What is the actual behavior?

WiFi responds that FTM Initiator mode is not supported although documentation indicates this should work.

I (13831) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
E (13831) wifi:FTM Initiator mode not supported!
E (13841) ftm_station: Failed to start FTM session

### Steps to reproduce.

1. Verify FTM Initiator support and FTM Responder support is enabled.
2. Optionally enable FTM report logging (tested with enabled and disabled). 
3. WiFi FTM example built for ESP32-C6 using idf.py or VScode (tested with both environments). 
4. Flashed on two separate ESP32-C6 devices
5. Device A put in softAP mode
6. Optionally connect device B in STA mode (tested with and without this step)
7. Device B: "ftm -I -s <ssid of A>

### Debug Logs.

```plain
ESP-ROM:esp32c6-20220919
Build:Sep 19 2022
rst:0x1 (POWERON),boot:0x1c (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:2
load:0x40875720,len:0x16b8
load:0x4086c110,len:0xe84
load:0x4086e610,len:0x304c
entry 0x4086c11a
I (23) boot: ESP-IDF v5.4 2nd stage bootloader
I (24) boot: compile time Jan 15 2025 16:53:00
[... 84 lines cut ...]
 |                                                              |
 |  1. Start SoftAP with command 'ap <SSID> <password>'         |
 |  2. Optionally add '-c <ch>' to set channel, '-b 40' for 40M |
 |  3. Use 'help' for detailed information for all parameters   |
 ================================================================

Type 'help' to get the list of commands.
Use UP/DOWN arrows to navigate through command history.
Press TAB when typing command name to auto-complete.
I (691) main_task: Returned from app_main()
ftm> scan
I (2771) ftm_station: sta start to scan
I (5271) ftm_station: [test][rssi=-33][FTM Responder]
I (5271) ftm_station: [Pretty fly for a WiFi][rssi=-59]
I (5271) ftm_station: [esterierwifi?][rssi=-60]
I (5281) ftm_station: [esterierwifi?][rssi=-88]
I (5281) ftm_station: [Pretty fly for a WiFi][rssi=-90]
I (5281) ftm_station: [kelder][rssi=-92]
I (5291) ftm_station: sta scan done
ftm> ftm -I -s test
I (13831) ftm_station: Requesting FTM session with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
E (13831) wifi:FTM Initiator mode not supported!
E (13841) ftm_station: Failed to start FTM session
ftm> wifi:FTM Initiator mode not supported!wifi:FTM Initiator mode not supported!I (13831) ftm_station: Requesting FTM sessio                                                                ftm> wifi:FTM Initiator mode not supported!wifi:FTM Initiator mode not supported!I (13831) ftm_station: Requesting FTM sessio                                                                n with Frm Count - 32, Burst Period - 200mSec (0: No Preference)
```

nachiketkukade, 2025-01-24:
@thielemans , ESP32-C6 ECO1 has a hardware limitation due to which FTM Initiator mode cannot be supported on it. C6 ECO2 resolved this and supports FTM Initiator as expected.

thielemans, 2025-01-24:
Thank you @nachiketkukade for the clarification. Is there by any chance a newer development kit with separate model / EAN available that always feature the new chip revision? Much to our surprise we bought these development kits with apparently still the old the revision just weeks ago at a major electronics retailer.

Alternatively a development kit with a different chip model would also be fine as we do not specifically need the C6. Only requirement at this point is for FTM to absolutely work.

thielemans, 2025-02-27:
We eventually bought S3 devkits and still have C3 devkits on backorder. Confirmed to work as intended on S3.
Still disappointed about the rather hidden documentation on this feature not working on the ESP32-C6-DevKitC-1 (or at least with the initial revision which is still the current stock at retailers).

closed 2025-02-27 by thielemans

nachiketkukade, 2025-02-28:
> Is there by any chance a newer development kit with separate model / EAN available that always feature the new chip revision?
Sorry I'm not aware about this.

> Still disappointed about the rather hidden documentation on this feature not working on the ESP32-C6-DevKitC-1 

@thielemans , apologies for the trouble caused. Currently we don't have a feature compatibility matrix that mentions hardware versions, we only have one document for recommended IDF versions. We'll try to find the best way to communicate this info in future