espressif/esp-idf #15217: BLE Mesh Stack Overrides Custom BLE Advertising Parameters

lebrislo, 2025-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-MINI-1

### Power Supply used.

USB

### What is the expected behavior?

When using the ESP BLE Mesh stack along with BLE GATT advertising, I encounter an issue where my custom BLE advertising data is overridden by the BLE Mesh stack after the advertising duration ends. Specifically:

- I configure BLE advertising parameters to include my vendor-specific service UUID in the complete list of service UUIDs.

- Advertising works as expected for the specified duration (e.g., 19 seconds).

- After the BLE advertising duration ends, only the BLE Mesh Proxy Service UUID (0x1828) is advertised, and my vendor-specific UUID is no longer included.

This behavior prevents the coexistence of BLE Mesh Proxy and custom BLE GATT advertising.

The BLE advertising should continue periodically with the configured parameters, including both the BLE Mesh Proxy Service UUID (0x1828) and my vendor-specific service UUID.

### What is the actual behavior?

After the BLE advertising duration ends, the ESP BLE Mesh stack takes control of the advertising process and only advertises the Mesh Proxy Service UUID (0x1828). My vendor-specific UUID is no longer included.

### Steps to reproduce.

1.  Initialize the ESP BLE Mesh stack.
2.  Configure BLE advertising parameters as shown below:

```C
static esp_ble_mesh_ble_adv_param_t g_bleMeshAdvParam = {
    .interval       = 0x20,
    .adv_type       = ADV_TYPE_IND,
    .own_addr_type  = BLE_ADDR_TYPE_PUBLIC,
    .peer_addr_type = BLE_ADDR_TYPE_PUBLIC,
    .peer_addr      = {0},
    .duration       = 19000,
    .period         = 20000,
    .count          = ESP_BLE_MESH_BLE_ADV_INFINITE,
    .priority       = ESP_BLE_MESH_BLE_ADV_PRIO_HIGH,
};
```
3. Start BLE advertising with the above parameters.

4. Observe the advertising data using a BLE scanner app (e.g., nRF Connect).

### More Information.

Is there a supported way to include custom vendor-specific UUIDs alongside the Mesh Proxy Service UUID in the BLE advertising data?

If not, can you suggest an alternative approach or API updates to allow coexistence of BLE Mesh and custom BLE GATT advertising?

Thank you for your support!

forx157, 2025-01-17:
Hi, @lebrislo 

Do you mean mesh proxy adveritising data to carry your UUID information? This shouldn't be the case, your UUID information should be included in your own broadcast message!

forx157, 2025-01-20:
Hi, @lebrislo 

Why did you set adv data len and scan response len to 0?

How did you set the service uuid information when using ble mesh?

forx157, 2025-01-22:
Hi, @lebrislo 

Sorry for the late reply.

After my testing and confirmation, it is not possible to broadcast your own private service uuid when using ble mesh to coexist with ble adv, you can submit a feature request in github and we will support it in the future and sync the information to you.