espressif/esp-idf #15219: Using ulp_riscv_i2c.h and ulp_riscv_gpio.h in the same file leads to enumerator redeclarations of GPIO_NUM_*

Mychaylo-evox, 2025-01-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

It should compile the project.

### What is the actual behavior?

When compiling, IDE gives an error stating that GPIO_NUM_* is being redeclared, below are the libraries that are being mentioned:
v5.3.2/esp-idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h
v5.3.2/esp-idf/components/soc/esp32s3/include/soc/gpio_num.h

### Steps to reproduce.

Adding the libraries "ulp_riscv_gpio.h" and "ulp_riscv_i2c.h" in the main.c that is to be run in the risc-v coprocessor and (trying to) compiling it

### Build or installation Logs.

```plain
C:/Users/User/esp/v5.3.2/esp-idf/components/ulp/ulp_riscv/ulp_core/include/ulp_riscv_gpio.h:21:5: error: redeclaration of enumerator 'GPIO_NUM_1'
   21 |     GPIO_NUM_1 = 1,     /*!< GPIO1, input and output */
      |     ^~~~~~~~~~
C:/Users/User/esp/v5.3.2/esp-idf/components/soc/esp32s3/include/soc/gpio_num.h:19:5: note: previous definition of 'GPIO_NUM_1' with type 'enum <anonymous>'
   19 |     GPIO_NUM_1 = 1,     /*!< GPIO1, input and output */
      |     ^~~~~~~~~~
```

### More Information.

After looking around, I found out that "ulp_riscv_gpio.h" declares GPIO_NUM_\*, while "ulp_riscv_i2c.h" includes "hal/gpio_types.h" which also include "soc/gpio_num.h" which also declares GPIO_NUM_\*, creating the conflict.

atanisoft, 2025-01-15:
IIRC this was logged and fixed somewhat recently. You might need to refresh to the tip of the release-5.3 branch as it may not be available in that version of a release just yet

SoucheSouche, 2025-01-16:
Hello @Mychaylo-evox, thanks for pointing this one out!
This was indeed fixed from 5.4 onward and backported to 5.1 and 5.2 but for some reasons, it was never backported to 5.3. We will backport the fix to 5.3 as well.

Mychaylo-evox, 2025-01-16:
Thank you, can I know when it will be available for update?
And how can I get the new 'sync'? I don't know how to work with that and can't find any help online, I've been using the express instalation untill now, I tried it again but I still get the bug so I am assuming that it's not the express instalation that I should use to update the components folder with the new fix.

Thank you

SoucheSouche, 2025-01-16:
I will let you know when the fix will be available.
Are you using the ESP-IDF vscode extension ?

Mychaylo-evox, 2025-01-16:
Yes, I am using the extention.

brianignacio5, 2025-01-17:
You can try to run the `ESP-IDF: Open ESP-IDF Terminal` and type:

```
cd $Env:IDF_PATH
git fetch
git checkout release/v5.3
git pull
git submodule update --init --recursive
```

These steps will only update ESP-IDF itself and not ESP-IDF tools or ESP-IDF python Virtual environment packages. If you need to update those too, you can try using the Express install with `Find ESP-IDF` in your system option or running the `$Env:IDF_PATH/install.ps1` script.

Alvin1Zhang, 2025-01-26:
Thanks for reporting, the fix on release/5.3 is available https://github.com/espressif/esp-idf/commit/1ef5be3b399695645f18294f25d824ba4742c685, feel free to reopen with more updates.

closed 2025-01-26 by Alvin1Zhang