espressif/esp-idf #15222: MQTTS client connection to MQTT broker on IPV6 only instance.

Zeni241, 2025-01-16:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My mosquitto MQTT broker is hosted on AWS EC2, which only has IPV6 address attached to it (no IPV4 address, as it costs 😉 ).
The problem is I cannot connect my MQTT client on my ESP32 to this broker. I am sure this is due to IPV6 only address of my  EC2 because if I attach an IPV4 elastic IP to this instance, the MQTT client on ESP32 can easily connect to the broker on EC2. 

Any solution please.

euripedesrocha, 2025-01-17:
Hi @Zeni241, could you provide logs of the connection phase with the errors? Please enable VERBOSE log level for the components, refer to mqtt client examples on IDF for the list of most important ones and how to do it.

Zeni241, 2025-01-18:
Hi @euripedesrocha Thanks for your response . In fact I am not able to get IPV6 address for my ESP32 when I connect to wifi. I only get IPV4 address and this may be the reason I cannot connect to mqtt broker behind IPV6. 

To connect to wifi my code is

        `ESP_ERROR_CHECK(esp_netif_init());
        ESP_ERROR_CHECK(esp_event_loop_create_default());
        esp_netif_t *sta_netif = esp_netif_create_default_wifi_sta();
        assert(sta_netif);
       esp_netif_create_ip6_linklocal(sta_netif);
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        ESP_ERROR_CHECK(esp_wifi_init(&cfg));
        ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_STA));`

And in event handler I have added"

    `else if (event_base == IP_EVENT && event_id == IP_EVENT_GOT_IP6) 
    {
        ip_event_got_ip6_t *event = (ip_event_got_ip6_t *)event_data;
        ESP_LOGI(TAG, "got ipv6: " IPV6STR, IPV62STR(event->ip6_info.ip));
        if (event->ip6_info.ip.addr[0] == 0) {
            ESP_LOGW("IPv6", "Only link-local IPv6 assigned (fe80::). No global IPv6.");
        } else {
            ESP_LOGI("IPv6", "ESP32 has a global IPv6 address!");
        }
       
    } `

    
I have checked that my router supports SLAAC and has IPV6 global address. Can you please guide me how can I get IPV6 address when I connect ESP32 to wifi. Presently it get only IPV4 address.

nopnop2002, 2025-01-19:
>Can you please guide me how can I get IPV6 address when I connect ESP32 to wifi. 

The easiest way is to use example_connect.

https://github.com/espressif/esp-idf/blob/master/examples/protocols/mqtt/tcp/main/app_main.c#L159

Zeni241, 2025-01-19:
Thanks @euripedesrocha  for guiding me  I will go through this example and give my feedback here.

Zeni241, 2025-01-19:
Dear @euripedesrocha thanks for the link, Now I am able to get Global IPV6 address in my ESP-32 WIFI connection and now I can connect my MQTT client on my ESP32 to this broker behind IPV6 only address.
The main mistake I was making was that I should put the line  `esp_netif_create_ip6_linklocal(sta_netif);`  in wifi connected callback code (once the wifi is connected) instead of in wifi_ init() or at any other place.

closed 2025-01-23 by Zeni241

mickeyreg, 2026-10-22:
I'm learning OpenThread as an educational project: I'm building a Thread network with a sensor that sends data to Home Assistant over MQTT. Everything worked with Wi-Fi and IPv4. After setting up OpenThread I changed the broker address in the code to IPv6.

I'm using ESP-IDF v6.1 (ESP32-C6) and the code is based on the examples. My configuration looked like this:

```c
#define EXAMPLE_MQTT_BROKER_URI "mqtt://[fdb0:dcec:f6d1:0:944f:4848:99e1:660a]:1883"

const esp_mqtt_client_config_t mqtt_cfg = {
    .broker = {
        .address.uri = EXAMPLE_MQTT_BROKER_URI,
    },
};
```

When trying to connect, this is what I got:

* Thread started and the ping reply was fine:

```
esp32c6> ot ping fdb0:dcec:f6d1:0:944f:4848:99e1:660a
16 bytes from fdb0:dcec:f6d1:0:944f:4848:99e1:660a: icmp_seq=2 hlim=64 time=43ms
1 packets transmitted, 1 packets received. Packet loss = 0.0%. Round-trip min/avg/max = 43/43.000/43 ms.
Done
```

* but the MQTT client did not connect:

```
E (285305) esp-tls: couldn't get hostname for :fdb0:dcec:f6d1:0:944f:4848:99e1:660a: getaddrinfo() returns 202, addrinfo=0
E (285305) transport_base: Failed to open a new connection: 32769
E (285305) mqtt_client: Error transport connect
I (285305) ot_esp_cli: MQTT_EVENT_ERROR
I (285305) ot_esp_cli: Last error code reported from esp-tls: 0x8001
I (285305) ot_esp_cli: Last tls stack error number: 0x0
I (285305) ot_esp_cli: Last captured errno : 118 (Host is unreachable)
I (285305) ot_esp_cli: MQTT_EVENT_DISCONNECTED
```

After a long analysis with the help of Claude AI, I added the following fragment to the esp-tls component, in esp_tls.c (the first and last lines are from the original file, lines 246 and 247 in my copy):

```c
ESP_LOGD(TAG, "host:%s: strlen %lu", use_host, (unsigned long)hostlen);

/* an IPv6 literal should not go through DNS (e.g. Thread without IPv4/DNS) */
struct in6_addr tmp6;
if (inet_pton(AF_INET6, use_host, &tmp6) == 1) {
    hints.ai_family = AF_INET6;
    hints.ai_flags |= AI_NUMERICHOST;
}

int res = getaddrinfo(use_host, NULL, &hints, &address_info);
```

After this change the client connected:

```
I (53305) mqtt_client: Host address [fdb0:dcec:f6d1:0:944f:4848:99e1:660a]
I (53525) ot_esp_cli: MQTT_EVENT_CONNECTED
I (63685) ot_esp_cli: temperature read from DS18B20[0]: 26.38C
I (63685) ot_esp_cli: Temperature C Sensor published properly...
I (63695) ot_esp_cli: Published state: {"temp2":26.38} to topic: home/sslab_sek_00_dev_01/state
I (63785) ot_esp_cli: MQTT_EVENT_PUBLISHED, msg_id=63851
I (63845) ot_esp_cli: MQTT_EVENT_PUBLISHED, msg_id=42220
```

The above is mine, and the below is from Claude:

Additional findings: the problem is not in esp-mqtt or esp-tls itself. Calling `getaddrinfo()` directly with the same literal returns 202 both with `AF_UNSPEC` and `AF_INET6`, but returns 0 when `AI_NUMERICHOST` is set. A raw socket (`inet6_aton()` + `connect()`) to the same address works. The device has no IPv4 address and no DNS server (Thread only); the same code works on Wi-Fi with IPv4. I did not find out which `EAI_FAIL` return in lwIP's `netdb.c` is responsible, so the patch above is only a workaround.

mickeyreg, 2026-10-28:
Thanks for a hint. I found the source of the problem myself and disabled in menuconfig:
  _Component config → OpenThread → Thread Core Features → Enable DNS64 Client [set: OFF]_
and IPv6 connection started to work :)