espressif/esp-idf #15227: ESP32 S3 TINYUSB 2个CDD 1个MSC 加载失败

qlwz, 2025-01-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4.0

### Espressif SoC revision.

ESP-ROM:esp32s3-20210327

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-box-lite

### Power Supply used.

USB

### What is the expected behavior?

支持2个CDC和1个MSC

### What is the actual behavior?

2个CDC和1个MSC win无法加载。
2个CDC win正常使用
1个CDC 1个MSC  win正常使用

### Steps to reproduce.

1. 复制tusb_composite_msc_serialdevice示例
2. 修改sdkconfig.defaults -> CONFIG_TINYUSB_CDC_COUNT=2
3. 编译及烧录

### Debug Logs.

```plain
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4 2nd stage bootloader
I (27) boot: compile time Jan 17 2025 15:22:03
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 16MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00005000
I (65) boot:  1 otadata          OTA data         01 00 0000e000 00002000
I (71) boot:  2 app0             OTA app          00 10 00010000 00270000
I (78) boot:  3 app1             OTA app          00 11 00280000 00270000
I (84) boot:  4 spiffs           Unknown data     01 82 004f0000 00019000
I (91) boot:  5 coredump         Unknown data     01 03 00509000 00010000
I (97) boot:  6 custom_data      unknown          40 00 00519000 00001000
I (104) boot:  7 storage          Unknown data     01 81 0051a000 00ae6000
I (110) boot: End of partition table
I (114) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ca10h ( 51728) map
I (130) esp_image: segment 1: paddr=0001ca38 vaddr=3fc92b00 size=02f50h ( 12112) load
I (133) esp_image: segment 2: paddr=0001f990 vaddr=40374000 size=00688h (  1672) load
I (137) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2b694h (177812) map
I (175) esp_image: segment 4: paddr=0004b6bc vaddr=40374688 size=0e42ch ( 58412) load
I (193) esp_image: segment 5: paddr=00059af0 vaddr=600fe100 size=0001ch (    28) load
I (199) boot: Loaded app from partition at offset 0x10000
I (199) boot: Disabling RNG early entropy source...
I (211) cpu_start: Multicore app
I (220) cpu_start: Pro cpu start user code
I (220) cpu_start: cpu freq: 160000000 Hz
I (220) app_init: Application information:
I (220) app_init: Project name:     tusb_composite
I (224) app_init: App version:      1
I (228) app_init: Compile time:     Jan 17 2025 15:21:43
I (233) app_init: ELF file SHA256:  77d427415...
I (237) app_init: ESP-IDF:          v5.4
I (241) efuse_init: Min chip rev:     v0.0
I (245) efuse_init: Max chip rev:     v0.99 
I (249) efuse_init: Chip rev:         v0.1
I (253) heap_init: Initializing. RAM available for dynamic allocation:
I (259) heap_init: At 3FC97288 len 00052488 (329 KiB): RAM
I (264) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (269) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (274) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (281) spi_flash: detected chip: gd
I (283) spi_flash: flash io: dio
I (286) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (292) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (299) main_task: Started on CPU0
I (339) main_task: Calling app_main()
I (339) example_main: Initializing storage...
I (339) example_main: Initializing wear levelling
I (349) example_main: Reading file
I (349) example_main: Read from file: 'Hello World!'
I (349) example_main: USB Composite initialization
W (349) tusb_desc: No Device descriptor provided, using default.
W (359) tusb_desc: No FullSpeed configuration descriptor provided, using default.
W (369) tusb_desc: No String descriptors provided, using default.
I (369) tusb_desc: 
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 239         │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 2           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 1           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4001      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (539) TinyUSB: TinyUSB Driver installed
I (539) example_main: USB Composite initialization DONE
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
```

peter-marcisovsky, 2025-01-17:
Hi @qlwz  

If I got the translation correctly, you are trying to use `tusb_composite_msc_serialdevice` example.

It works fine with `1 CDC` and `1 MSC` interfaces.
But it does not work with `2 CDC` and `1 MSC` interfaces.

It this is your case, then you have hit a HW lkimitation of esp32s3, regarding the number of Endpoints. 
Please refer to the [docs](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_device.html#features) about number of endpoints avaliable for esp32s3

The MSC device occupies 2 Endpoints, CDC interface occupies 3 Endpoints. So with the configuration 2 CDC and 1 MSC you are already over the limit. There should be some error log in the console output about the number of endpoints. We will check.

roma-jam, 2025-01-17:
Hi @qlwz 

What version of tinyusb component do you use? 

You can check it from the build log, it should be something like: 

```
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/esp_tinyusb (1.6.0)
NOTICE: [2/3] espressif/tinyusb (0.17.0~2)
NOTICE: [3/3] idf (5.5.0)
```

qlwz, 2025-01-17:
NOTICE: Processing 3 dependencies:
NOTICE: [1/3] espressif/esp_tinyusb (1.6.0)
NOTICE: [2/3] espressif/tinyusb (0.17.0~2)
NOTICE: [3/3] idf (5.4.0)

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0xa (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fce2810,len:0x15a0
load:0x403c8700,len:0x4
load:0x403c8704,len:0xd20
load:0x403cb700,len:0x2ee4
entry 0x403c8928
I (27) boot: ESP-IDF v5.4 2nd stage bootloader
I (27) boot: compile time Jan 17 2025 19:28:33
I (27) boot: Multicore bootloader
I (27) boot: chip revision: v0.1
I (30) boot: efuse block revision: v1.2
I (33) boot.esp32s3: Boot SPI Speed : 80MHz
I (37) boot.esp32s3: SPI Mode       : DIO
I (41) boot.esp32s3: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (49) boot: Partition Table:
I (52) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (71) boot:  2 factory          factory app      00 00 00010000 00100000
I (78) boot:  3 storage          Unknown data     01 81 00110000 00100000
I (84) boot: End of partition table
I (87) esp_image: segment 0: paddr=00010020 vaddr=3c030020 size=0ca10h ( 51728) map
I (104) esp_image: segment 1: paddr=0001ca38 vaddr=3fc92b00 size=02f50h ( 12112) load
I (107) esp_image: segment 2: paddr=0001f990 vaddr=40374000 size=00688h (  1672) load
I (110) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=2b694h (177812) map
I (149) esp_image: segment 4: paddr=0004b6bc vaddr=40374688 size=0e42ch ( 58412) load
I (162) esp_image: segment 5: paddr=00059af0 vaddr=600fe100 size=0001ch (    28) load
I (168) boot: Loaded app from partition at offset 0x10000
I (168) boot: Disabling RNG early entropy source...
I (179) cpu_start: Multicore app
I (188) cpu_start: Pro cpu start user code
I (188) cpu_start: cpu freq: 160000000 Hz
I (189) app_init: Application information:
I (189) app_init: Project name:     tusb_composite
I (193) app_init: App version:      1
I (196) app_init: Compile time:     Jan 17 2025 19:28:13
I (202) app_init: ELF file SHA256:  266a84c46...
I (206) app_init: ESP-IDF:          v5.4
I (209) efuse_init: Min chip rev:     v0.0
I (213) efuse_init: Max chip rev:     v0.99 
I (217) efuse_init: Chip rev:         v0.1
I (221) heap_init: Initializing. RAM available for dynamic allocation:
I (228) heap_init: At 3FC97288 len 00052488 (329 KiB): RAM
I (233) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (238) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (243) heap_init: At 600FE11C len 00001ECC (7 KiB): RTCRAM
I (249) spi_flash: detected chip: gd
I (252) spi_flash: flash io: dio
W (254) spi_flash: Detected size(16384k) larger than the size in the binary image header(4096k). Using the size in the binary image header.
I (267) sleep_gpio: Configure to isolate all GPIO pins in sleep state
I (273) sleep_gpio: Enable automatic switching of GPIO sleep configuration
I (280) main_task: Started on CPU0
I (300) main_task: Calling app_main()
I (300) example_main: Initializing storage...
I (300) example_main: Initializing wear levelling
I (300) example_main: Reading file
I (310) example_main: Read from file: 'Hello World!'
I (310) example_main: USB Composite initialization
W (310) tusb_desc: No Device descriptor provided, using default.
W (320) tusb_desc: No FullSpeed configuration descriptor provided, using default.
W (330) tusb_desc: No String descriptors provided, using default.
I (330) tusb_desc: 
┌─────────────────────────────────┐
│  USB Device Descriptor Summary  │
├───────────────────┬─────────────┤
│bDeviceClass       │ 239         │
├───────────────────┼─────────────┤
│bDeviceSubClass    │ 2           │
├───────────────────┼─────────────┤
│bDeviceProtocol    │ 1           │
├───────────────────┼─────────────┤
│bMaxPacketSize0    │ 64          │
├───────────────────┼─────────────┤
│idVendor           │ 0x303a      │
├───────────────────┼─────────────┤
│idProduct          │ 0x4001      │
├───────────────────┼─────────────┤
│bcdDevice          │ 0x100       │
├───────────────────┼─────────────┤
│iManufacturer      │ 0x1         │
├───────────────────┼─────────────┤
│iProduct           │ 0x2         │
├───────────────────┼─────────────┤
│iSerialNumber      │ 0x3         │
├───────────────────┼─────────────┤
│bNumConfigurations │ 0x1         │
└───────────────────┴─────────────┘
I (500) TinyUSB: TinyUSB Driver installed
I (500) example_main: USB Composite initialization DONE
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED
dfifo_alloc 196: ASSERT FAILED
dcd_edpt_open 533: ASSERT FAILED
usbd_open_edpt_pair 1256: ASSERT FAILED
mscd_open 250: ASSERT FAILED
process_set_config 1064: ASSERT FAILED
process_control_request 804: ASSERT FAILED

qlwz, 2025-01-17:
> Hi [@qlwz](https://github.com/qlwz)
>
> If I got the translation correctly, you are trying to use `tusb_composite_msc_serialdevice` example.
>
> It works fine with `1 CDC` and `1 MSC` interfaces. But it [...]

You're right, I test  1 CDC and 1 MSC and1 HID OK ，  3 Endpoints + 2 Endpoints + 1 Endpoints = 6 Endpoints

closed 2025-01-17 by qlwz