espressif/esp-idf #15228: GDMA buffers init fail when flash encryption enabled

Zer0-bit, 2025-01-17:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.0+sha.cfea4f7c98

### Espressif SoC revision.

ESP32-S3 (N16R8)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

Sunton 4.3 RGB LCD

### Power Supply used.

USB

### What is the expected behavior?

Hi, i've been trying to drive a 800x480 ESP32-S3 LCD module and all seems to be working well with with no encryption, GDMA buffers are initialising no issues and everything works well.

```#ifdef LCD_NATIVE
#pragma once

#include <TAMC_GT911.h>
#include "esp_err.h"
#include "esp_log.h"
#include "esp_lcd_panel_ops.h"
#include "esp_lcd_panel_rgb.h"
#include "esp_lcd_panel_io.h"

#include "driver/spi_master.h"
#include "driver/i2c.h"
#include "driver/gpio.h"
#include "driver/touch_sensor.h"
#include "esp_private/gdma_link.h"
#include "esp_private/gdma.h"
#include "esp_private/gdma_link.h"

#define LCD_H_RES 800
#define LCD_V_RES 480

//
// min / max determined by printing x/y coordinates while pressing different locations on the display
// can be reproduced by printing the numbers in process_coordinates before calling the map function
//

//
// Serious distortion going above 18Mhz:
//
#define LCD_PIXEL_CLOCK_HZ     (12 * 1000000)

//
[... 198 lines cut ...]
file: "frontend-controls/embedded-ui/driver/esp_drv.h" line 157
func: esp_err_t EspNativeDisplayController::initializeLCD(lv_disp_drv_t*)
expression: esp_lcd_new_rgb_panel(&panel_config, &panel_handle)

abort() was called at PC 0x40382fdf on core 1

Backtrace: 0x40377316:0x3fcce310 0x40382fe9:0x3fcce330 0x40389ccd:0x3fcce350 0x40382fdf:0x3fcce3d0 0x4202fead:0x3fcce400 0x4202ffb7:0x3fcce4e0 0x42030087:0x3fcce500 0x42030d34:0x3fcce530 0x42099ec3:0x3fcce550 0x403840e6:0x3fcce570
  #0  0x40377316 in panic_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/panic.c:463
  #1  0x40382fe9 in esp_system_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x40389ccd in abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/newlib/abort.c:38
  #3  0x40382fdf in _esp_error_check_failed at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp_system/esp_err.c:49
  #4  0x4202fead in EspNativeDisplayController::initializeLCD(_lv_disp_drv_t*) at frontend-controls/embedded-ui/driver/esp_drv.h:157 (discriminator 2)
  #5  0x4202ffb7 in lcd_init() at frontend-controls/embedded-ui/ui.cpp:149
  #6  0x42030087 in lvglPinnedCoreTask() at frontend-controls/embedded-ui/ui.cpp:309
  #7  0x42030d34 in setup() at frontend-controls/main.cpp:202
  #8  0x42099ec3 in loopTask(void*) at ./platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:59
  #9  0x403840e6 in vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/port.c:139

ELF file SHA256: 0c7f495ca

E (1676) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
E (1684) esp_core_dump_elf: Elf write init failed!
E (1688) esp_core_dump_common: Core dump write failed with error=-1
Rebooting...
```

suda-morris, 2025-01-20:
Hi @Zer0-bit Thanks for the report. This indeed is a bug.

I think currently a quick workaround is to change value of the `LCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE`  to 4032

https://github.com/espressif/esp-idf/blob/adf5319639b7e66e4f80868b14fee1acb73672c7/components/esp_lcd/priv_include/esp_lcd_common.h#L36

Can you have a try?

Zer0-bit, 2025-01-20:
> Hi [@Zer0-bit](https://github.com/Zer0-bit) Thanks for the report. This indeed is a bug.
> 
> I think currently a quick workaround is to change value of the `LCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE` to 4032
> 
> [esp-idf/components/esp_lcd/priv_include/esp_lcd_common.h](https://github.com/espressif/esp-idf/blob/adf5319639b7e66e4f80868b14fee1acb73672c7/components/esp_lcd/priv_include/esp_lcd_common.h#L36)
> 
> Line 36 in [adf5319](/espressif/esp-idf/commit/adf5319639b7e66e4f80868b14fee1acb73672c7)
> 
>  #define LCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE 4095 
> Can you have a try?

Hi @suda-morris - thank you for having a look into this, I should be able to get to this in 2 days and will write back the results.

Zer0-bit, 2025-01-23:
Hi @suda-morris - sadly since i'm using the arduino-espressif32 framework i have no direct access to the file you pointing to but i had set this as a platofrmio **build_flags** define which should in theory redefine that with the new value, see below:
```
build_flags =
	-DCONFIG_GDMA_CTRL_FUNC_IN_IRAM=1
	-DCONFIG_GDMA_ISR_IRAM_SAFE=1
	-DLCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE=4032
```
And sadly this has not yielded any positive outcome, still getting the **_no more space for buffer mounting_** message, ofc all of this only happens when encryption is enabled, i keep double checking against non-encrypted hw and it works no issues there.
Pls let me know whether i can do anything else to try and debug this issue.

Zer0-bit, 2025-01-24:
On another note i tried to dance around a bit with setting different panel params and if i set ` .bounce_buffer_size_px = (3 * LCD_H_RES),` or to any other value really it keeps throwing the "buffer not aligned to 16" error, i ofc tried setting values that are aligned to 16 with no success, again not an issue at all on the non encrypted variant.

suda-morris, 2025-02-07:
Hi @Zer0-bit 

> Hi [@suda-morris](https://github.com/suda-morris) - sadly since i'm using the arduino-espressif32 framework i have no direct access to the file you pointing to but i had set this as a platofrmio [...]

It doesn't work because arduino library has pre-compiled the esp-idf into an library, so the `-DDLCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE` doesn take effect actually.

Zer0-bit, 2025-02-08:
> Hi [@Zer0-bit](https://github.com/Zer0-bit)
>
>
> It doesn't work because arduino library has pre-compiled the esp-idf into an library, so the `-DDLCD_DMA_DESCRIPTOR_BUFFER_MAX_SIZE` doesn take effect [...]

Thank you for looking into this @suda-morris, just so I understand is there a possibility of  you providing a branch if the arduino-espressiff32 with this define set to the value you mention so I can test or maybe there's already a branch in the works where this is included?

Zer0-bit, 2025-02-10:
> Hi [@Zer0-bit](https://github.com/Zer0-bit) We already merged a fix to the master branch and will backport to esp-idf v5.3 soon, we will make sure it can be included in the next arduino release.

Thank you for the quick actions, much appreciated!

suda-morris, 2025-02-10:
Here is a quick patch:

[0001-fix-dma-also-consider-buffer-alignment-when-calculat.patch.txt](https://github.com/user-attachments/files/18731544/0001-fix-dma-also-consider-buffer-alignment-when-calculat.patch.txt)

BatuhanKaratas, 2025-02-13:
Hi,

**Environment:**

**IDF Version:**
 ESP-IDF: v5.5.0 (master branch - 0d6099e)

**Espressif SoC Revision:**
ESP32-S3 (N16R8)

**Development Environment:**
VS Code IDE with ESP-IDF Extension (1.8.1)

**Security Features Configuration:**

```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_V2_RSA_SUPPORTED=y
CONFIG_SECURE_BOOT_V2_PREFERRED=y
CONFIG_SECURE_BOOT_V2_RSA_ENABLED=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="secure_boot_signing_key.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
# CONFIG_SECURE_BOOT_V2_ALLOW_EFUSE_RD_DIS is not set
CONFIG_SECURE_BOOT_FLASH_BOOTLOADER_DEFAULT=y
CONFIG_SECURE_BOOT_INSECURE=y
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_AES128=y
# CONFIG_SECURE_FLASH_ENCRYPTION_AES256 is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT=y
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE is not set
CONFIG_SECURE_FLASH_HAS_WRITE_PROTECTION_CACHE=y

#
# Potentially insecure options
#
# CONFIG_SECURE_BOOT_ALLOW_JTAG is not set
CONFIG_SECURE_BOOT_ALLOW_SHORT_APP_PARTITION=y
CONFIG_SECURE_BOOT_ALLOW_UNUSED_DIGEST_SLOTS=y
CONFIG_SECURE_FLASH_UART_BOOTLOADER_ALLOW_ENC=y
# CONFIG_SECURE_FLASH_UART_BOOTLOADER_ALLOW_CACHE is not set
# CONFIG_SECURE_FLASH_REQUIRE_ALREADY_ENABLED is not set
CONFIG_SECURE_FLASH_SKIP_WRITE_PROTECTION_CACHE=y
# end of Potentially insecure options

CONFIG_SECURE_FLASH_ENCRYPT_ONLY_IMAGE_LEN_IN_APP_PART=y
CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
# CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE is not set
CONFIG_SECURE_INSECURE_ALLOW_DL_MODE=y
# end of Security features
```

When "CONFIG_SECURE_FLASH_ENC_ENABLED" is enabled and in the Efuse table "SPI_BOOT_CRYPT_CNT (BLOCK0)" is set to Enable R/W (0b111), I encounter the same error described by @Zer0-bit when trying to initialize the RGB panel, as shown in the logs below:

```
I (00:00:01.843) LVGL_MANAGER: Initialize RGB panel
E (00:00:01.888) gdma: gdma_link_mount_buffers(169): buffer not aligned to 16
E (00:00:01.888) lcd_panel.rgb: lcd_rgb_panel_init_trans_link(957): mount DMA restart buffer failed
E (00:00:01.894) lcd_panel.rgb: esp_lcd_new_rgb_panel(346): init DMA link failed
E (00:00:01.903) LVGL_MANAGER: LCD RGB panel construction failed.
E (00:00:01.909) lcd_panel: esp_lcd_panel_del(28): invalid panel handle
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x4201a723
file: "./inf_components/inf_lvgl/src/LVGLManager.cpp" line 146
func: void inf::inf_component::LVGLManager::init()
expression: lcdInit()

abort() was called at PC 0x4038c5d3 on core 1

```
Although my IDF version includes your fix([cfca6c6](https://github.com/espressif/esp-idf/commit/cfca6c606e256f38888e94c3e41ed19a4000f65d)), the issue still persists. Could you please re-evaluate this?

suda-morris, 2025-02-14:
Hi @BatuhanKaratas Can you do me a favor, if comment out this line can help?

https://github.com/espressif/esp-idf/blob/0d6099ec533c4b647fb7a7b0b8942bc7aeb82f90/components/esp_hw_support/dma/gdma.c#L398

Zer0-bit, 2025-02-14:
Just to confirm - i'm going to manually build the esp32-arduino-lib for the s3 variant with the fixes form master backported into 5.3 since that seems to be the only latest supported idf version for now by the arduino fw and test as well just need a bit more time since this is a bit more involved, would be great if the guys over in the esp32-arduino repo would push the trigger on one of their pipelines to build a test branch ofc but as of now haven't gotten any replies there so will have to go the longer route :)

BatuhanKaratas, 2025-02-14:
@sudomorris Commenting out the mentioned line resolved the issue. It appears that GDMA_LL_AHB_RX_BURST_NEEDS_ALIGNMENT is enabled on the ESP32-S3. In this case, assuming data_burst is enabled, would int_mem_alignment = MAX(int_mem_alignment, 4); be set to 4? Would this be the final solution in its current state?

Zer0-bit, 2025-02-14:
@suda-morris - sadly i have just tested with the latest fixes + commented out the line you mention above and i get the alignment error now as well.
```
E (815) gdma: gdma_link_mount_buffers(169): buffer not aligned to 16
E (816) lcd_panel.rgb: lcd_rgb_panel_init_trans_link(1086): mount DMA restart buffer failed
E (819) lcd_panel.rgb: esp_lcd_new_rgb_panel(350): init DMA link failed
ESP_ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x4203037d
  #0  0x4203037d in EspNativeDisplayController::initializeLCD(_lv_disp_drv_t*) at frontend-controls/embedded-ui/driver/esp_drv.h:163 (discriminator 2)

file: "Esp32ExceptionDecoder: failed to call C:\Users\victo\.platformio\packages\toolchain-xtensa-esp-elf\bin\xtensa-esp32s3-elf-addr2line.exe: Command '['C:\\Users\\victo\\.platformio\\packages\\toolchain-xtensa-esp-elf\\bin\\xtensa-esp32s3-elf-addr2line.exe', '-fipC', '-e', 'C:\\GIT\\gaggiuino-merge\\.pio\\build\\frontend-controls-upgrade\\firmware.elf', '0x4203037d']' returned non-zero exit status 3221225786.
frontend-controls/embedded-ui/driver/esp_drv.h" line 163
func: esp_err_t EspNativeDisplayController::initializeLCD(lv_disp_drv_t*)
expression: esp_lcd_new_rgb_panel(&panel_config, &panel_handle)

abort() was called at PC 0x4038331f on core 1

Backtrace: 0x403773b2:0x3fcce930 0x40383329:0x3fcce950 0x4038a0ed:0x3fcce970 0x4038331f:0x3fcce9f0 0x4203037d:0x3fccea20 0x42030487:0x3fcceb00 0x42030557:0x3fcceb20 0x42031384:0x3fcceb50 0x4209b073:0x3fcceb70 0x40383fc6:0x3fcceb90
  #0  0x403773b2 in panic_abort at /home/ubuntu/esp32-arduino-lib-builder/build/../esp-idf/components/esp_system/panic.c:463
  #1  0x40383329 in esp_system_abort at /home/ubuntu/esp32-arduino-lib-builder/build/../esp-idf/components/esp_system/port/esp_system_chip.c:92
  #2  0x4038a0ed in abort at /home/ubuntu/esp32-arduino-lib-builder/build/../esp-idf/components/newlib/abort.c:38
  #3  0x4038331f in _esp_error_check_failed at /home/ubuntu/esp32-arduino-lib-builder/build/../esp-idf/components/esp_system/esp_err.c:49
```

suda-morris, 2025-02-17:
> Commenting out the mentioned line resolved the issue. It appears that GDMA_LL_AHB_RX_BURST_NEEDS_ALIGNMENT is enabled on the ESP32-S3. In this case, assuming data_burst is enabled, would [...]

Thanks @BatuhanKaratas LCD only uses the "TX Path" of the DMA engine. So this `int_mem_alignment = MAX(int_mem_alignment, 4);` won't apply. 

I think the root cause is, when external memory encryption is enabled, DMA need a more strict alignment for the buffer.

suda-morris, 2025-02-17:
Hi @Zer0-bit looks like you're facing the same issue as @BatuhanKaratas does. I'm sorry. Have you correctly comment out the line I mentioned in https://github.com/espressif/esp-idf/issues/15228#issuecomment-2658159912? 

Could you help print the value of `int_mem_alignment` at the end of the `gdma_config_transfer` for sure?

Zer0-bit, 2025-02-17:
Hi @suda-morris - yeah can confirm the alignment error I get now is with the line mentioned in #15228 comment commented out.
I'll print out the `int_mem_alignment` and come back with the results.

Zer0-bit, 2025-02-19:
Hi @suda-morris and @simeonAT - sorry for being frank but I'm not sure what the resolution is on this ticket as given I'm the original creator with the original problem the commit that closes this ticket for the second time still doesn't solve the issue for me, maybe I did miss smth in which case sorry and pls let me know what did I miss if possible 😅

suda-morris, 2025-02-20:
Hi @Zer0-bit Now I realized you're not using the "bounce buffer" mode, so the fix workd for @BatuhanKaratas but not for you. 

Can you comment out this line and try again?

https://github.com/espressif/esp-idf/blob/c71d74e2f853b1135c63f47e349f2a08f63f3e01/components/esp_hw_support/dma/gdma.c#L397

Zer0-bit, 2025-02-20:
Hi @suda-morris - I'm actually using the bounce buffer mode, if you see my first post the config looks like this:
```
     .data_width = 16,
     .num_fbs = 2,
     .hsync_gpio_num = LCD_PIN_HSYNC,
     .vsync_gpio_num = LCD_PIN_VSYNC,
     .de_gpio_num = LCD_PIN_DE,
     .pclk_gpio_num = LCD_PIN_PCLK,
     .disp_gpio_num = LCD_PIN_DISP_EN,
     .data_gpio_nums = {
         LCD_PIN_DATA0, LCD_PIN_DATA1, LCD_PIN_DATA2, LCD_PIN_DATA3, LCD_PIN_DATA4, LCD_PIN_DATA5, LCD_PIN_DATA6, LCD_PIN_DATA7,
         LCD_PIN_DATA8, LCD_PIN_DATA9, LCD_PIN_DATA10, LCD_PIN_DATA11, LCD_PIN_DATA12, LCD_PIN_DATA13, LCD_PIN_DATA14, LCD_PIN_DATA15
     },
     .flags = {
      .fb_in_psram = true
     }
    };
```
Was under the impression that `.num_fbs = 2` is what enables this mode or am I mistaken and there's more actions to do?

Zer0-bit, 2025-02-21:
Hi @suda-morris - good news is that commenting out the line you mention [in this message](https://github.com/espressif/esp-idf/issues/15228#issuecomment-2670235828) did bring back the image, tried with one and two frame buffers and the image is there, though there's quite a lot of frame shift towards the right, attached video to show.

https://github.com/user-attachments/assets/6fad84db-6061-4452-b0c2-3f7919f7e0f7

Alvin1Zhang, 2025-03-06:
Thanks for reporting, fix on release/5.3 is available at https://github.com/espressif/esp-idf/commit/6a1cf6bb85b138e0f4a119ee3a52b893d01ed12a, feel free to reopen if the issue still happens. Thanks.

closed 2025-03-06 by Alvin1Zhang

lbxhacker, 2025-03-14:
This problem still exists on ESP-IDF5.3.2. In my use case :i80_controller, this problem will occur when flash encryption in sdkconfig is enabled. Even in the development mode, I have enabled the encryption function in IDF4.4.2, which is the same environment, while IDF4. 4.2 This does not happen

lbxhacker, 2025-03-14:
Even if you comment out the ext and int lines, the result is still inaccurate and the screen refresh is very abnormal

lbxhacker, 2025-03-14:
Oh no, my problem has been solved now. The main reason why I need to annotate ext and int is that I modified esp_lcd_i80_bus_config_t bus_config = {... The code is now restored and the problem has been resolved

owen63, 2025-05-09:
@suda-morris I have found the same problem. My setup:
esp32s3 
esp-idf v5.4.1
encrypted flash
double buffers in PSRAM, no bounce buffers

The suggested fix to comment out this code fixes my problem:
`ext_mem_alignment = MAX(ext_mem_alignment, GDMA_ACCESS_ENCRYPTION_MEM_ALIGNMENT); `

Can you advise on the fix status?