espressif/esp-idf #15230: strptime not working as expected

Ruubah, 2025-01-18:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Info:
ESP32-C6
ESP-IDF v5.5 (master branch)
`idf.py --version
ESP-IDF v5.5-dev-847-gcb3ac7429c-dirty`

I'm trying to read a string in format "(YYMMDDhhmmssX)"
`
#include <time.h>
#include <stdio.h>
char* X = NULL;    
struct tm tm;

const char* str_data = "(250111163740W)";
X = strptime(str_data , "(%y%m%d%H%M%S", &tm);
ESP_LOGI(TEST_TAG, "X is: %p\r\n", X);
`

`
TEST: X is: 0x0
`

X is still NULL (0x0) but it should point to the character W which it doesn't.
Am I miss interpreting how strptime should work?

Tried just compiling a program with gcc in Ubuntu.

`
#define _GNU_SOURCE
#include <stdio.h>
#include <time.h>

int main()
{
    struct tm tm;
    char *X = NULL;

    const char* str_data = "(250111163740W)";
    X = strptime(str_data, "(%y%m%d%H%M%S", &tm);

    printf("X is: %c, X value: %p\r\n", *X, X);

    return 0;
}
`

`
X is: W, X value: 0x555ddc8d3011
`

Have a hard time understanding if it is possible to see the strptime implementation used in esp-idf or if it's just strptime in the newlib that is used without changes.

SoucheSouche, 2025-01-20:
Hello @Ruubah, thanks for reporting this issue. I will take a look at it and come back to you when I find out what is causing this.

Lapshin, 2025-01-20:
Hi @Ruubah , it uses Newlib's `strptime` implementation without modifications.

The problem with the code you posted is that it does not follow `strptime` specification.

The key points:

> There should be whitespace or other alphanumeric
>       characters between any two field descriptors.

And

> If strptime() fails to match all of the format string and
>       therefore an error occurred, the function returns NULL.

So, you have an invalid `str_data` to pass into `strptime`.

As for me, it's better to know if parsing failed than to have unexpected results as in your Ubuntu's example

Ruubah, 2025-01-20:
@Lapshin, thanks. That explains it, missed that. Guess the test with gcc in Ubuntu threw me off a little. 

With the answer I can't understand how I really missed it, because I know that I wondered why it wasn't possible to specify how many characters per format like for e.g %2H, %HH, or %H%H.

And we agree on unexpected results, it should return NULL in this case.

closed 2025-01-20 by Ruubah