espressif/esp-idf #15233: A/D continuous parameter calculation

zamek42, 2025-01-19:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi All, 
I need help to understand continuous ad configuration of ESP IDF. 

I want to digitize 6 analog inputs with esp idf. Each channel should take a sample for 20ms in every 100us, it is 200 samples per channel.
The calculation process:
1. frame size: number of channels * number of samples * size of adc_digi_output_data_t. ( 6 * 200 * 2 = 2400 )
2. sample frequency:  10000 samples/seconds * number of channels ( 10000 * 6 = 60Khz )
I used the adc continuous sample application for the test.
3. I set the dma buffer size to 2 times the frame_size.
4. I turned on the AD log level to DEBUG, but I didn't see any log messages from ad driver.
5. I set the s_on_pool_ovf function to my own implementation, which only incremented a variable if a buffer overflow occurred.

In ad_init I print the ad parameters:
`     I (680) ad: CHANNEL_NUMS:6, SAMPLES_PER_CHANNEL:200, sizeof(adc_digi_output_data_t):2, FRAME_SIZE:2400, sample freq:60000  `

I change a gpio pin in the ad DMA interrupt, so I can check the interrupt frequency with an oscilloscope, which is strangely 25ms instead of the desired 20ms.

I haven't processed the incoming data yet, I just calculated the number of data per channel in an array in each cycle.
The result of this was very strange (elen is the read_len from adc_continuous_read ):
`
       A/D cycles:3697, timeouts:0, size errors:0, invalid channels:0, ovf:0, elen:2400
`
`
       ch0:1400 ch1:199 ch2:199 ch3:200 ch4:201 ch5:201
`

In total, there are 2400 samples, but 1400 of them are on the first channel, while the other channels each have 200 samples.
When I tried to set frame size to number of samples instead of bytes, I get the similar results, ch0 is 7 times larger than others:

` I (679) ad: CHANNEL_NUMS:6, SAMPLES_PER_CHANNEL:200, sizeof(adc_digi_output_data_t):2, FRAME_SIZE:1200, sample freq:60000 `
`     

    A/D cycles:3911, timeouts:0, size errors:0, invalid channels:0, ovf:0, elen:1200 `
`
    ch0:700 ch1:99 ch2:99 ch3:100 ch4:101 ch5:101 
`
here is the code: 

`

      #include <stdbool.h>
      #include <stdint.h>
      #include <string.h>
      #include <stdio.h>
      #include <strings.h>
      #include <sys/stat.h>
      #include "esp_attr.h"
      #include "esp_err.h"
      #include "esp_console.h"
      #include "argtable3/argtable3.h"
      #include "freertos/projdefs.h"
      #include "hal/adc_types.h"
      #include "portmacro.h"
      #include "esp_log.h"
      #include "freertos/task.h"
      #include "freertos/semphr.h"
      #include "esp_adc/adc_continuous.h"
      #include "driver/gpio.h"
      #include "ad.h"

      #undef LOG_LOCAL_LEVEL
      #define LOG_LOCAL_LEVEL ESP_LOG_INFO
[... 186 lines cut ...]
                    static int fn_cmd(int argc, char **argv) {
                      int err=arg_parse(argc, argv, (void **) &args);
                      if (err!=0) {
                        help();
                        return 0;
                      }  
                      
                      print_stat();  
                      return 0;
                    }
                      
                      static void register_cmd() {
                        args.phase = arg_int0("pP", "phase", "<n>", "phase index 1 up to 3");
                        args.end=arg_end(0);
                        const esp_console_cmd_t cmd = {
                          .command=CMD,
                          .help="",
                          .hint="",
                          .func=&fn_cmd
                          };
                        
                        ESP_ERROR_CHECK(esp_console_cmd_register(&cmd));  
                        }

`

zamek42, 2025-01-19:
couple of hours later: when I don't use ADC_CHANNEL_0 it works well. Seems to be this channel has some special priority, or is it bug?

FrankBJensen, 2025-01-22:
ESP32-S3-N16R8V

zamek42, 2025-01-22:
Hi @Bruce297, I have an ESP-32S devkit-C with 4Mb flash.

crivi30, 2025-03-26:
I have a related question on the ADC topic. Let me know if I need to open another thread.

### Question:

Based on @zamek42's code and official doc, it seems that in `.conv_frame_size`, we need to consider how many channels we are using:

```c
#define FRAME_SIZE ((CHANNEL_NUMS) * (SAMPLES_PER_CHANNEL) * sizeof(adc_digi_output_data_t))
```

However, in the ADC library (specifically in `adc_continuous.c`), I noticed that memory is allocated as follows (conv_frame_size*num channels):

```c
adc_ctx->rx_dma_buf = heap_caps_calloc(INTERNAL_BUF_NUM, hdl_config->conv_frame_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA | MALLOC_CAP_8BIT);
```

This implies that the correct `FRAME_SIZE` definition should be:

```c
#define FRAME_SIZE ((SAMPLES_PER_CHANNEL) * sizeof(adc_digi_output_data_t))
```

without multiplying by the number of channels in advance. Is this correct?

### Additional Observation:

Following @zamek42's code, the memory allocated for the conversion frame appears to be larger than the max buffer size because it is **not** multiplied by the number of channels:

```c
adc_ctx->ringbuf_storage = heap_caps_calloc(1, hdl_config->max_store_buf_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_8BIT);
adc_ctx->rx_dma_buf = heap_caps_calloc(INTERNAL_BUF_NUM, hdl_config->conv_frame_size, MALLOC_CAP_INTERNAL | MALLOC_CAP_DMA | MALLOC_CAP_8BIT);
```

Does anyone know the reason behind this allocation?
Thank you in advance!