espressif/esp-idf #15237: ADC in continous mode STOPS when using FATFS

FrankBJensen, 2025-01-20:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.0

### Espressif SoC revision.

ESP32-S3-N16R8V

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

External 3.3V

### What is the expected behavior?

That the ADC keeps sampling continously

### What is the actual behavior?

The ADC stops sampling with an TIMEOUT error, when I use FatFS to store a file.

### Steps to reproduce.

1. I have this task running, sampling from 4 inputs in continous mode. I never stops the task, it is free running, and I just grab data, when I need it.

```
MyADs::MyADs() {

    setStamData(STAT_ADS, "", 0);
    setTaskInfo(TASK_ADS, ADS_TASK, ADS_CORE);

    uint16_t vars = sizeof(reglist_ads)/sizeof(regStruct_ads);
        GVs->ZNZ        = (MBUS*)heap_caps_malloc(sizeof(MBUS[vars]), MALLOC_CAP_SPIRAM);
    if (GVs->ZNZ != NULL) ESP_LOGW("MEM", "ZNZ initialized OK, size: %i, len: %i", heap_caps_get_allocated_size(GVs->ZNZ), sizeof(MBUS[vars]));

    std::fill(GVs->ZNZ, GVs->ZNZ + vars, MBUS());

    result = (uint8_t*)heap_caps_malloc(ADC_BUFFER_SIZE, MALLOC_CAP_8BIT | MALLOC_CAP_DMA); // MALLOC_CAP_SPIRAM);
    std::fill(result, result + ADC_BUFFER_SIZE, 0);

    adc_event_group = xEventGroupCreate();
    adc_progress = xEventGroupCreate();

    begin();
}

MyADs::~MyADs() {

    if (taskHandle != NULL) {
[... 420 lines cut ...]
fclose(inFile);
```

### Debug Logs.

```plain
There is no crash, but the FATFS runs once every hour, and then its like the ADC DMA simply stops

---- Everything running fine, calculating from ADC every second
I (550218) ads: Calculated: 0:-0.00 1:-0.00 2:-0.00
Save file: /fatfs/GRAPH-1737360000.bin
Save file: /fatfs/METER_GRID-1737360000.bin
Save file: /fatfs/METER_SUN-1737360000.bin
I (552088) kmq: Compare1: 49: DSP alarm code - 8400 0.00
E (552148) ads: Error ESP_ERR_TIMEOUT
E (552148) ads: TIMEOUT, LOGGING STOPPED!
```

### More Information.

I tried to separate the 2 tasks on different cores - does not help.
I have tried to increase the ADC buffers, and funny enough, when the buffer is real big, like 10K, sometimes it surveives the filewrite, but ADC still stops most of the times
There is no overrun of the buffer
There is no memory leaks or nothing
The task just keeps looping with ESP_ERR_TIMEOUT and do not recover

FrankBJensen, 2025-01-20:
Making this trick when there is a timeout, makes it run another hour, until FATFS is active again

```
adc_continuous_stop(adc_handle);
vTaskDelay(1);
adc_continuous_start(adc_handle);
```

safocl, 2025-01-20:
> gpio_isr_handler_add(IO_PW, gpio_isr_handler, (void *)IO_PW);

is this code correct? (`IO_PW` reinterpret to `void*`). If gpio_isr_handler uses this pointer, perhaps this is undefined behavior?

FrankBJensen, 2025-01-20:
Hi Safocl.

I only use this interupt to detect, if incomming is 50 or 60Hz. It does not interfear with the ADC at all, except I set the sampling frequency to 40 or 48 KHz when initializing, in order to have same number of samples for a period (200 samples).

```
void MyADs::gpio_isr_handler(void *arg)
{
    pulse_data.timestamp = esp_timer_get_time(); // uS
    if (pulse_data.timestamp - pulse_data.last_timestamp < 15000) return;

    pulse_data.count = pulse_data.count +1;
    
    pulse_data.pulse_width = pulse_data.timestamp - pulse_data.last_timestamp;
    pulse_data.hz = 1.0 / ((double)pulse_data.pulse_width * 1e-6);

    pulse_data.last_timestamp = pulse_data.timestamp;
}
```

This also do not explain, why the error occurs, exactly every whole hour, when I run the FATFS task. Also at 14:00, 15:00, 16:00 etc.

FrankBJensen, 2025-01-20:
Here is a standalone code, ready for you to compile and run. It replicates the fault. As soon as the fasfs is called, the ADC have a TIMEOUT, and do not recover

```
extern "C"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"

#include "nvs_flash.h"
#include "esp_system.h"
#include "esp_timer.h"

#include <cmath>
#include "driver/gpio.h"
#include "driver/gptimer.h"
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_timer.h"
#include "driver/timer.h"

#include <algorithm>
#include "esp_vfs_fat.h"

#define SCOP_BUF            210
#define SCOP_CYC              5
#define NUM_SAMPLES          50      // 100   195   450 svarer til 4.5 sekund, oversampling value
#define ADS_STACK          6000      //  3500
#define ADC_BUFFER_SIZE    2048      // 10240
#define ADC_FRAME_SIZE      256      //  2560

#define ADC_TASK_STOP_BIT       (1 << 0)
#define ADC_TASK_OVERFLOW_BIT   (1 << 1)
#define ADC_TASK_TIMEOUT_BIT    (1 << 2)
#define ADC_TASK_READY_BIT      (1 << 3)

static const char *TAGFS = "FATFS";
bool fatfs_Struct(void* data, size_t size, const char* name, long FileFimeStamp);

struct TESTFILE {
    char test[10000];
};

class MyADs {

    private:
        TaskHandle_t taskHandle = NULL;

        static void _task_adc(void* parameter);
        static void calculateRMS(uint16_t grafCount, uint16_t baseline);
        static void IRAM_ATTR gpio_isr_handler(void *arg);
        void begin();

    public:
        MyADs();
        ~MyADs();
};

typedef struct {
    float graf_average = 0;
    int sample_sum = 0;
    int count = 0;
} point_data_t;

typedef struct {
    point_data_t cp[SCOP_BUF] = { 0 };
    float baseline = 0;
[... 416 lines cut ...]
    sprintf(buf, "/fatfs/%s-%ld.bin", name, FileFimeStamp);
    printf("Save file: %s\n", buf);

    // Serialize the struct to a file
    FILE* outFile = fopen(buf, "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", buf);
        perror("Error"); // This will print a more detailed error
        return false;
    }

    size_t written = 0;
    while (written < size) {
        size_t to_write = (size - written) < 512 ? (size - written) : 512;
        fwrite(data + written, 1, to_write, outFile);
        written += to_write;
        vTaskDelay(1); // Delay for 1 millisecond
    }

//            fwrite(data, size, 1, outFile);
    fclose(outFile);

    return true;
}
```

safocl, 2025-01-20:
> bool startADC = true;

this variable is not changed from code. it always is `true`.
the following code is unreachable:
> if (ADs != NULL && !startADC) {
>                 delete ADs;
>                 ADs = NULL;
>             }

safocl, 2025-01-21:
[driver/timer.h](https://github.com/espressif/esp-idf/blob/master/components/driver/deprecated/driver/timer.h) is deprecated

safocl, 2025-01-21:
> fwrite(data + written, 1, to_write, outFile);

data is of type `void*` -- pointer arithmetic is not allowed.

safocl, 2025-01-21:
> EventBits_t bits = xEventGroupWaitBits(adc_event_group, ADC_TASK_STOP_BIT, pdFALSE, pdTRUE, 0);

this code wait `ADC_TASK_STOP_BIT` bits be activate and:
> [Potentially] block to wait for one or more bits to be set within a previously created event group.
but only ADC destructor set it...

safocl, 2025-01-21:
i did execute this [code](https://godbolt.org/z/s4vTqqTbq) (with some modifications) and it gave an [error](https://godbolt.org/z/KarY8KT7Y).

> E (7280) main: testfile don't allocated.

FrankBJensen, 2025-01-21:
You need to activate PSRAM in sdkconfig.

FrankBJensen, 2025-01-21:
I have simplified the code, the error persists.
Now I dont use PSRAM, so you should be able to run the code without PSRAM.

Log
```
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18851) ADC: Data OK
I (18861) ADC: Data OK
I (18861) ADC: Data OK
I (18861) ADC: Data OK
I (18861) ADC: Data OK
I (18861) ADC: Data OK
I (18861) ADC: Data OK
Save file: /fatfs/test.bin
I (18871) ADC: Data OK
I (18871) ADC: Data OK
I (18871) ADC: Data OK
I (18871) ADC: Data OK
I (18871) ADC: Data OK
I (18871) ADC: Data OK
E (23871) ADC: Error ESP_ERR_TIMEOUT
E (23871) ADC: TIMEOUT!
E (28871) ADC: Error ESP_ERR_TIMEOUT
E (28871) ADC: TIMEOUT!
```

code
```
extern "C"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"

#include "nvs_flash.h"
#include "esp_system.h"
#include "esp_timer.h"

#include <cmath>
#include "driver/gpio.h"
#include "driver/gptimer.h"
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_timer.h"
#include "driver/timer.h"

#include <algorithm>
#include "esp_vfs_fat.h"

#define SCOP_BUF            210
#define SCOP_CYC              5
#define NUM_SAMPLES          50      // 100   195   450 svarer til 4.5 sekund, oversampling value
#define ADS_STACK          6000      //  3500
#define ADC_BUFFER_SIZE    2048      // 10240
#define ADC_FRAME_SIZE      256      //  2560

#define ADC_TASK_STOP_BIT       (1 << 0)
#define ADC_TASK_OVERFLOW_BIT   (1 << 1)
#define ADC_TASK_TIMEOUT_BIT    (1 << 2)
#define ADC_TASK_READY_BIT      (1 << 3)

static const char *TAGFS = "FATFS";
[... 153 lines cut ...]
*/
                break; // stop med at hente data fra ADC
            }
        }
    }
}

bool fatfs_Struct(void* data, size_t size) {

    printf("Save file: /fatfs/test.bin\n");

    // Serialize the struct to a file
    FILE* outFile = fopen("/fatfs/test.bin", "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", "/fatfs/test.bin");
        perror("Error"); // This will print a more detailed error
        return false;
    }

    fwrite(data, size, 1, outFile);
    fclose(outFile);

    return true;
}
```

FrankBJensen, 2025-01-21:
Unless I made some stupid mistake, that I have overlooked, it seems to me, like FATFS is stopping the DMA and when the ADC buffer is empty, it goes into TIMEOUT, due to there is not comming data from the DMA anymore.

FrankBJensen, 2025-01-21:
Hmm, this is getting stranger and stranger.

- I have tried to set char test_file[10] in stead of 10000, this makes no difference, still same result with timeout. So filesize does not seem to matter.

- I have tried to increase ADC_BUFFER_SIZE to 20480 just to make an insane big buffer, still same result with timeout. So buffer size does not seem to matter.

- Now it gets interesting, messing with ADC_FRAME_SIZE. When i set this to 1500, still same result with timeout. But when I increase it to 2000 - IT WORKS. No timeout. So the size of the frame matters - WHY ?

- Then I tried to set back the char test_file[10000], then I am back, with same fault, the timeout.

- Then I increase the ADC_FRAME_SIZE to 3000, AND IT WORKS AGAIN.

SO what is the strange relationship between the filesize that I write with FATFS and the ADC_FRAME_SIZE ?? What is going on here??

AND not to forget. When I write, IT WORKS, I do not know, if it keeps working. Remember, in my original code, it works sometimes, and fails sometimes, even though both file size and frame_size if fixed. But in order to evaluate why this is, I need to understand, the relationship between FATFS and ADC.

Notes:
I have also upgraded to 5.3.1, made no difference.
I have also tried to pin the _task_adc to the other core, made no difference.

FrankBJensen, 2025-01-21:
OK, I can now confirm, that even though it works immediately after modifications with no timeout, it fails over time, and goes into a timeout loop.

safocl, 2025-01-21:
> if (test_file != NULL)

`if (test_file == NULL) PRINT_ERROR ` is better for program execution.

safocl, 2025-01-21:
> SO what is the strange relationship between the filesize that I write with FATFS and the ADC_FRAME_SIZE ?? What is going on here??

maybe there's a memory overflow going on here -- or an error that you don't catch and continue the program?

FrankBJensen, 2025-01-21:
I do have test_file == NULL check in my real code. This is minimized code to reproduce the fault I am having.

The only code running is the one you see here. Nothing else. I dont see any memory overflow anywhere. The system is running flawlessly for days, until I save a file using FATFS.

Have you tried running the code? It should now be ready to run even without PSRAM.

FrankBJensen, 2025-01-21:
I do not even have to write anything into the file, this generates same error

Does FATFS use DMA? I can not find any clear answers.

```
bool fatfs_Struct(void* data, size_t size) {

    FILE* outFile = fopen("/fatfs/test.bin", "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", "/fatfs/test.bin");
        perror("Error"); // This will print a more detailed error
        return false;
    }

    uint32_t written = 0;
//    written = fwrite((uint8_t*)data, size, 1, outFile);
    fclose(outFile);

    printf("/fatfs/test.bin have been written, size: %li %lu\n", (uint32_t)size, written);

    return true;
}
```

FrankBJensen, 2025-01-21:
The error occurs when fclose is called - not fopen.

safocl, 2025-01-21:
> I do have test_file == NULL check in my real code. This is minimized code to reproduce the fault I am having.

the program continues to work even after an error, and further memory access will generate undefined behavior.

safocl, 2025-01-21:
[this code](https://godbolt.org/z/MsYhT45no) generates [this log](https://godbolt.org/z/d4G97Mje7)

> I (20819) ads: About to start ADC
> E (25999) ads: Error ESP_ERR_TIMEOUT
> E (26029) ads: TIMEOUT, LOGGING STOPPED!
> E (31049) ads: Error ESP_ERR_TIMEOUT
> E (31059) ads: TIMEOUT, LOGGING STOPPED!
> E (36079) ads: Error ESP_ERR_TIMEOUT
> E (36099) ads: TIMEOUT, LOGGING STOPPED!
> Save file: /fatfs/TESTFILE-1000.bin
> E (38339) FATFS: Failed to open file for writing: /fatfs/TESTFILE-1000.bin
> Error: No such file or directory
> E (41119) ads: Error ESP_ERR_TIMEOUT
> E (41139) ads: TIMEOUT, LOGGING STOPPED!

the error is reached even before working with the file.

FrankBJensen, 2025-01-21:
Please try the latest much later code

extern "C"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"

#include "nvs_flash.h"
#include "esp_system.h"
#include "esp_timer.h"

#include <cmath>
#include "driver/gpio.h"
#include "driver/gptimer.h"
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"
#include "esp_adc/adc_oneshot.h"
#include "esp_timer.h"
#include "driver/timer.h"

#include <algorithm>
#include "esp_vfs_fat.h"

#define SCOP_BUF            210
#define SCOP_CYC              5
#define NUM_SAMPLES          50      // 100   195   450 svarer til 4.5 sekund, oversampling value
#define ADS_STACK          6000      //  3500
#define ADC_BUFFER_SIZE    2048      // 10240
#define ADC_FRAME_SIZE      256      //  2560

#define ADC_TASK_STOP_BIT       (1 << 0)
#define ADC_TASK_OVERFLOW_BIT   (1 << 1)
#define ADC_TASK_TIMEOUT_BIT    (1 << 2)
#define ADC_TASK_READY_BIT      (1 << 3)

static const char *TAGFS = "FATFS";
static const char *TAG = "ADC";

bool fatfs_Struct(void* data, size_t size);
static void _task_adc(void* parameter);

struct TESTFILE {
    char test[10000];
};

wl_handle_t s_wl_handle;
EventGroupHandle_t adc_event_group = NULL;
adc_cali_handle_t adc_cali_handle;
adc_continuous_handle_t adc_handle{nullptr};

const char *path = "/fatfs";
uint8_t* result = NULL;
uint32_t ret_num = 0;

extern "C" void app_main(void)
{
    vTaskDelay(pdMS_TO_TICKS(3000));

    TESTFILE* test_file;
    test_file  = (TESTFILE*)heap_caps_malloc(sizeof(TESTFILE), MALLOC_CAP_DEFAULT);
if (test_file != NULL) ESP_LOGW("MEM", "Testfile initialized OK, size: %i, len: %i", heap_caps_get_allocated_size(test_file), sizeof(TESTFILE));
    *test_file = TESTFILE();

    // Mount configuration
    esp_vfs_fat_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
[... 121 lines cut ...]
                vTaskDelay(1);
*/
                break; // stop med at hente data fra ADC
            }
        }
    }
}

bool fatfs_Struct(void* data, size_t size) {

    printf("Save file: /fatfs/test.bin\n");

    // Serialize the struct to a file
    FILE* outFile = fopen("/fatfs/test.bin", "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", "/fatfs/test.bin");
        perror("Error"); // This will print a more detailed error
        return false;
    }

    fwrite(data, size, 1, outFile);
    fclose(outFile);

    return true;
}

safocl, 2025-01-21:
> The error occurs when fclose is called - not fopen.

my code do not reach to `fclose`

FrankBJensen, 2025-01-21:
Please try the latest code I just sent, its much simpler.

My partition.csv looks like this, there need to be a storage section

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x009000, 0x6000,
phy_init, data, phy,     0x00f000, 0x1000,
factory,  app,  factory, 0x010000, 2M,
otadata,  data, ota,     0x210000, 0x2000,
ota_0,    app,  ota_0,   0x220000, 2M,
ota_1,    app,  ota_1,   0x420000, 2M,
storage,  data, fat,     ,         1M,
```

and I have this in my platformio.ini

`board_build.partitions = partition.csv`

FrankBJensen, 2025-01-21:
I will make sure to exclude driver/timer.h in my real code, thank you.

safocl, 2025-01-21:
hmmm... qemu don't work with ADC? my hardware normally evaluates [this code](https://godbolt.org/z/KobcfhGW8), but qemu don't (produce TIMEOUT ERROR)...

safocl, 2025-01-21:
> # ESP-IDF Partition Table
> # Name,   Type, SubType, Offset,  Size, Flags
> nvs,      data, nvs,     0x009000, 0x6000,
> phy_init, data, phy,     0x00f000, 0x1000,
> factory,  app,  factory, 0x010000, 2M, [...]

my hardware is esp32 with 4Mib flash mem...

FrankBJensen, 2025-01-21:
If you use an emulator, do you then get data fra the ADC? I dont know....

Try this, since OTA is not in this simple example

```
# ESP-IDF Partition Table
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x009000, 0x6000,
phy_init, data, phy,     0x00f000, 0x1000,
factory,  app,  factory, 0x010000, 1M,
storage,  data, fat,     ,         1M,
```

safocl, 2025-01-21:
> [this code](https://godbolt.org/z/MsYhT45no) generates [this log](https://godbolt.org/z/d4G97Mje7)

this code on real esp32 (nodemcu-32) hardware **[do not produce ERROR](https://godbolt.org/z/crj6a3Exx)**

safocl, 2025-01-21:
> Try this, since OTA is not in this simple example

I can't compile your code - it reports the error `not find header files (deprecated)`.

FrankBJensen, 2025-01-21:
I have used Platformio for many years, but I do consider changing to espidf.

Hm, strange. I just compiled this code below, and it produces this log:

```
I (18850) ADC: Data OK
I (18860) ADC: Data OK
I (18860) ADC: Data OK
I (18860) ADC: Data OK
I (18870) ADC: Data OK
I (18870) ADC: Data OK
I (18870) ADC: Data OK
I (18870) ADC: Data OK
I (18880) ADC: Data OK
I (18880) ADC: Data OK
I (18880) ADC: Data OK
I (18890) ADC: Data OK
I (18890) ADC: Data OK
I (18890) ADC: Data OK // this after 15 seconds and a LOT of Data OK's
File opened
I (18900) ADC: Data OK
I (18900) ADC: Data OK
I (18900) ADC: Data OK
E (23900) ADC: Error ESP_ERR_TIMEOUT
E (23900) ADC: TIMEOUT!
E (28900) ADC: Error ESP_ERR_TIMEOUT
E (28900) ADC: TIMEOUT!
E (33900) ADC: Error ESP_ERR_TIMEOUT
E (33900) ADC: TIMEOUT!
```

```
extern "C"

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include <cmath>
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"

#include "esp_vfs_fat.h"

#define ADS_STACK          6000      //  3500
#define ADC_BUFFER_SIZE    20480      // 10240
#define ADC_FRAME_SIZE      512      //  2560

#define ADC_TASK_STOP_BIT       (1 << 0)
#define ADC_TASK_OVERFLOW_BIT   (1 << 1)
#define ADC_TASK_TIMEOUT_BIT    (1 << 2)
#define ADC_TASK_READY_BIT      (1 << 3)

static const char *TAGFS = "FATFS";
static const char *TAG = "ADC";

bool fatfs_Struct(void* data, size_t size);
static void _task_adc(void* parameter);

char test_file[10];

wl_handle_t s_wl_handle;
EventGroupHandle_t adc_event_group = NULL;
adc_cali_handle_t adc_cali_handle;
adc_continuous_handle_t adc_handle{nullptr};

const char *path = "/fatfs";
uint8_t* result = NULL;
uint32_t ret_num = 0;
[... 134 lines cut ...]
                break; // stop med at hente data fra ADC
            }
        }
    }
}

bool fatfs_Struct(void* data, size_t size) {

    FILE* outFile = fopen("/fatfs/test.bin", "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", "/fatfs/test.bin");
        perror("Error"); // This will print a more detailed error
        return false;
    }

    printf("File opened\n");

//  uint32_t  written = fwrite((uint8_t*)data, size, 1, outFile);
    fclose(outFile);

//    printf("/fatfs/test.bin have been written, size: %li %lu\n", (uint32_t)size, written);

    return true;
}
```

safocl, 2025-01-21:
@FrankBJensen deleting the code to use FAT produces normal execution?

FrankBJensen, 2025-01-21:
Yes. Then it runs for months.

I have 7 products running my software. I only use FAT if I want to store statestics. I am only using FAT on 2 of these 7 units, the task with FAT is disabled on the 5 pcs.. And only the 2 units with FAT fails. The other 5 units never fail. One has been running more then 90 days.

This is the reason why I began to investigate this error. And then suddenly I discovered, that the error occured exactly at a whole hour, eg. 14:00:00 or 15:00:00, and that is exactly the time, where I run my statestics task. 

Then I began pulling the code to bits and pieces, to find out, what caused this, ending up in me narrowing it down to the FAT save procedure. And this is where this thread started, and I copied my actual code snippets to a separate project, in order to better investigate, and have some real compilable code to post here.

FrankBJensen, 2025-01-21:
Strange. Can you see what line? I have no idea, what cache it talks about.

At least you have Data OK now :-)

FrankBJensen, 2025-01-21:
maybe you need to add this to platformio.ini in order to see, what line causes the panic

```
monitor_filters = direct, esp32_exception_decoder
build_flags =
    -DLOG_LOCAL_LEVEL=ESP_LOG_DEBUG
```

safocl, 2025-01-21:
> maybe you need to add this to platformio.ini in order to see, what line causes the panic

i don't use pio...

safocl, 2025-01-21:
the last debug report:
```
I (9644) ADC: Data OK
fatfs_Struct( test_file, sizeof( test_file ) ); start
fopen( /fatfs/test.bin, wb ); start
V (9654) vfs_fat: vfs_fat_open: path="/test.bin", flags=601, mode=1b6
I (9654) ADC: Data OK
V (9654) ff_diskio_spiflash: ff_wl_read - pdrv=0, sector=3, count=1
D (9664) wl_flash: read - src_addr= 0x00003000, size= 0x00001000
V (9664) wl_flash: calcAddr - addr= 0x00003000 -> result= 0x00004000, dummy_addr= 0x00000000
V (9674) ff_diskio_spiflash: ff_wl_read - pdrv=0, sector=4, count=1
D (9684) wl_flash: read - src_addr= 0x00004000, size= 0x00001000
V (9684) wl_flash: calcAddr - addr= 0x00004000 -> result= 0x00005000, dummy_addr= 0x00000000
V (9694) ff_diskio_spiflash: ff_wl_read - pdrv=0, sector=5, count=1
D (9704) wl_flash: read - src_addr= 0x00005000, size= 0x00001000
V (9704) wl_flash: calcAddr - addr= 0x00005000 -> result= 0x00006000, dummy_addr= 0x00000000
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400e0868  PS      : 0x00060034  A0      : 0x8008524f  A1      : 0x3ffb10c0  
A2      : 0x00000000  A3      : 0xff000000  A4      : 0x8008b660  A5      : 0x3ffba600  
A6      : 0x3ffc4c58  A7      : 0x3ff420c0  A8      : 0x80085184  A9      : 0x3ffba6a4  
A10     : 0x3ffc4c5c  A11     : 0x3ffca7ac  A12     : 0x3ffb10c4  A13     : 0x3ffb10c8  
A14     : 0x00000001  A15     : 0x3ffb2d2c  SAR     : 0x00000018  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400e0865:0x3ffb10c0 0x4008524c:0x3ffb1100 0x40083431:0x3ffb1120 0x4008ba1d:0x3ffba640 0x4008e11b:0x3ffba660 0x4008493f:0x3ffba6e0 0x400dd3d2:0x3ffba720 0x400dac1a:0x3ffba750 0x400dc241:0x3ffba770 0x400dab06:0x3ffba7b0 0x400d6af9:0x3ffba7d0 0x400d6a0a:0x3ffba800 0x400d6dc3:0x3ffba820 0x400d7904:0x3ffba840 0x400d7ddd:0x3ffba860 0x400d7f17:0x3ffba890 0x400d9c69:0x3ffba8f0 0x4011f3f5:0x3ffba930 0x40100ce2:0x3ffba950 0x40100d75:0x3ffba980 0x400d632a:0x3ffba9a0 0x400d65e4:0x3ffba9c0 0x4011e64b:0x3ffbaa00 0x40087e35:0x3ffbaa30

ELF file SHA256: 577cd0c89

CPU halted.
```

FrankBJensen, 2025-01-21:
What do this return (earlier in the code)? Is the filesystem mounted?

```
esp_err_t ret = esp_vfs_fat_spiflash_mount_rw_wl(path, "storage", &mount_config, &s_wl_handle);

if (ret != ESP_OK) {
    ESP_LOGE(TAGFS, "Failed to mount FATFS (%s)", esp_err_to_name(ret));
} else {
    ESP_LOGI(TAGFS, "FAT filesystem mounted");
}
```

safocl, 2025-01-21:
> What do this return (earlier in the code)? Is the filesystem mounted?

yes -- it is successfully

FrankBJensen, 2025-01-21:
That is strange. If I enable this logging below, 

```
esp_log_level_set("vfs_fat", ESP_LOG_VERBOSE);
esp_log_level_set("ff_diskio_spiflash", ESP_LOG_VERBOSE);
esp_log_level_set("wl_flash", ESP_LOG_VERBOSE);
```

Then I get this log (and note, timeout just after write......)
```
I (18880) ADC: Data OK
I (18880) ADC: Data OK
I (18890) ADC: Data OK
I (18890) ADC: Data OK
I (18890) ADC: Data OK
I (18900) ADC: Data OK
I (18900) ADC: Data OK
I (18900) ADC: Data OK
I (18910) ADC: Data OK
D (18910) wl_flash: read - src_addr= 0x00003000, size= 0x00001000
File opened
D (18910) wl_flash: erase_range - start_address= 0x00003000, size= 0x00001000
D (18910) wl_flash: erase_sector - sector= 0x00000003
I (18910) ADC: Data OK
I (18910) ADC: Data OK
I (18910) ADC: Data OK
D (18940) wl_flash: write - dest_addr= 0x00003000, size= 0x00001000
E (23910) ADC: Error ESP_ERR_TIMEOUT
E (23910) ADC: TIMEOUT!
E (28910) ADC: Error ESP_ERR_TIMEOUT
E (28910) ADC: TIMEOUT!
E (33910) ADC: Error ESP_ERR_TIMEOUT
E (33910) ADC: TIMEOUT!
```

safocl, 2025-01-21:
[log](https://godbolt.org/z/aedxnqzYE) with [code](https://godbolt.org/z/KEcd5hncG)

FrankBJensen, 2025-01-21:
So it fails here
`FILE * outFile = fopen( "/fatfs/test.bin", "wb" );`
But that should be basic stuff, not relying on any of our code.

There must be something wrong with the mount or something....... You do have 4mb flash, and you dont adress more than that?

Can you compile using my sdkconfig? Just in case, there is a setting somewhere with cache?

safocl, 2025-01-21:
> Can you compile using my sdkconfig?

my hardware is not **ESP32-S3** -- but it is **ESP32**

FrankBJensen, 2025-01-21:
OK. But the file system should work on any ESP32, I guess.... 

Do you have another project, where the filesystem is working?

FrankBJensen, 2025-01-21:
Googling, something like this keeps comming up: 

> Ensure Proper Memory Allocation: Make sure that the memory regions used by the WL layer are correctly allocated and not in cached regions when the cache is disabled.

Maybe try turning wear leveling OFF ?

safocl, 2025-01-21:
> Do you have another project, where the filesystem is working?

no

FrankBJensen, 2025-01-21:
I made a new file-only project from scratch.

I made the partition 512K like you, and that gave me some problems, so I had to change sector and wear level size to 512 bytes, in order for it to mount successfully. 

Maybe you should try changing sector and wear level size to 512 bytes??
And see if this compiles and run ??

log
```
FATFS: FAT filesystem mounted
File opened
/fatfs/test.bin have been written, size: 10 1

```
code
```
#include "esp_vfs_fat.h"
#include "esp_log.h"

char test_file[10];
const char *path = "/fatfs";
wl_handle_t s_wl_handle;

static const char *TAGFS = "FATFS";

extern "C" void app_main(void)
{
    vTaskDelay(pdMS_TO_TICKS(3000));

    esp_log_level_set("vfs_fat", ESP_LOG_VERBOSE);
    esp_log_level_set("ff_diskio_spiflash", ESP_LOG_VERBOSE);
    esp_log_level_set("wl_flash", ESP_LOG_VERBOSE);

    // Mount configuration
    esp_vfs_fat_mount_config_t mount_config = {
        .format_if_mount_failed = true,
        .max_files = 5,
        .allocation_unit_size = CONFIG_WL_SECTOR_SIZE
    };

    // Mount the partition named "storage" as FAT with wear leveling
    esp_err_t ret = esp_vfs_fat_spiflash_mount_rw_wl(path, "storage", &mount_config, &s_wl_handle);

    if (ret != ESP_OK) {
        ESP_LOGE(TAGFS, "Failed to mount FATFS (%s)", esp_err_to_name(ret));
        if (ret == ESP_FAIL) {
            ESP_LOGE(TAGFS, "General failure, possibly due to incorrect partition size or label.");
        } else if (ret == ESP_ERR_NO_MEM) {
            ESP_LOGE(TAGFS, "Out of memory.");
        } else if (ret == ESP_ERR_NOT_FOUND) {
            ESP_LOGE(TAGFS, "Partition not found.");
        } else if (ret == ESP_ERR_INVALID_STATE) {
            ESP_LOGE(TAGFS, "Invalid state, possibly due to incorrect wear leveling configuration.");
        }
        while (1) {
            vTaskDelay(pdMS_TO_TICKS(10));
        }
    } else {
        ESP_LOGI(TAGFS, "FAT filesystem mounted");
    }

    FILE* outFile = fopen("/fatfs/test.bin", "wb");
    if (outFile == NULL) {
        ESP_LOGE("FATFS", "Failed to open file for writing: %s", "/fatfs/test.bin");
        perror("Error"); // This will print a more detailed error
        while (1) {
            vTaskDelay(pdMS_TO_TICKS(10));
        }
    }

    printf("File opened\n");

    uint32_t  written = fwrite((uint8_t*)test_file, sizeof(test_file), 1, outFile);
    fclose(outFile);

    printf("/fatfs/test.bin have been written, size: %u %lu\n", sizeof(test_file), written);

    while (1) {
        vTaskDelay(pdMS_TO_TICKS(10));
    }
}
```

safocl, 2025-01-21:
> Maybe you should try changing sector and wear level size to 512 bytes??

yes, i do it.

FrankBJensen, 2025-01-22:
OK, that seems to work. Based on that, I made this modified code. If you use same project, and paste this, then file should work. The fault with timeout still persists. Please try this

Log
```
I (18396) ADC: Data OK
I (18396) ADC: Data OK
I (18406) ADC: Data OK
I (18406) ADC: Data OK
I (18406) ADC: Data OK
I (18416) ADC: Data OK
I (18416) ADC: Data OK
I (18416) ADC: Data OK
I (18426) ADC: Data OK
I (18426) ADC: Data OK
I (18426) ADC: Data OK
I (18436) ADC: Data OK
I (18436) ADC: Data OK
I (18436) ADC: Data OK
I (18436) ADC: Data OK
File opened
/fatfs/test.bin have been written, size: 10 1
E (23446) ADC: Error ESP_ERR_TIMEOUT
E (23446) ADC: TIMEOUT!
E (28446) ADC: Error ESP_ERR_TIMEOUT
E (28446) ADC: TIMEOUT!
```

Code
```
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"

#include "esp_log.h"
#include <cmath>
#include "esp_adc/adc_continuous.h"
#include "esp_adc/adc_cali.h"
#include "esp_adc/adc_cali_scheme.h"

#include "esp_vfs_fat.h"

#define ADS_STACK          6000      //  3500
#define ADC_BUFFER_SIZE    20480      // 10240
#define ADC_FRAME_SIZE      512      //  2560

#define ADC_TASK_STOP_BIT       (1 << 0)
#define ADC_TASK_OVERFLOW_BIT   (1 << 1)
#define ADC_TASK_TIMEOUT_BIT    (1 << 2)
#define ADC_TASK_READY_BIT      (1 << 3)

static const char *TAGFS = "FATFS";
static const char *TAG = "ADC";

bool fatfs_Struct(void* data, size_t size);
static void _task_adc(void* parameter);

char test_file[10];

wl_handle_t s_wl_handle;
EventGroupHandle_t adc_event_group = NULL;
adc_cali_handle_t adc_cali_handle;
adc_continuous_handle_t adc_handle{nullptr};

const char *path = "/fatfs";
uint8_t* result = NULL;
uint32_t ret_num = 0;

extern "C" void app_main(void)
{
    vTaskDelay(pdMS_TO_TICKS(3000));

[... 138 lines cut ...]

        while (1) {

            esp_err_t ret_adc = adc_continuous_read(adc_handle, result, ADC_FRAME_SIZE, &ret_num, 5000);

            if (ret_adc == ESP_OK) {
                ESP_LOGI(TAG, "Data OK");
            } else if (ret_adc == ESP_ERR_TIMEOUT) {

                ESP_LOGE(TAG, "Error ESP_ERR_TIMEOUT");
                xEventGroupSetBits(adc_event_group, ADC_TASK_TIMEOUT_BIT);

/* This code will resume ADC if enabled
                vTaskDelay(1);
                adc_continuous_stop(adc_handle);
                vTaskDelay(1);
                adc_continuous_start(adc_handle);
                vTaskDelay(1);
*/
                break; // stop med at hente data fra ADC
            }
        }
    }
}
```

safocl, 2025-01-22:
> Code

```
I (18460) ADC: Data OK
Guru Meditation Error: Core  0 panic'ed (Cache disabled but cached memory region accessed). 

Core  0 register dump:
PC      : 0x400e3400  PS      : 0x00060034  A0      : 0x80086758  A1      : 0x3ffb1090  
A2      : 0x00000000  A3      : 0x00001000  A4      : 0xc1b4c868  A5      : 0xc1b4c868  
A6      : 0xffffdfff  A7      : 0x3ffc0258  A8      : 0x80086660  A9      : 0xc1b4c868  
A10     : 0x3ffc025c  A11     : 0x3ffc6224  A12     : 0x3ffb109c  A13     : 0x3ffb10a0  
A14     : 0x3ffb1810  A15     : 0x00000000  SAR     : 0x00000000  EXCCAUSE: 0x00000007  
EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x00000000  

Backtrace: 0x400e33fd:0x3ffb1090 0x40086755:0x3ffb10d0 0x40084025:0x3ffb1100 0x4008e75d:0x3ffb3b70 0x4008e7a5:0x3ffb3ba0 0x40091226:0x3ffb3be0 0x40091a59:0x3ffb3c30 0x400920df:0x3ffb3c60 0x40091a12:0x3ffb3c90 0x4009205e:0x3ffb3cc0 0x40085cb7:0x3ffb3d40 0x400def2d:0x3ffb3da0 0x400dc6e0:0x3ffb3dd0 0x400de0ad:0x3ffb3e00 0x400dc8bd:0x3ffb3e50 0x400de589:0x3ffb3ea0 0x400dc3e2:0x3ffb3ef0 0x400d7969:0x3ffb3f20 0x400d7794:0x3ffb3f50 0x400d7e2a:0x3ffb3f80 0x400d7e94:0x3ffb3fb0 0x400d84c4:0x3ffb3fe0 0x400d8eaa:0x3ffb4010 0x400d92fd:0x3ffb4040 0x400db57f:0x3ffb40b0 0x400f74ed:0x3ffb40f0 0x400e3b06:0x3ffb4120 0x400e3b99:0x3ffb4150 0x400d723e:0x3ffb4170 0x400f6718:0x3ffb41c0 0x40089e7a:0x3ffb41f0
```

FrankBJensen, 2025-01-22:
I dont understand. And you use same project, where you have set sector and wearlevel to 512 bytes?

safocl, 2025-01-22:
> I dont understand. And you use same project, where you have set sector and wearlevel to 512 bytes?

```
╰─>$ grep "512" sdkconfig 
CONFIG_SOC_SHA_SUPPORT_SHA512=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=512
CONFIG_HTTPD_MAX_URI_LEN=512
# CONFIG_FATFS_SECTOR_512 is not set
CONFIG_MBEDTLS_SHA512_C=y
CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED=y
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE=512

╰─>$ idf.py menuconfig
[...]

╰─>$ grep "512" sdkconfig
CONFIG_SOC_SHA_SUPPORT_SHA512=y
CONFIG_HTTPD_MAX_REQ_HDR_LEN=512
CONFIG_HTTPD_MAX_URI_LEN=512
CONFIG_FATFS_SECTOR_512=y
CONFIG_MBEDTLS_SHA512_C=y
CONFIG_MBEDTLS_ECP_DP_BP512R1_ENABLED=y
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE=512
```

FrankBJensen, 2025-01-22:
I dont understand. It compiles and runs fine here at my place, that exact code.
In what line of code does it fail? Is it still on fopen ?

safocl, 2025-01-22:
> I dont understand. It compiles and runs fine here at my place, that exact code.
> In what line of code does it fail? Is it still on fopen ?

and I don't understand -- probably on 
`FILE* outFile = fopen(“/fatfs/test.bin”, “wb”);`

FrankBJensen, 2025-01-22:
Yes, but thats the exact code from the file-only example, so why does it fail now..... 
I dont know how to solve that. I dont have that problem.
And I still have the problem, that ADC is stopping on the fclose command....