espressif/esp-idf #15240: [BLE][IDF v5.3.x/5.4.0] Unable to Keep Periodic Advertising Running After Stopping Extended Advertising / Cannot Set Large Extended Adv Intervals

iamcalledusman, 2025-01-20:
Hello Espressif Team,

I am facing an issue with Extended Advertising and Periodic Advertising in ESP-IDF, specifically:

Stopping Extended Advertising Also Stops Periodic Advertising

According to the Bluetooth 5.0 spec, it should be possible to continue periodic advertising even after stopping extended advertising on the same handle. However, in my testing with ESP-IDF, once I call esp_ble_gap_ext_adv_stop(...), the periodic advertising also stops.
Large Extended Advertising Intervals Appear Clamped at ~40 s

When I set .interval_min and .interval_max to larger 24-bit values (e.g., 0x1FFFF for ~81 s), I still observe an actual advertising interval closer to 40 s. This suggests the controller might be falling back to the 16-bit legacy limit, even though I am using a non-legacy extended advertising property (NONCONN_NONSCANNABLE_UNDIRECTED).
My requirement is to have a short burst of extended advertising to establish periodic sync, and then either:

Stop extended advertising entirely while keeping periodic advertising on the same handle, or
Use a truly large extended advertising interval (e.g., ~81 s or more, up to the 24-bit limit of ~2.9 hours) to minimize power consumption.
However, neither approach seems possible in practice with the current ESP-IDF

**Environment**
Module or Chip: ESP32-C6
ESP-IDF Version: v5.3.2/v5.4.0(tried both)
Compiler: riscv32-esp-elf-gcc
Operating System: Windows
Power Supply: USB
Example code : https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble_50/periodic_adv

```
#define EXT_ADV_HANDLE   0

static esp_ble_gap_ext_adv_params_t ext_adv_params = {
    .type           = ESP_BLE_GAP_SET_EXT_ADV_PROP_NONCONN_NONSCANNABLE_UNDIRECTED,
    .interval_min   = 0x1FFFF,  // aiming for ~81 s
    .interval_max   = 0x1FFFF,
    .channel_map    = ADV_CHNL_ALL,
    .filter_policy  = ADV_FILTER_ALLOW_SCAN_ANY_CON_ANY,
    .primary_phy    = ESP_BLE_GAP_PHY_1M,  // Tried with CODED as well 
    .max_skip       = 0,
    .secondary_phy  = ESP_BLE_GAP_PHY_2M, // // Tried with CODED as well 
    .sid            = 0,
    .scan_req_notif = false,
    .own_addr_type  = BLE_ADDR_TYPE_RANDOM,
    .tx_power       = EXT_ADV_TX_PWR_NO_PREFERENCE,
};

static esp_ble_gap_periodic_adv_params_t periodic_adv_params = {
    .interval_min = 24000,  // ~30 s
    .interval_max = 24000,
    .properties   = 0,
};
```

**Observed Behavior**
Stopping Extended Advertising

esp_ble_gap_ext_adv_stop(1, &ext_adv[EXT_ADV_HANDLE]);
This also kills periodic advertising on the same handle. The logs confirm both advertisements stop.

Large Interval Clamped
If I do not stop extended advertising, I set .interval_min = 0x1FFFF (aiming for ~81 s). In practice, using a BLE sniffer or a scanning device, I see the device advertising at ~40 s intervals, indicating a fallback to the legacy 16-bit maximum.

Expected Behavior
According to the BLE Specification: We can keep periodic advertising running even if we stop sending extended advertising PDUs, or we can set an extended interval up to 0xFFFFFF (~2.9 hours) for truly extended advertising.

In either case, we would see either zero extended adv overhead (only periodic PDUs) or a very infrequent extended adv interval.

**Question / Request**
Is this a known limitation or bug in the ESP-IDF BLE controller stack?
Are there workarounds to achieve “periodic only” without extended PDUs or to set the interval larger than ~40 s?
Is there any plan to support stopping extended advertising while keeping periodic advertising active on the same handle, in line with the BLE spec?
Any guidance or solutions would be greatly appreciated.

**Wireshark Logs**

You can see that I set the interval of 0x1FFFF but getting PDU almost every 41 seconds, Tried with coded phy as well but still the same behavior.

[image]

Thank you!

ZhouXiao616, 2025-01-23:
Hi @iamcalledusman , thanks for you feedback!

We noticed that you mentioned two issues:

1. Extended advertising interval could not exceed 40.95s
2. Stopping extended adveritising also stops periodic advertising

The first one is a known issue and we're fixing it, the fix would be merged to IDF as soon as possible. :)

For the second one, we tried to reproduce the issue locally with ESP32C6 on release/v5.3 and release/v5.4. In our modification to https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble_50/periodic_adv, we called `esp_ble_gap_ext_adv_stop` function right after calling `esp_ble_gap_periodic_adv_start` function, but periodic advertising didn't stop. We also tried to set periodic advertising interval to around 30s, it still worked well.

Related codes (on release/v5.4):

``` C

void app_main(void)
{
    ...

    // create static random address
    esp_bd_addr_t rand_addr;
    esp_ble_gap_addr_create_static(rand_addr);

    test_sem = xSemaphoreCreateBinary();

    // 2M phy extend adv, Non-Connectable and Non-Scannable Undirected advertising
    ESP_LOG_BUFFER_HEX(LOG_TAG, rand_addr, ESP_BD_ADDR_LEN);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_params(EXT_ADV_HANDLE, &ext_adv_params_2M), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_set_rand_addr(EXT_ADV_HANDLE, rand_addr), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_ext_adv_data_raw(EXT_ADV_HANDLE, sizeof(raw_ext_adv_data_2m), &raw_ext_adv_data_2m[0]), test_sem);

    // start all adv
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_start(NUM_EXT_ADV, &ext_adv[0]), test_sem);

    // set periodic adv param
    FUNC_SEND_WAIT_SEM(esp_ble_gap_periodic_adv_set_params(EXT_ADV_HANDLE, &periodic_adv_params), test_sem);

#if (CONFIG_BT_BLE_FEAT_PERIODIC_ADV_ENH)
    // set periodic adv raw data
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_periodic_adv_data_raw(EXT_ADV_HANDLE, sizeof(periodic_adv_raw_data), &periodic_adv_raw_data[0], false), test_sem);
    // start periodic adv, include the ADI field in AUX_SYNC_IND PDUs
    FUNC_SEND_WAIT_SEM(esp_ble_gap_periodic_adv_start(EXT_ADV_HANDLE, true), test_sem);
    while (1) {
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        // just update the Advertising DID of the periodic advertising, unchanged data
        FUNC_SEND_WAIT_SEM(esp_ble_gap_config_periodic_adv_data_raw(EXT_ADV_HANDLE, 0, NULL, true), test_sem);
    }
#else
    // set periodic adv raw data
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_periodic_adv_data_raw(EXT_ADV_HANDLE, sizeof(periodic_adv_raw_data), &periodic_adv_raw_data[0]), test_sem);
    FUNC_SEND_WAIT_SEM(esp_ble_gap_periodic_adv_start(EXT_ADV_HANDLE), test_sem);
#endif

    // Stop extended advertising right after starting periodic advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_stop(NUM_EXT_ADV, &ext_adv[0].instance), test_sem);

    return;
}
```

Related logs:
```
I (528) PERIODIC_ADV_DEMO: Extended advertising params set, status 0, instance 0
I (538) PERIODIC_ADV_DEMO: Extended advertising random address set, status 0, instance 0
I (548) PERIODIC_ADV_DEMO: Extended advertising data set, status 0, instance 0
I (558) PERIODIC_ADV_DEMO: Extended advertising start, status 0, instance numble 1
I (558) PERIODIC_ADV_DEMO: Periodic advertising params set, status 0, instance 0
I (568) PERIODIC_ADV_DEMO: Periodic advertising data set, status 0, instance 0
I (578) PERIODIC_ADV_DEMO: Periodic advertising start, status 0, instance 0
I (578) PERIODIC_ADV_DEMO: Extended advertising start, status 0, instance numble 1
I (588) main_task: Returned from app_main()
```

Checking your log, we found that periodic advertising didn't even start, could you please share your codes so we can help you debug the issue?

PS: We noticed that you tried to stop extended advertising by calling `esp_ble_gap_ext_adv_stop(1, &ext_adv[EXT_ADV_HANDLE]);` and it seems to be incorrect. If you call `esp_ble_gap_ext_adv_start(NUM_EXT_ADV, &ext_adv[0]);` to start extended advertising, you should call `esp_ble_gap_ext_adv_stop(NUM_EXT_ADV, &ext_adv[0].instance);` to stop extended advertising.

We hope it would help. Looking forward to your reply.

iamcalledusman, 2025-01-23:
Hi @ZhouXiao616!

Thank you so much for your time and assistance! 

1. Extended advertising interval could not exceed 40.95s
    
      Thank you for fixing it in the coming esp-idf version!

2. Stopping extended advertising also stops periodic advertising

**Periodic Advertisement LOG FILE** 
```
I (406) main_task: Calling app_main()
I (416) BLE_INIT: Using main XTAL as clock source
I (426) BLE_INIT: ble controller commit:[b623f7a]
I (426) BLE_INIT: Bluetooth MAC: 40:4c:ca:51:42:da
I (426) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (476) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
I (686) PERIODIC_ADV_DEMO: Extended advertising params set, status 0, instance 0
I (686) PERIODIC_ADV_DEMO: Extended advertising random address set, status 0, instance 0
I (686) PERIODIC_ADV_DEMO: Extended advertising data set, status 0, instance 0
I (696) PERIODIC_ADV_DEMO: Extended advertising start, status 0, instance numble 1
I (706) PERIODIC_ADV_DEMO: Periodic advertising params set, status 0, instance 0
I (716) PERIODIC_ADV_DEMO: Periodic advertising data set, status 0, instance 0
I (726) PERIODIC_ADV_DEMO: ESP_GAP_BLE_PERIODIC_ADV_START_COMPLETE_EVT, status 0, instance 0
I (726) PERIODIC_ADV_DEMO: Extended advertising stop, status 0, instance numble 1
I (736) main_task: Returned from app_main()
```

**Periodic Sync LOG FILE** 
```
I (405) main_task: Calling app_main()
I (425) BLE_INIT: Using main XTAL as clock source
I (425) BLE_INIT: ble controller commit:[b623f7a]
I (425) BLE_INIT: Bluetooth MAC: 40:4c:ca:51:62:5e
I (425) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (475) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
I (685) PERIODIC_SYNC: Extended scanning params set, status 0
I (685) PERIODIC_SYNC: Extended scanning start, status 0
I (685) main_task: Returned from app_main()
I (6165) extended adv data is......... : 02 01 06 02 0a eb 13 09 45 53 50 5f 4d 55 4c 54
I (6165) extended adv data is......... : 49 5f 41 44 56 5f 38 30 4d 53
I (6175) PERIODIC_SYNC: Start create sync with the peer device ESP_MULTI_ADV_80MS
I (6185) PERIODIC_SYNC: Periodic advertising create sync, status 0

```

[periodic_adv & Sync.zip](https://github.com/user-attachments/files/18526552/periodic_adv.Sync.zip)

I attached both the project periodic_adv as well as the periodic_sync 

I enabled the following but still got the same response after disabling it.
**#if (CONFIG_BT_BLE_FEAT_PERIODIC_ADV_ENH)**

I noticed the same response, after calling esp_ble_gap_ext_adv_stop(NUM_EXT_ADV, &ext_adv[0].instance); both the periodic and ext adv pdu stopped. 

Thank you again for your response!

Looking forward to your reply.

iamcalledusman, 2025-01-24:
Hi @ZhouXiao616!

Thank you so much for your time and assistance!

After a small change , now periodic adv is  working fine 

```
#if (CONFIG_BT_BLE_FEAT_PERIODIC_ADV_ENH)
    // set periodic adv raw data
    FUNC_SEND_WAIT_SEM(esp_ble_gap_config_periodic_adv_data_raw(EXT_ADV_HANDLE, sizeof(periodic_adv_raw_data), &periodic_adv_raw_data[0], false), test_sem);
    // start periodic adv, include the ADI field in AUX_SYNC_IND PDUs
    FUNC_SEND_WAIT_SEM(esp_ble_gap_periodic_adv_start(EXT_ADV_HANDLE, true), test_sem);

  
    vTaskDelay(5000 / portTICK_PERIOD_MS);
    
    // Stop extended advertising right after starting periodic advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_stop(NUM_EXT_ADV, &ext_adv[0].instance), test_sem);

    while (1) {
        vTaskDelay(2000 / portTICK_PERIOD_MS);
        // just update the Advertising DID of the periodic advertising, unchanged data
        FUNC_SEND_WAIT_SEM(esp_ble_gap_config_periodic_adv_data_raw(EXT_ADV_HANDLE, 0, NULL, true), test_sem);
    }

```

Added the delay and stop the function 

```

 vTaskDelay(5000 / portTICK_PERIOD_MS);
    
    // Stop extended advertising right after starting periodic advertising
    FUNC_SEND_WAIT_SEM(esp_ble_gap_ext_adv_stop(NUM_EXT_ADV, &ext_adv[0].instance), test_sem);

```

Again thank You!

ZhouXiao616, 2025-02-02:
Hi @iamcalledusman, glad to hear that!

Alvin1Zhang, 2025-04-10:
Thanks for reporting, fix on release/5.4 is available at https://github.com/espressif/esp-idf/commit/dd9dbfccd615346abd882f89e90c7e190aa7cd65, feel free to reopen if the issue still happens.

closed 2025-04-10 by Alvin1Zhang