espressif/esp-idf #15241: Allow FF_FS_NOFSINFO to be configured via menuconfig

try-except, 2025-01-20:
### Is your feature request related to a problem?

When using the `f_getfree` function from the FatFs library, sometimes the wrong value for the amount of available clusters is returned. FatFs offers the option to request a full scan of the FAT when calling `f_getfree` for the first time via the `FF_FS_NOFSINFO` macro defined in `ffconf.h`, but currently in ESP-IDF this is a hardcoded value and can't be set per-project. This is a little bit annoying when working in a team for example because everyone has to manually change the value in their ESP-IDF installation.

### Describe the solution you'd like.

Just as `FF_FS_LOCK` or `FF_FS_TIMEOUT`, it would be very nice to have the option to set the value for `FF_FS_NOFSINFO` in `menuconfig`.

### Describe alternatives you've considered.

Changing the vale manually whenever I catch the `f_getfree` function returning bogus values.

adokitkat, 2025-01-23:
Good point, thank you.

try-except, 2025-01-23:
I think setting it to 0 as the default is ok since that's what FatFs uses as the default, so I personally would try to mimic FatFs and their documentation. Whenever I want a full scan I just set it to 3, that way FatFs doesn't "trust" any previous values. I haven't really tried the other options.

closed 2025-03-24 by espressif-bot