espressif/esp-idf #15242: Symbol  defined in multiple locations

chegewara, 2025-01-21:
Hi team,
can we do something with this when we are using some components from registry? It is a bit annoying or confusing and im not sure how it may impact configurations (which one is used)
```
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:26
info: INFO: Symbol ESP_WIFI_STATIC_TX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:78
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:62
info: INFO: Symbol ESP_WIFI_DYNAMIC_TX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:80
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:64
info: INFO: Symbol ESP_WIFI_TX_BUFFER_TYPE defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:85
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:69
info: INFO: Symbol ESP_WIFI_STATIC_TX_BUFFER_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:90
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:74
info: INFO: Symbol ESP_WIFI_CACHE_TX_BUFFER_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:89
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:105
info: INFO: Symbol ESP_WIFI_DYNAMIC_TX_BUFFER_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:102
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:118
info: INFO: Symbol ESP_WIFI_STATIC_RX_MGMT_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:129
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:145
info: INFO: Symbol ESP_WIFI_DYNAMIC_RX_MGMT_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:131
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:147
info: INFO: Symbol ESP_WIFI_DYNAMIC_RX_MGMT_BUF defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:151
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:135
info: INFO: Symbol ESP_WIFI_RX_MGMT_BUF_NUM_DEF defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:156
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:140
info: INFO: Symbol ESP_WIFI_CSI_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:150
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:166
info: INFO: Symbol ESP_WIFI_AMPDU_TX_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:175
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:159
info: INFO: Symbol ESP_WIFI_TX_BA_WIN defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:166
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:182
info: INFO: Symbol ESP_WIFI_AMPDU_RX_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:178
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:194
info: INFO: Symbol ESP_WIFI_RX_BA_WIN defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:200
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:184
info: INFO: Symbol ESP_WIFI_AMSDU_TX_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:199
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:215
info: INFO: Symbol ESP_WIFI_NVS_ENABLED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:206
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:222
info: INFO: Symbol ESP_WIFI_TASK_PINNED_TO_CORE_0 defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:235
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:219
info: INFO: Symbol ESP_WIFI_TASK_PINNED_TO_CORE_1 defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:221
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:237
info: INFO: Symbol ESP_WIFI_SOFTAP_BEACON_MAX_LEN defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:225
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:241
info: INFO: Symbol ESP_WIFI_MGMT_SBUF_NUM defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:264
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:248
info: INFO: Symbol ESP_WIFI_IRAM_OPT defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:274
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:255
[... 186 lines cut ...]
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./Kconfig.soc_wifi_caps.in:45
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./Kconfig.soc_wifi_caps.in:129
[... cut ...]
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:116
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:132
info: INFO: Choice ESP_WIFI_TASK_CORE_ID defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:228
    /home/chegewara/programming/open-source/tests/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/Kconfig:212
```

Thanks

chegewara, 2025-01-22:
Hi @igrr 
sorry for poor info about build environment.
I believe it is with components like esp_wifi_remote and esp_hosted on esp32-p4 and probably all esp-idf supported versions.
I tested with 
- esp32-p4 devkit
- esp-idf master 0f0068fff3ab159f082133aadfa9baf4fc0c7b8d
- idf_component.yml
```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_wifi_remote: "*"
  espressif/esp_hosted: "*"
  ## Required IDF version
  idf:
    version: ">=5.4.0"
```

From this single example
```
info: INFO: Symbol SLAVE_SOC_WIFI_SUPPORTED defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:281
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:233
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:73
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:189
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:5
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:33
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:117
    /home/chegewara/programming/open-source/esp32P4_projects/ble-mesh-p4/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.soc_wifi_caps.in:161
```
im guessing that `if SLAVE_IDF_TARGET_ESP32XX` in Kconfig are ignored. I understand it is from esp_wifi_remote component, but i believe it may be general cmake/Kconfig issue, thats why i posted it here. Or maybe this check should be checked different way, im not Kconfig expert, sorry:
```
if SLAVE_IDF_TARGET_ESP32C6 == 'y'
```

BTW i just found some inconsistency in menuconfig, no idea why:
```
IDF_SLAVE_TARGET(=esp32c6)                                          <----------------- C6
SLAVE_CHIPSET_ESP32(=n) "Slave as ESP32"
SLAVE_CHIPSET_ESP32C2(=n) "Slave as ESP32C2"
SLAVE_CHIPSET_ESP32C3(=n) "Slave as ESP32C3"
SLAVE_CHIPSET_ESP32C5(=n) "Slave as ESP32C5"
SLAVE_CHIPSET_ESP32C6(=y) "Slave as ESP32C6"         <---------------- C6
SLAVE_CHIPSET_ESP32S2(=n) "Slave as ESP32S2"
SLAVE_CHIPSET_ESP32S3(=n) "Slave as ESP32S3"
SLAVE_IDF_TARGET_ESP32(=y) "esp32"                            <----------------- why here?
SLAVE_IDF_TARGET_ESP32C2(=n) "esp32c2"
SLAVE_IDF_TARGET_ESP32C3(=n) "esp32c3"
SLAVE_IDF_TARGET_ESP32C5(=n) "esp32c5"
SLAVE_IDF_TARGET_ESP32C6(=n) "esp32c6"                   <-------------- not here?
SLAVE_IDF_TARGET_ESP32C61(=n) "esp32c61"
SLAVE_IDF_TARGET_ESP32S2(=n) "esp32s2"
SLAVE_IDF_TARGET_ESP32S3(=n) "esp32s3"
```

Thanks

chegewara, 2025-01-22:
Another conflict example is between esp-idf wifi component and esp_wifi_remote component when using it with esp32-p4
```
info: INFO: Choice ESP_WIFI_MGMT_RX_BUFFER defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:132
    /home/chegewara/programming/open-source/esp32P4_projects/lvgl-test1/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.wifi.in:110
info: INFO: Choice ESP_WIFI_TASK_CORE_ID defined in multiple locations (see below). Please check if this is a correct behavior or a random name match:
    /home/chegewara/programming/open-source/esp32P4_projects/lvgl-test1/managed_components/espressif__esp_wifi_remote/./idf_v5.5/Kconfig.wifi.in:206
    /home/chegewara/programming/esp-idf/master/components/esp_wifi/Kconfig:228
```
In this case i believe some Kconfig checks are missing in esp-idf wifi component. There is more like this.

david-cermak, 2025-01-30:
Thanks for reporting, we will look into this. The issue is that symbols defined under native Wi-Fi component get duplicated by the injected symbols from the wifi-remote component. The definitions are mutually exclusive (i.e. either defined inside different `#if` blocks which are not enabled for one config), but kconfiglib doesn't account for it.
We will try to address it in the build tools or within the wifi-remote component. sorry for the inconvenience.

Honza0297, 2025-02-04:
Hello @chegewara! Just to complement my colleagues, this is caused by the new esp-idf-kconfig functionality introduced in https://github.com/espressif/esp-idf-kconfig/releases/tag/v2.4.0, reporting potential issues when one config name is defined in multiple locations. I am already working on a better output of this, sorry for the inconvenience.

chegewara, 2025-02-04:
Hi,
i dont complain seeing when i find bugs/issues, it is normal with that complex framework, especially when so many people working on it.
Like i said, i am completely noob in kconfig thats why my concerns
- if its possible and how to solve such situations
- which config is used when we have 2 versions, especially if one is inactive, like with esp_hosted and esp-idf wifi

Thanks

PS the problem i have is espressif team attitude sometimes
i dont pay for support, like other clients, but i am spending a lot of time trying to find root cause to describe it best and help to solve problems
its hard to believe but my time costs too, and nobody is paying me for it

Honza0297, 2025-02-05:
I'll try to answer you in detail. 

**TLDR**:

> if its possible

Yes, configuration system per-se allows multiple config name definition. But it is very error prone and we would like to highlight situations when it happens. 

> how to solve this situations 

As a user; report to the affected component and let its developers to fix it. 
As a component developer; try to rewrite Kconfig in a way that it does not need this.

> which config is used when we have 2 versions... 

It depends on the order in which Kconfig files are loaded. The process is deterministic, but unclear for humans. Multiple definitions are "half-merged" together (e.g. help and prompt are preserved for every definition). 

> ...especially if one is inactive

That's the problem. In parse time, we generally cannot tell which config option is inactive and which value (from which location) it will get. To be extra safe, I opted to report all multiple definitions, which unfortunately flooded the output. 
In the time of writing configuration to e.g. sdkconfig file, configuration system first looks to the **last** location where config option was defined and picks first default value whose condition is met. If none of the default values in the last location have their conditions met, it moves to the  penultimate location etc. 

**Detailed answer:**

I'll start very widely;  kconfig configuration system per-se allows multiple definitions of the same symbol name and there may be use cases which cannot be (easily) solved by other means (e.g. you need to redefine a config option from third party component for your specific use). 
However, it may happen that two or more components define the same config option unintentionally with e.g. different default values.  Because configuration option can have only one value in the given time, configuration system needs to resolve it somehow. Kconfig system first looks to the **last** location where config option was defined and picks first default value whose condition is met. If none of the default values in the last location have their conditions met, it moves to the  penultimate location etc. 

The process of which Kconfig file is loaded in which order (and thus which location will be the last one) is set by the relationships between Kconfig files (via `source` entries) and is deterministic, but unclear for the user. 
To highlight the situation when one config option is defined in multiple locations, the aforementioned check has been implemented. However, in some places, multiple definition of the same config option are used intentionally and a flood of info statements is produced. 

To prevent this, I'll implement a a way how the component developers will be able to mark some config options as allowed to be set multiple times. Then, they will have a mean to restrict the number of these info statements to the minimum. 

I hope I answered your question sufficiently. All of this will be described in greater detail in the [esp-idf-kconfig documentation](https://docs.espressif.com/projects/esp-idf-kconfig/en/latest/) once I'll find time.  If you have further questions, we can definitely discuss the topic further, but probably not in this issue as we are drifting from the merit. :)

chegewara, 2025-02-22:
Hi team,
looks like it works nicely with esp_wifi_remote#0.7.0 component.

Thanks

closed 2025-02-22 by chegewara