espressif/esp-idf #15248: (components/fatfs): possible array index out of bounds

safocl, 2025-01-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The functions in _**components/fatfs/diskio/diskio_wl.c**_ potentially may produce access to out of bounds of **ff_wl_handles** array:
https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/fatfs/diskio/diskio_wl.c#L35
**pdrv** is a function argument and it is not checked to be **less** than **FF_VOLUMES** value (that puts from the configuration files).

safocl, 2025-01-21:
this check exists here.
https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/fatfs/diskio/diskio_wl.c#L86

RathiSonika, 2025-01-28:
Hello @safocl, While using wear leveling, it must be registered through the diskio layer using the [ff_diskio_register_wl_partition()](https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/components/fatfs/diskio/diskio_wl.c#L86) function, where the necessary checks are implemented. During this registration process, the [ff_diskio_get_drive()](https://github.com/espressif/esp-idf/blob/master/components/fatfs/diskio/diskio.c#L54) function is called to obtain the pdrv, which will only be assigned if it is available (and within range of FF_VOLUMES). The ff_wl_read(),ff_wl_write() functions are internally invoked when using POSIX functions like read and write. In these cases, you don't directly pass the pdrv—it is managed internally based on the file path, ensuring the out-of-bounds issue is avoided. You can refer to the [example](https://github.com/espressif/esp-idf/blob/0f0068fff3ab159f082133aadfa9baf4fc0c7b8d/examples/storage/fatfs/fatfsgen/main/fatfsgen_example_main.c) for a clearer understanding of how this works.

safocl, 2025-01-28:
@RathiSonika  these functions are not `static` - maybe they should be marked as `static`?

RathiSonika, 2025-01-28:
Yes, it should be static and it will be updated accordingly. Thanks!

closed 2025-02-13 by espressif-bot