espressif/esp-idf #15249: IDF 5.4 release version failed to build libbtdm_app.a for ESP32S3, cannot link

aircable, 2025-01-21:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-300-gc8bb53292d

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Compiling Bluetooth apps for the ESP32S3 target should succeed.

### What is the actual behavior?

At the end of the build I get this error:
`xtensa-esp-elf/bin/ld: cannot find -lbtdm_app: No such file or directory`

Turns out, the files in the ESP-IDF are missing, they have not been compiled for the ESP32S3 target:
```
components/bt/controller/lib_esp32c3_family/esp32c3/libbtdm_app_flash.a
...esp32c3/libbtdm_app.a
...esp32s3/libbtdm_app.a
...esp32s3/libbtdm_app_flash.a
```

My previous version v5.4-dev-4375-gf420609c33 does create these files and the link succeeds.

### Steps to reproduce.

use a Bluetooth app from the example
idf.py set-target esp32s3
idf.py build
error in linker

### Build or installation Logs.

```plain
`xtensa-esp-elf/bin/ld: cannot find -lbtdm_app: No such file or directory`
```

### More Information.

Does not work on 5.4 release, does not work on 5.5 master (different error)

esp-zhp, 2025-01-23:
(base) zhanghaipeng@FA002031:~/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server$ git log
commit c8bb53292d08d6449a09823cf554e62ac839cd8c (HEAD)
Merge: cf1692c96c f96a118ebf
Author: Mahavir Jain <mahavir@espressif.com>
Date:   Mon Jan 13 15:57:45 2025 +0800

    Merge branch 'fix/return_esp_err_t_for_httpd_req_get_url_query_str_v5.4' into 'release/v5.4'
    
    feat(https_server): Added checks to verify if uri is empty (v5.4)
    
    See merge request espressif/esp-idf!36286

commit cf1692c96c615a5dabf66d4355a7b4accfb6084b
Merge: c6637ae369 605fb6e894
Author: morris <maoshengrong@espressif.com>
Date:   Mon Jan 13 15:12:26 2025 +0800

    Merge branch 'bugfix/reset_axi_gdma_gracefully_v5.4' into 'release/v5.4'
    
    fix(gdma): stop the axi gdma gracefully on CPU SW reset (v5.4)
    
    See merge request espressif/esp-idf!36283

commit c6637ae3691baf75d92b87b4608c615262baddc6
Merge: 1ceb4e4682 3f4dd2bf35
Author: Rahul Tank <rahul.tank@espressif.com>
Date:   Mon Jan 13 13:24:50 2025 +0800
(base) zhanghaipeng@FA002031:~/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server$ idf.py set-target esp32s3;idf.py build
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/zhanghaipeng/.espressif/python_env/idf5.4_py3.11_env/bin/python -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=0 /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server"...
-- Existing sdkconfig '/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig' renamed to '/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.old'.
-- Found Git: /usr/bin/git (found version "2.17.1") 
-- The C compiler identification is GNU 14.2.0
-- The CXX compiler identification is GNU 14.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
-- Project sdkconfig file /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig
Loading defaults file /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.defaults...
warning: unknown kconfig symbol 'BT_LE_50_FEATURE_SUPPORT' assigned to 'n' in /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.defaults
Loading defaults file /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.defaults.esp32s3...
warning: unknown kconfig symbol 'BT_LE_50_FEATURE_SUPPORT' assigned to 'n' in /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.defaults.esp32s3
-- Compiler supported targets: xtensa-esp-elf
-- Found Python3: /home/zhanghaipeng/.espressif/python_env/idf5.4_py3.11_env/bin/python (found version "3.11.4") found components: Interpreter 
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "gatt_server_demos" version: v5.4-300-gc8bb53292d
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build/esp-idf/esp_system/ld/sections.ld.in
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.api.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.bt_funcs.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32s3/ld/esp32s3.rom.libgcc.ld
[... 57 lines cut ...]
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
Generated /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build/bootloader/bootloader.bin
[115/115] cd /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blued.../bluetooth/bluedroid/ble/gatt_server/build/bootloader/bootloader.bin
Bootloader binary size 0x5210 bytes. 0x2df0 bytes (36%) free.
[1346/1347] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32s3 image...
Merged 2 ELF sections
Successfully created esp32s3 image.
[... cut ...]
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 80m 0x0 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/gatt_server_demos.bin
or from the "/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/build" directory
 python -m esptool --chip esp32s3 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
(base) zhanghaipeng@FA002031:~/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server$

esp-zhp, 2025-01-23:
@aircable 
It seems like there might be an issue with your IDF environment. Could you try reinstalling the IDF and testing again?  
I didn't encounter any issues during compilation on my side. Above is the log from my compilation.

chegewara, 2025-01-23:
```
warning: unknown kconfig symbol 'BT_LE_50_FEATURE_SUPPORT' assigned to 'n' in /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/bluedroid/ble/gatt_server/sdkconfig.defaults.esp32s3
```
which depends on
```
Direct dependencies (=y):
     BT_BLE_ENABLED(=y)
  && (BT_CONTROLLER_ENABLED(=y) && SOC_BLE_50_SUPPORTED(=y)) || BT_CONTROLLER_DISABLED(=n)  (=y)
  && BT_BLUEDROID_ENABLED(=y)
```

This log is suggesting that BT/BLE is disabled or misconfigured for some reason.
You should check menuconfig settings.

closed 2025-05-08 by Alvin1Zhang