espressif/esp-idf #15252: ESP-I2C Occasionally Invalid Readings

elecrain, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

For a project, I'm using an RTC IC with I2C. Sometimes, esp reads wrong values from registers. I've made a test with logic analyzer and osciloscope and esp logs. Evertime esp makes a reading, it prints out the 6 register and the return of the i2c_master_write_read_device function. When the problem occurs, oscilloscope and logic anayzer matches out but esp prints are different. Esp prints out that it reads 1 byte different. 

This problem only occurs in every 3-5 Hours. What might be the reason ? 

[image]

mythbuster5, 2025-01-22:
esp32?

elecrain, 2025-01-22:
Yes, I'm using ESP32-WROVER-E.

mythbuster5, 2025-01-22:
I guess you encounter the same issue with https://github.com/espressif/esp-idf/issues/12860 please try patch there,.

elecrain, 2025-01-22:
Yes, it looks like we have the same issue with #12860. I've made the code update on source file and restarted the test process. I will be updating here if the problem persist to continue. Thanks!

mythbuster5, 2025-01-23:
Any result?

elecrain, 2025-01-23:
Yesterday, I've had a problem with test setup, I've started to testing today and its being running for 3 hours with no problem. But i will be keep testing for at least a couple of days.

closed 2025-02-03 by elecrain