espressif/esp-idf #15253: [ble-mesh] [esp32-p4] some features are not working

chegewara, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi team,

recently esp-hosted component has been released with bluedroid support. I made some quick testing to check if it is working with bluedroid ble mesh and mostly it is working, so i am happy with it.

I found 2 features that are not working:
- `BT_BLE_50_FEATURES_SUPPORTED` - im not sure if its general issue or just with ble mesh; when this option is enabled then after calling esp_ble_mesh_node_prov_enable i get this error
```
E (3878) BT_HCI: CC evt: op=0x200b, status=0xc
E (3888) BLE_MESH: start_le_scan, Invalid status 12
E (3888) BLE_MESH: starting scan failed (err -1)
```
- ble mesh v1.1 cant be enabled due to many `undefined reference to bt_mesh_prov_XXX` errors

Thanks

forx157, 2025-01-23:
Hi, @chegewara 

Thanks for your report!
could you share us with your compile log and sdkconfig?

chegewara, 2025-01-23:
Hi @forx157 
more or less this is sdkconfig.defaults
```
# This file was generated using idf.py save-defconfig. It can be edited manually.
# Espressif IoT Development Framework (ESP-IDF) 5.5.0 Project Minimal Configuration
#
CONFIG_IDF_TARGET="esp32p4"
CONFIG_ESPTOOLPY_FLASHSIZE_16MB=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_BT_ENABLED=y
CONFIG_BT_BTU_TASK_STACK_SIZE=4512
CONFIG_BT_GATTS_SEND_SERVICE_CHANGE_MANUAL=y
CONFIG_BT_ALLOCATION_FROM_SPIRAM_FIRST=y
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BT_BLE_50_FEATURES_SUPPORTED=n
CONFIG_BLE_MESH=y
CONFIG_BLE_MESH_V11_SUPPORT=n                                                  # this is causing build problems when its is Y
CONFIG_BLE_MESH_NODE=y
CONFIG_BLE_MESH_PROVISIONER=y
CONFIG_BLE_MESH_PB_GATT=y
CONFIG_BLE_MESH_TX_SEG_MSG_COUNT=10
CONFIG_BLE_MESH_RX_SEG_MSG_COUNT=10
CONFIG_BLE_MESH_CFG_CLI=y
CONFIG_BLE_MESH_GENERIC_ONOFF_CLI=y
CONFIG_BLE_MESH_GENERIC_LEVEL_CLI=y
CONFIG_BLE_MESH_GENERIC_SERVER=y
CONFIG_SPIRAM=y
CONFIG_SPIRAM_SPEED_200M=y
CONFIG_ESP_WIFI_EXTRA_IRAM_OPT=n
CONFIG_ESP_WIFI_SLP_IRAM_OPT=n
CONFIG_ESP_ENABLE_BT_BLUEDROID=y
CONFIG_ESP_MAX_SIMULTANEOUS_SYNC_RPC_REQUESTS=15
CONFIG_ESP_MAX_SIMULTANEOUS_ASYNC_RPC_REQUESTS=15
CONFIG_SLAVE_IDF_TARGET_ESP32C6=y
CONFIG_IDF_EXPERIMENTAL_FEATURES=y
```
logs with V11 support enabled (stripped) - looks like precompiled binary for ble mesh v1.1 is not linked
```
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/net.c:1617:(.text.bt_mesh_net_relay+0x3e): undefined reference to `bt_mesh_net_retrans_match'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/net.c:1215:(.text.bt_mesh_net_send+0x68): undefined reference to `bt_mesh_tag_relay'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/transport.c:678:(.text.bt_mesh_trans_send+0xc2): undefined reference to `bt_mesh_tag_send_segmented'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/transport.c:1180:(.text.bt_mesh_ctl_send+0x4c): undefined reference to `bt_mesh_tag_send_segmented'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/access.c:363:(.text.mod_init+0xa8): undefined reference to `comp_0'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/access.c:981:(.text.model_send+0x66): undefined reference to `bt_mesh_valid_security_cred'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/access.c:48:(.text.bt_mesh_model_foreach+0x12): undefined reference to `comp_0'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/access.c:248:(.text.bt_mesh_model_elem+0x0): undefined reference to `comp_0'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/cfg_srv.c:234:(.text.comp_data_get+0x18): undefined reference to `bt_mesh_comp_page_check'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/prov_node.c:478:(.text.send_confirm+0x18a): undefined reference to `bt_mesh_prov_conf_salt_epa'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/core/prov_node.c:790:(.text.prov_random+0x152): undefined reference to `bt_mesh_prov_conf_epa'
/home/chegewara/programming/esp-idf/master/components/bt/esp_ble_mesh/lib/ext.c:4947:(.text.bt_mesh_v11_ext_init+0x1c): undefined reference to `bt_mesh_v11_init'
```

There is running log when i enable `BT_BLE_50_FEATURES_SUPPORTED` 
```
I (1844) transport: Slave chip Id[12]
I (1844) vhci_drv: Host BT Support: Enabled
I (1844) vhci_drv:      BT Transport Type: VHCI
I (1854) H_SDIO_DRV: Received INIT event
I (2674) EXAMPLE: ESP_BLE_MESH_PROV_REGISTER_COMP_EVT, err_code 0
I (2674) EXAMPLE: BLE Mesh Node initialized
E (2674) BT_HCI: CC evt: op=0x200b, status=0xc
E (2684) BLE_MESH: start_le_scan, Invalid status 12
E (2684) BLE_MESH: starting scan failed (err -1)
I (2684) EXAMPLE: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code -1
I (2694) main_task: Returned from app_main()
```

I have BLE 5.0 support enabled on C6 esp hosted slave (by default).

Thanks

forx157, 2025-01-23:
Hi, @chegewara 
Sorry for the error, but this is due to the fact that the ble mesh library does not yet support esp32p4, so mesh 1.1 functionality is not currently available on esp32p4, but mesh 1.0 functionality is not affected. In addition, if you really need it for your development, we will provide you with a preview version after the Chinese New Year holiday for you to use.

Regarding the errors generated after turning on ble 5.0, you can turn on the option BLE_MESH_USE_BLE_50 (currently only supported on esp32c3, esp32s3 platforms in the master branch), and of course, if you really need ble 5.0 for your development, we can also provide you with a patch after the Chinese New Year holiday to facilitate your testing.

chegewara, 2025-01-23:
Hi @forx157 
no worries, i am just playing with it and i only wanted to test esp hosted with bluedroid ble mesh, because nimble does not support v1.1 features.

At the moment i dont have particular needs for it (no clients so far)so i will play with features that are available.

Regarding BLE 50, i thought all chips series C, including C6, should support it (not just C2 and S3).

Thanks for support

chegewara, 2025-01-23:
BTW there may be few features missing on P4 (and with any hosted based app) because they are in BT controller and API is not available, like
```
esp_ble_tx_power_set
```

Thanks

forx157, 2025-01-23:
Hi, @chegewara 
BLE Mesh on BLE 5.0 functionality will be available on the esp32c6, esp32h2, esp32c61, esp32c5 platforms in the near future.

about your question:
```
BTW there may be few features missing on P4 (and with any hosted based app) because they are in BT controller and API is not available, like

esp_ble_tx_power_set
Thanks
```
I'll invite others to answer this question.

jwh-hutchison, 2025-12-01:
Is there any update on adding the `esp_ble_tx_power_set` feature?

Thanks for support