espressif/esp-idf #15256: 在使用i2s_std进行音频输入时出现错误

rfRedfeng, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3.2

### Espressif SoC revision.

ESP32-S3(QFN56)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

无

### Power Supply used.

USB

### What is the expected behavior?

通过I2S接口获取INP441的I2S数据进行解析音频数据，后将音频数据发送到串口，用上位机实现播放获取到的音频信息

### What is the actual behavior?

出现乱码与电平报错，通过声音保存进行声纹分析发现存在严重的削波失真，怀疑数据出现移位，并尝试进行修复后

### Steps to reproduce.

1.在默认编写时采用了官方推荐的I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO)宏定义进行初始化设置
```c
void i2s_mic_init() {
    // 配置I2S通道
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER);
    chan_cfg.auto_clear = true;
    chan_cfg.dma_desc_num =  8; 
    chan_cfg.dma_frame_num = bufferlen;
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, NULL, &rx_chan_handle));
    // 标准I2S配置
    i2s_std_config_t std_cfg = {
        .clk_cfg = {
            .sample_rate_hz = I2S_SAMPLE_RATE,
            .clk_src = I2S_CLK_SRC_PLL_160M, // 设置时钟源
            .mclk_multiple = I2S_MCLK_MULTIPLE_384,
        },
        .slot_cfg = I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO),
        .gpio_cfg = {
            .bclk = I2S_BCK,
            .ws = I2S_WS,
            .din = I2S_DATA,
            .mclk = I2S_GPIO_UNUSED,
            .dout = I2S_GPIO_UNUSED,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_STEREO; // 立体声模式
    std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_LEFT;    // 仅左声道有效
    // 初始化I2S
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan_handle, &std_cfg));
    ESP_ERROR_CHECK(i2s_channel_enable(rx_chan_handle));
} 

void app_main(void) {

    uart_config_t uart_cfg = {
        .baud_rate = 921600,
        .data_bits = UART_DATA_8_BITS,
        .parity = UART_PARITY_DISABLE,
        .stop_bits = UART_STOP_BITS_1,
        .flow_ctrl = UART_HW_FLOWCTRL_DISABLE,
    };
    uart_param_config(UART_NUM_0, &uart_cfg);
    uart_driver_install(UART_NUM_0, 4096, 0, 0, NULL, 0);

    i2s_mic_init();

    while (1) {

        esp_err_t res = i2s_channel_read(rx_chan_handle, &data_buf, bufferlen * sizeof(int16_t) , &bytes_read, portMAX_DELAY);
        int16_t muestras = bytes_read / 2;
        if (res == ESP_OK && muestras > 0) {
            uart_write_bytes(UART_NUM_0, buffer, bytes_read);
        }
    }
}
```
期间代码无产生任何的报错，因此我尝试对代码进行修正，在将i2s由I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO)改为手动配置后，现象解除音频正常产生。
```c
void i2s_mic_init() {
    // 配置I2S通道
    i2s_chan_config_t chan_cfg = I2S_CHANNEL_DEFAULT_CONFIG(I2S_NUM_0, I2S_ROLE_MASTER);
    chan_cfg.auto_clear = true;
    chan_cfg.dma_desc_num =  8; // DMA缓冲区数量
    chan_cfg.dma_frame_num = bufferlen; // 每个DMA缓冲区的长度
    ESP_ERROR_CHECK(i2s_new_channel(&chan_cfg, NULL, &rx_chan_handle));

    // 标准I2S配置
    i2s_std_config_t std_cfg = {
        .clk_cfg = {
            .sample_rate_hz = I2S_SAMPLE_RATE,
            .clk_src = I2S_CLK_SRC_PLL_160M, // 直接设置时钟源
            .mclk_multiple = I2S_MCLK_MULTIPLE_384,
        },
        .slot_cfg = {
        .data_bit_width = I2S_DATA_BIT_WIDTH_16BIT,
        .slot_bit_width = I2S_SLOT_BIT_WIDTH_16BIT, 
        .slot_mode = I2S_SLOT_MODE_STEREO,
        .slot_mask = I2S_STD_SLOT_LEFT,
        .ws_width = 16,               
        .ws_pol = false,             
        .bit_shift = true,            
    },
    .gpio_cfg = {
            .bclk = I2S_BCK,
            .ws = I2S_WS,
            .din = I2S_DATA,
            .mclk = I2S_GPIO_UNUSED,
            .dout = I2S_GPIO_UNUSED,
            .invert_flags = {
                .mclk_inv = false,
                .bclk_inv = false,
                .ws_inv = false,
            },
        },
    };
    std_cfg.slot_cfg.slot_mode = I2S_SLOT_MODE_STEREO; // 立体声模式
    std_cfg.slot_cfg.slot_mask = I2S_STD_SLOT_LEFT;    // 仅左声道有效
    // 初始化I2S
    ESP_ERROR_CHECK(i2s_channel_init_std_mode(rx_chan_handle, &std_cfg));
    ESP_ERROR_CHECK(i2s_channel_enable(rx_chan_handle));
}
```

### Debug Logs.

```plain

```

### More Information.

再代码执行成功后我将怀疑的目光投向I2S_STD_PHILIPS_SLOT_DEFAULT_CONFIG(I2S_DATA_BIT_WIDTH_16BIT, I2S_SLOT_MODE_MONO)，通过定义发现其在定义中将 .slot_bit_width 设置为了 I2S_SLOT_BIT_WIDTH_AUTO 及自动定义其对应段的槽位宽度，观察i2s_channel_init_std_mode()设置函数，其在对于I2S_SLOT_BIT_WIDTH_AUTO 处理中采用了当其配置为自动时，将其等于数据位宽
‘’‘c
//  i2s_std.c 328行
    /* If the slot bit width changed, then need to update the clock */
    uint32_t slot_bits = slot_cfg->slot_bit_width == I2S_SLOT_BIT_WIDTH_AUTO ? slot_cfg->data_bit_width : slot_cfg->slot_bit_width;
    if (std_cfg->slot_cfg.slot_bit_width == slot_bits) {
        ESP_GOTO_ON_ERROR(i2s_std_set_clock(handle, &std_cfg->clk_cfg), err, TAG, "update clock failed");
    }
’‘’
但在之后的数据判断时存在逻辑矛盾 此条件判断 槽位宽度是否等于自动计算后的值，但若当前配置的 slot_bit_width 是 AUTO，则 std_cfg->slot_cfg.slot_bit_width 的值是 I2S_SLOT_BIT_WIDTH_AUTO，而 slot_bits 是计算出的实际位宽，两者永远不相等。使得此条件永远为 false，导致时钟更新逻辑无法触发。在我尝试在if前直接使用时钟设置i2s_std_set_clock(handle, &std_cfg->clk_cfg)，代码能够正常运作

希望官方可以尽快修正这个bug，谢谢

L-KAYA, 2025-01-23:
@rfRedfeng 非常感谢您指出该问题！

检查驱动后这里确实存在两处错误：
1. 初始化阶段保存配置信息时，仅将`slot_bit_width`保存为`AUTO`，并未更新成实际真实值，导致此处取值时取到的是 `AUTO` (即`0`)；
2. 此处 `if` 的逻辑应为 `if (std_cfg->slot_cfg.slot_bit_width != slot_bits)`，而不是 `==`；

顺便询问下，您是在后面调用了 `i2s_channel_reconfig_std_slot` 并更改了 `data_width` 是吗？因为这个逻辑是在 reconfig 中的，而不是 init 中。如果没有调用 `i2s_channel_reconfig_std_slot` 或者调用后未改变数据位宽应该不会遇到这个问题。

## 修复补丁

如果方便的话可以试一下这个修复补丁是否有效

[0001-fix-i2s-fixed-incorrect-logic-in-slot-reconfig.patch](https://github.com/user-attachments/files/18515157/0001-fix-i2s-fixed-incorrect-logic-in-slot-reconfig.patch)

rfRedfeng, 2025-01-23:
是的，我在后面进行调试时使用了 `i2s_channel_reconfig_std_slot` 进行了动态修改测试产生了该问题，即`data_width` 一直无法被正常的运用

closed 2025-01-23 by rfRedfeng