espressif/esp-idf #15257: How to use esp_intr_alloc_intrstatus to register an ISR on HP core woken up by LP core

pat-bert, 2025-01-22:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Dear Espressif Team,

I'm working with the ESP32C6 and ESP-IDF5.4 to read some sensor measurements via the LP I2C.
The LP core is notifying the HP core via `ulp_lp_core_wakeup_main_processor()` when it is done and waits for 1 min.
Currently I'm only using light sleep mode.

According to the [technical reference manual](https://www.espressif.com/sites/default/files/documentation/esp32-c6_technical_reference_manual_en.pdf#subsection.12.8) this should trigger a PMU interrupt that can be routed via the interrupt matrix to a CPU software interrupt.

I'm trying to register an interrupt callback for handling the notification with the following code:

```
static void isr(void *arg)
{
    gpio_set_direction(GPIO_NUM_15, GPIO_MODE_OUTPUT);
    gpio_set_level(GPIO_NUM_15, 1);
}

extern "C" void app_main(void)
{
        intr_handle_t interruptHandle{};
        ESP_ERROR_CHECK(esp_intr_alloc_intrstatus(ETS_PMU_INTR_SOURCE, 0, PMU_HP_LP_CPU_COMM_REG, PMU_LP_TRIGGER_HP, isr, nullptr, &interruptHandle));
        ESP_ERROR_CHECK(esp_intr_enable(interruptHandle));
        esp_intr_enable_source(ETS_PMU_INTR_SOURCE);
        ESP_ERROR_CHECK(esp_sleep_enable_ulp_wakeup());

        ESP_ERROR_CHECK(ulp_lp_core_load_binary(ulp_lp_sensor_bin_start, (ulp_lp_sensor_bin_end - ulp_lp_sensor_bin_start)));

        /* Start the program */
        ulp_lp_core_cfg_t cfg{};
        cfg.wakeup_source = ULP_LP_CORE_WAKEUP_SOURCE_HP_CPU;

        ESP_ERROR_CHECK(ulp_lp_core_run(&cfg));

        ... // start other FreeRTOS tasks

        vTaskSuspend(nullptr);
}
```

I already verified that the LP-core is running correctly by
- polling a shared variable on the HP-core and reading the measurements
- registering a light sleep exit callback via `esp_pm_light_sleep_register_cbs()` to handle the measurements

My understanding is that registering an ISR would be more flexible since I can already read measurements before the HP-core actually goes to sleep. Currently with using the light sleep exit callback I see a delay until the first measurement arrives because the HP is still busy with other tasks and doesn't go to sleep immediately.

Is my understanding of `esp_intr_alloc_intrstatus` incorrect or do I need to do some additional steps to enable the interrupt?
The programming guide unfortunately does not provide an example for using a software interrupt.

ESP-Marius, 2025-01-23:
Hi,

You are almost there, but missed a few points:
* `esp_intr_alloc_intrstatus` takes the interrupt status register and mask, which is PMU_HP_INT_ST_REG, PMU_SW_INT_ST in this case. 
* `esp_intr_enable_source(ETS_PMU_INTR_SOURCE);` is not needed

Now the interrupt allocator is setup correctly, but this only takes care of the interrupt matrix and the CPU side of things, you also have to configure the peripheral that will trigger the interrupt (the PMU in our case).
To do this simply enable the interrupt with ` REG_SET_BIT(PMU_HP_INT_ENA_REG, PMU_SW_INT_ENA);`

Remember to also clear it in your interrupt handler with `REG_SET_BIT(PMU_HP_INT_CLR_REG, PMU_SW_INT_CLR);`

> The programming guide unfortunately does not provide an example for using a software interrupt.

From the CPU/interrupt matrix point of view there isnt really a difference between an interrupt that is triggered by HW vs writing to a register, it is all the same.

Hope this is enough to get it working for you.

pat-bert, 2025-01-23:
Hi,

Thanks a lot for the fast and detailed support! 
That solved my issue immediately.

closed 2025-01-23 by pat-bert