espressif/esp-idf #15258: ESP32-S3 using CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN

thekurtovic, 2025-01-22:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.3.2.241210

### Espressif SoC revision.

ESP32-S3 (QFN56) (revision v0.1)

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

Allow scanning while a connect is in progress when enabling `CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN` option.

### What is the actual behavior?

Scanning does not stop, but no new advertisements are received after initiating connection. Scan cannot be stopped or restarted.

### Steps to reproduce.

-Test with https://github.com/h2zero/esp-nimble-cpp/tree/master/examples/NimBLE_Async_Client
-Enable CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN

### Debug Logs.

<details>
<summary>Async example CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN=y</summary>

```
Starting NimBLE Async Client
I (428) BLE_INIT: BT controller compile version [edff442]
I (429) BLE_INIT: Bluetooth MAC: 34:85:18:45:43:6e
I (431) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (475) NimBLE: GAP procedure initiated: stop advertising.

I (478) NimBLE: GAP procedure initiated: discovery;
I (478) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (485) NimBLE: duration=5000ms
I (489) NimBLE:

Advertised Device found: Name: , Address: d1:bb:4d:b5:30:72, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
I (1689) NimBLE: GAP procedure initiated: connect;
I (1694) NimBLE: peer_addr_type=1 peer_addr=
I (1699) NimBLE: d1:bb:4d:b5:30:72
I (1703) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (1717) NimBLE:

Connected to: d1:bb:4d:b5:30:72
peer address: d1:bb:4d:b5:30:72
Services:

I (7585) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (7587) NimBLE: GAP procedure initiated: discovery;
I (7593) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (7602) NimBLE: duration=5000ms
I (7605) NimBLE:

E (7608) NimBLE: 0x20C : BLE_ERR_CMD_DISALLOWED (Command Disallowed)

d1:bb:4d:b5:30:72 Disconnected, reason = 534 - Starting scan
I (7631) NimBLE: GAP procedure initiated: discovery;
I (7631) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (7640) NimBLE: duration=5000ms
I (7644) NimBLE:

E (7647) NimBLE: 0x20C : BLE_ERR_CMD_DISALLOWED (Command Disallowed)
```

</details>

<details>
<summary>Async example CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN=n</summary>

```
Starting NimBLE Async Client
I (428) BLE_INIT: BT controller compile version [edff442]
I (429) BLE_INIT: Bluetooth MAC: 34:85:18:45:43:6e
I (431) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (474) NimBLE: GAP procedure initiated: stop advertising.

I (477) NimBLE: GAP procedure initiated: discovery;
I (477) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (484) NimBLE: duration=5000ms
I (488) NimBLE:

Advertised Device found: Name: SHIELD, Address: 68:dd:58:d5:52:0b, manufacturer data: ffff65636635323533326465643734323730, serviceUUID: 0x3456, txPower: -15
Advertised Device found: Name: , Address: d0:1e:c7:d8:b5:4f, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
I (3038) NimBLE: GAP procedure initiated: connect;
I (3043) NimBLE: peer_addr_type=1 peer_addr=
I (3048) NimBLE: d0:1e:c7:d8:b5:4f
I (3052) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (3065) NimBLE:

Connected to: d0:1e:c7:d8:b5:4f
peer address: d0:1e:c7:d8:b5:4f
Services:

I (3491) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (3494) NimBLE: GAP procedure initiated: discovery;
I (3499) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (3508) NimBLE: duration=5000ms
I (3512) NimBLE:

d0:1e:c7:d8:b5:4f Disconnected, reason = 534 - Starting scan
I (3531) NimBLE: GAP procedure initiated: discovery;
I (3531) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (3539) NimBLE: duration=5000ms
I (3543) NimBLE:

Advertised Device found: Name: , Address: d1:bb:4d:b5:30:72, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
I (4043) NimBLE: GAP procedure initiated: connect;
I (4048) NimBLE: peer_addr_type=1 peer_addr=
I (4053) NimBLE: d1:bb:4d:b5:30:72
I (4057) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (4071) NimBLE:

Connected to: d1:bb:4d:b5:30:72
peer address: d1:bb:4d:b5:30:72
Services:

I (4515) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (4518) NimBLE: GAP procedure initiated: discovery;
I (4523) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (4532) NimBLE: duration=5000ms
I (4536) NimBLE:

d1:bb:4d:b5:30:72 Disconnected, reason = 534 - Starting scan
I (4544) NimBLE: GAP procedure initiated: discovery;
I (4549) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (4558) NimBLE: duration=5000ms
I (4562) NimBLE:

Advertised Device found: Name: SHIELD, Address: 68:dd:58:d5:52:0b, manufacturer data: ffff65636635323533326465643734323730, serviceUUID: 0x3456, txPower: -15
Advertised Device found: Name: , Address: d1:bb:4d:b5:30:72, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
I (4868) NimBLE: GAP procedure initiated: connect;
I (4873) NimBLE: peer_addr_type=1 peer_addr=
I (4878) NimBLE: d1:bb:4d:b5:30:72
I (4882) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (4896) NimBLE:

Connected to: d1:bb:4d:b5:30:72
peer address: d1:bb:4d:b5:30:72
Services:

I (5566) NimBLE: GAP procedure initiated: terminate connection; conn_handle=1 hci_reason=19

I (5569) NimBLE: GAP procedure initiated: discovery;
I (5574) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (5583) NimBLE: duration=5000ms
I (5587) NimBLE:

d1:bb:4d:b5:30:72 Disconnected, reason = 534 - Starting scan
I (5596) NimBLE: GAP procedure initiated: discovery;
I (5601) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (5609) NimBLE: duration=5000ms
```

</details>

### More Information.

I'm using NimBLE to handle BLE in my project. https://github.com/h2zero/esp-nimble-cpp
The es32-s3 acts as a central, and will constantly be scanning for peripheral devices. There are 4 peripherals avilable. When a peripheral is found, the esp32-s3 will connect to it, and should continue scanning while the connect operation is in progress. Currently when `CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN` is enabled, the esp32-s3 will only get one advertisement, attempt connecting, and then no new advertisements are received. If I go back and disable the config option, everything works fine. 
I'd like to get this option working to test initiating multiple connections at once.

thekurtovic, 2025-01-23:
I've tried testing this again with esp32-c6, also appears to have issues. It will fail to connect immediately, unlike the s3 which could  connect before getting stuck.

<details>
<summary>C6 Async example CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN=y</summary>

```
Starting NimBLE Async Client
I (413) BLE_INIT: Using main XTAL as clock source
I (417) BLE_INIT: ble controller commit:[b623f7a]
I (418) BLE_INIT: Bluetooth MAC: f0:f5:bd:02:63:a2
I (418) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (474) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
I (475) NimBLEDevice: BLE Host Task Started
I (479) NimBLE: GAP procedure initiated: stop advertising.

I (482) NimBLEDevice: NimBle host synced.
D (487) NimBLEScan: >> start: duration=5000
I (491) NimBLE: GAP procedure initiated: discovery;
I (496) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (505) NimBLE: duration=5000ms
I (509) NimBLE:

D (512) NimBLEScan: Scan started
D (515) NimBLEScan: << start()
I (568) NimBLEScan: New advertiser: 78:aa:c7:b2:64:6b
D (568) NimBLEScanCallbacks: Discovered: Name: , Address: 78:aa:c7:b2:64:6b, manufacturer data: ffff65636635323533326465643734323730, serviceUUID: 0x3456, txPower: -15
I (579) NimBLEScan: Scan response from: 78:aa:c7:b2:64:6b
Advertised Device found: Name: SHIELD, Address: 78:aa:c7:b2:64:6b, manufacturer data: ffff65636635323533326465643734323730, serviceUUID: 0x3456, txPower: -15
D (598) NimBLEScan: erase device: 78:aa:c7:b2:64:6b
I (611) NimBLEScan: New advertiser: d1:bb:4d:b5:30:72
D (611) NimBLEScanCallbacks: Discovered: Name: , Address: d1:bb:4d:b5:30:72, manufacturer data: ffff00000000000000000000000000
I (622) NimBLEScan: Scan response from: d1:bb:4d:b5:30:72
Advertised Device found: Name: , Address: d1:bb:4d:b5:30:72, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
D (643) NimBLEClient: >> connect(d1:bb:4d:b5:30:72)
I (649) NimBLE: GAP procedure initiated: connect;
I (655) NimBLE: peer_addr_type=1 peer_addr=
I (659) NimBLE: d1:bb:4d:b5:30:72
I (663) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (677) NimBLE:

E (680) NimBLE: 0x20C : BLE_ERR_CMD_DISALLOWED (Command Disallowed)

E (687) NimBLEClient: Failed to connect to d1:bb:4d:b5:30:72, rc=524;
Failed to connect
D (696) NimBLEScan: erase device: d1:bb:4d:b5:30:72
```

</details>

<details>
<summary>C6 Async example CONFIG_BT_NIMBLE_HOST_ALLOW_CONNECT_WITH_SCAN=n</summary>

```
Starting NimBLE Async Client
I (414) BLE_INIT: Using main XTAL as clock source
I (418) BLE_INIT: ble controller commit:[b623f7a]
I (419) BLE_INIT: Bluetooth MAC: f0:f5:bd:02:63:a2
I (419) phy_init: phy_version 320,348a293,Sep  3 2024,16:33:12
I (473) phy: libbtbb version: 04952fd, Sep  3 2024, 16:33:30
I (475) NimBLEDevice: BLE Host Task Started
I (478) NimBLE: GAP procedure initiated: stop advertising.

I (481) NimBLEDevice: NimBle host synced.
D (486) NimBLEScan: >> start: duration=5000
I (490) NimBLE: GAP procedure initiated: discovery;
I (495) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (504) NimBLE: duration=5000ms
I (508) NimBLE:

[... 99 lines cut ...]
D (3630) NimBLEScan: << stop()
I (3633) NimBLE: GAP procedure initiated: discovery;
I (3639) NimBLE: own_addr_type=0 filter_policy=0 passive=0 limited=0 filter_duplicates=1
I (3648) NimBLE: duration=5000ms
I (3652) NimBLE:

D (3655) NimBLEScan: Scan started
D (3659) NimBLEScan: << start()
D (3662) NimBLEClient: << handleGapEvent
I (3667) NimBLEScan: New advertiser: d0:1e:c7:d8:b5:4f
D (3673) NimBLEScanCallbacks: Discovered: Name: , Address: d0:1e:c7:d8:b5:4f, manufacturer data: ffff00000000000000000000000000
I (3685) NimBLEScan: Scan response from: d0:1e:c7:d8:b5:4f
Advertised Device found: Name: , Address: d0:1e:c7:d8:b5:4f, manufacturer data: ffff00000000000000000000000000, serviceUUID: 93f7de31-03ce-4010-87dd-276f553ce0c6
Found Our Device
D (3707) NimBLEClient: >> connect(d0:1e:c7:d8:b5:4f)
D (3713) NimBLEScan: >> stop()
D (3717) NimBLEScan: << stop()
I (3720) NimBLE: GAP procedure initiated: connect;
I (3725) NimBLE: peer_addr_type=1 peer_addr=
I (3730) NimBLE: d0:1e:c7:d8:b5:4f
I (3734) NimBLE:  scan_itvl=16 scan_window=16 itvl_min=24 itvl_max=40 latency=0 supervision_timeout=256 min_ce_len=0 max_ce_len=0 own_addr_type=0
I (3748) NimBLE:
```

</details>

rahult-github, 2025-01-24:
Hi @thekurtovic ,

I tried the sample bleprph and blecent examples in IDF for v5.3.2 on esp32s3. 

I do see the devices discovered after connection is made.  Am attaching a small change i did in blecent example for this ( i.e. start scanning after a connection is made ). 

Am also attaching sdkconfig on blecent side for your reference. 

can you please check if default IDF examples work for you or not ? 

[scan_changes.txt](https://github.com/user-attachments/files/18532034/scan_changes.txt)  Please apply this patch in $IDF_PATH. 

[scan_after_connect.txt](https://github.com/user-attachments/files/18532044/scan_after_connect.txt)  , this is the log on blecent side where we can see disc events being received after a successful connection. 

[sdkconfig_blecent.txt](https://github.com/user-attachments/files/18532048/sdkconfig_blecent.txt)  blecent sdkconfig for your reference.

thekurtovic, 2025-01-24:
@rahult-github Thanks for confirming that it should work. 
Also sorry for not providing a log with IDF examples. I work with a cpp wrapper library for nimble which has never given me issues like this, but perhaps the issue lies in how things are managed there.
I will test again with the IDF examples and report back.

rahult-github, 2025-03-12:
Hi @thekurtovic  ,  any updats ?

closed 2025-03-12 by thekurtovic