espressif/esp-idf #15265: `esp_lcd_panel_swap_xy()` and `esp_lcd_panel_mirror()` cause ST7789 LCD to tear

yiming564, 2025-01-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4

### Espressif SoC revision.

ESP32S3

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

I successfully initialized my ST7789 LCD (240 * 320) by following the [docs](https://docs.espressif.com/projects/esp-idf/zh_CN/v5.4/esp32s3/api-reference/peripherals/lcd/spi_lcd.html) and I managed to avoid tearing effect by using [triple buffering](https://en.wikipedia.org/wiki/Multiple_buffering#Triple_buffering).

However, after adding these two lines of code, the screen teared badly:

I wanted to rotate the screen, so I added the following code:

```
//	ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel_handle, true));
//	ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel_handle, false, true));
```

The screen should rotate $270 \degree$ and should not tear.

### What is the actual behavior?

The screen rotated successfully, but also tore barly.

I have confirmed that adding anyone of them can cause tearing effect. [Possible reason](https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/lcd_screen_tearing.html#)

Specifically, `esp_lcd_panel_swap_xy()` can cause the screen to tear to triangles, and `esp_lcd_panel_mirror()` can cause the screen to tear vertically.

### Steps to reproduce.

```cpp
static void init_lcd()
{
	ESP_LOGI(app_tag, "Init LCD controller...");
	spi_bus_config_t spi_bus_cfg = {
		.mosi_io_num = GPIO_MOSI,
		.miso_io_num = GPIO_MISO,
		.sclk_io_num = GPIO_SCLK,
		.quadwp_io_num = -1,
		.quadhd_io_num = -1,
		.max_transfer_sz = LCD_H_RES * LCD_V_RES * sizeof(color_t) / 16,
	};
	ESP_ERROR_CHECK(spi_bus_initialize(LCD_HOST, &spi_bus_cfg, SPI_DMA_CH_AUTO));
	esp_lcd_panel_io_spi_config_t io_cfg = {
		.cs_gpio_num = GPIO_CS,
		.dc_gpio_num = GPIO_DC,
		.spi_mode = 0,
		.pclk_hz = LCD_FREQ,
		.trans_queue_depth = 8,
		.on_color_trans_done = trans_done,
		.lcd_cmd_bits = 8,
		.lcd_param_bits = 8,
	};
	ESP_ERROR_CHECK(esp_lcd_new_panel_io_spi(LCD_HOST, &io_cfg, &io_handle));
	esp_lcd_panel_dev_config_t panel_cfg = {
		.reset_gpio_num = GPIO_RST,
		.rgb_ele_order = LCD_RGB_ELEMENT_ORDER_RGB,
		.data_endian = LCD_RGB_DATA_ENDIAN_LITTLE,
		.bits_per_pixel = 16,
	};
	ESP_ERROR_CHECK(esp_lcd_new_panel_st7789(io_handle, &panel_cfg, &panel_handle));
	ESP_ERROR_CHECK(esp_lcd_panel_reset(panel_handle));
	ESP_ERROR_CHECK(esp_lcd_panel_init(panel_handle));
	ESP_ERROR_CHECK(esp_lcd_panel_swap_xy(panel_handle, true));
	ESP_ERROR_CHECK(esp_lcd_panel_mirror(panel_handle, false, true));
	ESP_ERROR_CHECK(esp_lcd_panel_disp_on_off(panel_handle, true));
}
```

### Debug Logs.

```plain

```

### More Information.

I modified `esp_lcd_panel_st7789.c` and fixed the bug of `esp_lcd_panel_mirror()`:

```cpp
static esp_err_t panel_st7789_mirror(esp_lcd_panel_t *panel, bool mirror_x, bool mirror_y)
{
    st7789_panel_t *st7789 = __containerof(panel, st7789_panel_t, base);
    esp_lcd_panel_io_handle_t io = st7789->io;
    if (mirror_x) {
        st7789->madctl_val |= LCD_CMD_MX_BIT;
        st7789->madctl_val |= (1 << 2);
    } else {
        st7789->madctl_val &= ~LCD_CMD_MX_BIT;
        st7789->madctl_val &= ~(1 << 2);
    }
    if (mirror_y) {
        st7789->madctl_val |= LCD_CMD_MY_BIT;
        st7789->madctl_val |= (1 << 4);
    } else {
        st7789->madctl_val &= ~LCD_CMD_MY_BIT;
        st7789->madctl_val &= ~(1 << 4);
    }
    ESP_RETURN_ON_ERROR(esp_lcd_panel_io_tx_param(io, LCD_CMD_MADCTL, (uint8_t[]) {
        st7789->madctl_val
    }, 1), TAG, "io tx param failed");
    return ESP_OK;
}
```

However, I didn't find the way to change LCD refresh mode to reverse mode, so I created a issue here (～￣▽￣)～

By the way, I wonder if we can use Chinese to communicate, 我英语不好，表词达意不准确。

Maybe you need this: [ST7789V_SPEC_V1.0.pdf](https://github.com/user-attachments/files/18515131/ST7789V_SPEC_V1.0.pdf)

suda-morris, 2025-01-23:
Hi @yiming564 Thanks for sharing the background and the fix.

[image]

May I know if setging/clearing the ML/MH bits have resolved your "tearing" problem?

yiming564, 2025-01-23:
@suda-morris Thank you for the reply.

Yes. My code can resolve tearing problem in `esp_lcd_panel_mirror()`, but I didn't find a solution for `esp_lcd_panel_swap_xy()`.

Lzw655, 2025-02-05:
Hi @yiming564,

For SPI LCDs, the TE signal line is required to completely prevent tearing (refer to the [docs](https://docs.espressif.com/projects/esp-iot-solution/en/latest/display/lcd/lcd_screen_tearing.html#the-method-based-on-the-te-signal)). Are you using the TE signal?

yiming564, 2025-02-09:
> Hi [@yiming564](https://github.com/yiming564),
>
> For SPI LCDs, the TE signal line is required to completely prevent tearing (refer to the [...]

@Lzw655 I'm not using the TE signal.

Using TE signal or not actually has nothing to do with the tearing problem. In my case, the tearing problem was caused because ST7789 write the **real frame buffer (inside the screen module)** from Top to Bottom then from Left to Right (because I swapped xy), but the screen panel read the frame buffer from Left to Right then from Top to Bottom as usual. Such unsynchronization caused the triangle tearing.

P.S. I noticed your position is in Shanghai ... Maybe we can use Chinese to communicate?