espressif/esp-idf #15267: no Wifi Connection (maybe RAM Problem)

diplfranzhoepfinger, 2025-01-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

`
: 1244 bytes.
I (1247754) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1247754) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1247756) wifi:m f assoc rsp l=122

I (1248762) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1248762) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1248764) wifi:m f assoc rsp l=122

I (1249770) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1249771) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1249773) wifi:m f assoc rsp l=122

I (1250778) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1250778) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1250781) wifi:m f assoc rsp l=122

I (1254425) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1254425) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1254428) wifi:m f assoc rsp l=122

I (1255433) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1255433) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1255436) wifi:m f assoc rsp l=122

I (1255802) main: Core id: 0, prio: 1,  free heap: 4727936 bytes., minimum free heap: 4717560 bytes., minimum free stack: 1244 bytes.
I (1256441) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1256441) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1256444) wifi:m f assoc rsp l=122

I (1257449) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1257450) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
W (1257452) wifi:m f assoc rsp l=122

I (1258245) wifi:station: 00:93:37:e4:ca:37 leave, AID = 1, reason = 1, bss_flags is 34311266, bss:0x3c375e2c
I (1258245) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (1265804) main: Core id: 0, prio: 1,  free heap: 4729556 bytes., minimum free heap: 4717560 bytes., minimum free stack: 1244 bytes.
`

diplfranzhoepfinger, 2025-01-23:
after a reboot i got this: 

```
I (72188) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1, snd_ch_cfg:0x0
I (72188) wifi:station: 00:93:37:e4:ca:37 join, AID=1, bgn, 40U
I (72202) WiFi SoftAP: Station 00:93:37:e4:ca:37 joined, AID=1
I (73544) wifi:<ba-add>idx:2 (ifx:1, 00:93:37:e4:ca:37), tid:0, ssn:12, winSize:64
I (75680) main: Core id: 0, prio: 1,  free heap: 4727532 bytes., minimum free heap: 4717424 bytes., minimum free stack: 1420 bytes.
I (78449) esp_netif_lwip: DHCP server assigned IP to a client, IP is: 192.168.4.2
I (85682) main: Core id: 0, prio: 1,  free heap: 4727408 bytes., minimum free heap: 4717424 bytes., minimum free stack: 1420 bytes.
```

Alvin1Zhang, 2025-03-27:
Thanks for reporting.

hansw123, 2025-03-31:
@diplfranzhoepfinger  hi,could  you   provide your code and idf version, if can provide wifi wireshark capture will be better, thanks !

hansw123, 2025-03-31:
@diplfranzhoepfinger  and from the log ,we can see the free heap size is enough. the assoc resp allocated from the wifi mgmt long buffers（which max num is 32 ）,so the only reason the buffer is not enough, may be some other device is scaning, or some other
things take the buffer  successfuly. 
for the root cause we  need  more detail, could you provide your code and wifi packet capture?

hansw123, 2025-04-01:
@diplfranzhoepfinger 
you can refer this doc https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

nachiketkukade, 2025-04-11:
Duplicate of #15483

diplfranzhoepfinger, 2025-04-11:
> Duplicate of [#15483](https://github.com/espressif/esp-idf/issues/15483)

true !