espressif/esp-idf #15268: Bluetooth Root File Problem

elecrain, 2025-01-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have two computer at the office, let assume them as PC-A and PC-B. 

- PC-A is windows computer and its using idf v5.0.0.
- PC-B is mac computer and its also using idf v5.0.0. (the problem also arised in another windows computer)

We do not update idf versions to keep the project stable. The project using bluetooth to communicate with ios / android devices. 

Here is the problem: 

When the project is build on PC-A and downloaded into ESP32, all bluetooth functions work without a problem. But when the project transferred to PC-B, we only change c_cpp_prop json and settings json to give the correct paths. These path are taken from a local project on PC-B. However, even though that we can build the project successfully, when the code downloaded into "entirely empty flash" ESP32, bluetooth functions do not work, but all the other sections work. 

- Notice, when we first code the ESP32 from PC-A, then download the code again from PC-B without erase-flash, bluetooth works fine. 
- But when we download the code into ESP32 with flash-erase from PC-B, bluetooth does not work. 

I did not attach any log file since the code is exactly same on both cases. We thought some esp component files have mismatch but could not resolve this issue. 

What might be the possible solution ? 

Thanks !

BetterJincheng, 2025-01-24:
@elecrain please check if examples/bluetooth/bluedroid/classic_bt/a2dp_sink works with the PC-B.

elecrain, 2025-01-25:
I dont have the required hardware but the example project worked well, I've connected with my IOS Device and played song. When played song, code printed the data packages. 

Also notice, this problem is not specific to PC-B. When we move the project to any other PC, we have the same problem.

BetterJincheng, 2025-02-06:
@elecrain Please upload the sdkconfig files generated by PC-A and PC-B.

BTW, please try with `esptool.py erase_flash` on PC-B before download the code.

BetterJincheng, 2025-02-06:
@elecrain You can do a test like this: build the project on PC-A, then copy the resulting binary files to PC-B, and use PC-B to download these binary files.

```
esptool.py --chip esp32 -p /dev/ttyUSB6 -b 460800 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 40m --flash_size 2MB 0x1000 bootloader/bootloader.bin 0x10000 a2dp_sink.bin 0x8000 partition_table/partition-table.bin
```

The above are several bin files involved in the a2dp_stink project.

elecrain, 2025-02-10:
This is actually how we are distrubuting the project over many devices, PC-A creates binary files and we upload the binary to many devices through esp binary uploader tool. The problem arises when we build the project on PC-B and uploaded to entirely empty flash.

Notice, if the device is already programmed on PC-A and reprogrammed on PC-B without flash-erase, the problem does not occur.

BetterJincheng, 2025-02-11:
@elecrain Please use `esptool.py read_flash address size filename` to read out the two flash content.

possible value: `esptool.py read_flash 0, [your flash capacity] ./flash_content.bin`

We need to check if the flash contents are completely consistent in both cases.

BetterJincheng, 2025-02-11:
@elecrain please attach the whole log including building and downloading in both cases.

BetterJincheng, 2025-02-11:
@elecrain I build the a2dp_sink example on my ubuntu 22.04 and copy `build/bootloader/bootloader.bin`, `build/partition_table/partition-table.bin` and `/build/a2dp_sink.bin` to my laptop (ubuntu 24.04 wsl) and download with:
`python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/a2dp_sink.bin`

All things worked well...

BetterJincheng, 2025-02-27:
@elecrain Any updates?

elecrain, 2025-03-04:
We couldn't have a chance to try this on both machines, I will update it asap when we try to build on both machines.

elecrain, 2025-03-11:
We've solved the problem by just copying "components" file from PC-A to PC-B and now it works fine.

closed 2025-03-11 by elecrain