espressif/esp-idf #15271: Missing initializer for ext_clk_freq_hz

floitsch, 2025-01-23:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The `I2S_STD_CLK_DEFAULT_CONFIG` macro is not initializing the field `ext_clk_freq_hz` which is available on `SOC_I2S_HW_VERSION_2`. When compiling for platforms with that hw-version (like the S3), there are now warnigs:
```
warning: missing initializer for member 'i2s_std_clk_config_t::ext_clk_freq_hz' [-Wmissing-field-initializers]
```

L-KAYA, 2025-01-24:
@floitsch Thanks for reporting! May I ask your environment? Arduino or using idf.py directly?

floitsch, 2025-01-24:
idf.py directly.

closed 2025-02-13 by espressif-bot

wakwak-koba, 2025-02-14:
I applied the fix, but the error

> i2s_std.h:220:1: error: designator order for field 'i2s_std_clk_config_t::ext_clk_freq_hz' does not match declaration order in 'i2s_std_clk_config_t'

https://github.com/thelazt/esp-idf/commit/2bb9fb267fea88dd4e9a75d223fd50ad3d2e9020#diff-a1137c1c92942c76ba853b95de0e85e14fbc3898d5490e3c02baba63d02ff4e6R218-R219

Do you need to change the order ?

>   .ext_clk_freq_hz = 0, \
     .mclk_multiple = I2S_MCLK_MULTIPLE_256,