espressif/esp-idf #15276: SD Card UHS-I SDR104 support

PickaxeHit, 2025-01-24:
### Describe the solution you'd like.

I noticed that there is support for UHS-I's SDR/DDR50 in ESP-IDF, but there are still many UHS-I cards that natively have higher speeds (such as SDR104, which communicates at a frequency of 208MHz). After my testing, SDR104 (or actually SDR100) itself can work, but it is just unstable. The problem only occurs when reading:
```
E (1494) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6494) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_wait_for_idle timeout
```
But there is no error when writing.
I am not familiar with the SDMMC driver of IDF. I would like to understand why this happens? And whether support for SDR104 can be added in the future?

### Describe alternatives you've considered.

Code here:
```c
/* SD card and FAT filesystem example.
   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/

// This example uses SDMMC peripheral to communicate with SD card.

#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>
#include "esp_vfs_fat.h"

#include <sys/time.h>
#include "sdmmc_cmd.h"
#include "driver/sdmmc_host.h"
#include "sd_test_io.h"
#if SOC_SDMMC_IO_POWER_EXTERNAL
#include "sd_pwr_ctrl_by_on_chip_ldo.h"
#endif

#define EXAMPLE_MAX_CHAR_SIZE    64

static const char *TAG = "example";

#define MOUNT_POINT "/sdcard"
#define EXAMPLE_IS_UHS1    (CONFIG_EXAMPLE_SDMMC_SPEED_UHS_I_SDR50 || CONFIG_EXAMPLE_SDMMC_SPEED_UHS_I_DDR50)

#ifdef CONFIG_EXAMPLE_DEBUG_PIN_CONNECTIONS
const char* names[] = {"CLK", "CMD", "D0", "D1", "D2", "D3"};
const int pins[] = {CONFIG_EXAMPLE_PIN_CLK,
                    CONFIG_EXAMPLE_PIN_CMD,
                    CONFIG_EXAMPLE_PIN_D0
                    #ifdef CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
                    ,CONFIG_EXAMPLE_PIN_D1,
                    CONFIG_EXAMPLE_PIN_D2,
                    CONFIG_EXAMPLE_PIN_D3
                    #endif
                    };

const int pin_count = sizeof(pins)/sizeof(pins[0]);

#if CONFIG_EXAMPLE_ENABLE_ADC_FEATURE
const int adc_channels[] = {CONFIG_EXAMPLE_ADC_PIN_CLK,
                            CONFIG_EXAMPLE_ADC_PIN_CMD,
                            CONFIG_EXAMPLE_ADC_PIN_D0
                            #ifdef CONFIG_EXAMPLE_SDMMC_BUS_WIDTH_4
                            ,CONFIG_EXAMPLE_ADC_PIN_D1,
                            CONFIG_EXAMPLE_ADC_PIN_D2,
                            CONFIG_EXAMPLE_ADC_PIN_D3
                            #endif
                            };
#endif //CONFIG_EXAMPLE_ENABLE_ADC_FEATURE

[... 334 lines cut ...]
I (83) spi_flash: detected chip: winbond
I (84) spi_flash: flash io: qio
I (84) main_task: Started on CPU0
I (94) main_task: Calling app_main()
I (94) example: Initializing SD card
I (94) example: Using SDMMC peripheral
W (94) ldo: The voltage value 0 is out of the recommended range [500, 2700]
I (124) sdmmc_periph: host sampling edge is delayed by 2500 ps
Name: SD128
Type: SDHC
Speed: 200.00 MHz (limit: 208.00 MHz)
Size: 118900MB
CSD: ver=2, sector_size=512, capacity=243507200 read_bl_len=9
SSR: bus_width=4
[SD raw write                                           ] (10x)    6.220 ms    64.00 kB     10.048 MB/s
[SD raw read                                            ] (10x)    3.184 ms    64.00 kB     19.627 MB/s
[SD raw write                                           ] (10x)    7.520 ms   128.00 kB     16.621 MB/s
E (1494) sdmmc_req: sdmmc_host_wait_for_event returned 0x107
E (6494) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_wait_for_idle timeout
ESP_ERROR_CHECK failed: esp_err_t 0x107 (ESP_ERR_TIMEOUT) at 0x40009e86
```

Using Kingston CANVAS GO! Plus 128GB.

[image]

igrr, 2025-01-24:
> After my testing, SDR104 (or actually SDR100) itself can work, but it is just unstable. The problem only occurs when reading:
> I am not familiar with the SDMMC driver of IDF. I would like to understand why this happens? And whether support for SDR104 can be added in the future?

This is a limitation of ESP32-P4 hardware related to IO MUX and the clock tree, which we can't work around in software. The datasheet (which is to be released) will mention only SDR/DDR50 as supported modes.

I am not sure if this will be fixed in a future silicon revision of ESP32-P4, or only in some future ESP32-series chip after ESP32-P4.

PickaxeHit, 2025-01-25:
I roughly understand that the ESP32-P4 hardware IO cannot support frequencies up to 200MHz, which leads to read failures. Maybe I can try to use a logic analyzer to check what happened and compare the timing differences when successful/unsuccessful. 

Do I need to close this issue?

igrr, 2025-01-25:
> Maybe I can try to use a logic analyzer to check what happened and compare the timing differences when successful/unsuccessful.

It's related to the signal shape, I recommend using an oscilloscope instead if you want to look into this.

I don't think there is anything to be done in software here, unfortunately, until the new chip (revision) capable of 200 MHz signals on the IO MUX comes out. So yes, we can probably close this issue.

Alvin1Zhang, 2025-03-27:
Thanks for reporting, feel free to reopen.

closed 2025-03-27 by Alvin1Zhang

egonbeermat, 2025-11-27:
> Seems already supported?

You think this is now supported? I can't get a Sandisk Extreme Pro UHS-I to init over 50MHz.....