espressif/esp-idf #15279: About the NVS Partition Generator

KaeLL, 2025-01-24:
Why is the minimum partition size 12KiB? And given the following csv 

```csv
key,type,encoding,value
test,namespace,,
test,data,u32,0
```

With K pages, why are pages [1,K - 1] generated with status Full even though they're empty?

Demonstration below

After running `nvs_partition_gen.py generate nvs.csv nvs.bin 32768`...
<details>
<summary>nvs_tool.py -i -f text nvs.bin</summary>

```yaml
Page no. 0, Status: Full, Version: 2, CRC32: b9ba2d84, Page address: 0x0
 Entry state bitmap: fa ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 7e817bfa | test : 1
 001. Written, Namespace Index: 001, Type: uint32_t  , Span: 001, Chunk Index: 255, CRC32:   c5d00d | test : 0
 002. Empty
 ...
 125. Empty

Page no. 1, Status: Full, Version: 2, CRC32: 389f48a3, Page address: 0x1000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 2, Status: Full, Version: 2, CRC32: 6081e18b, Page address: 0x2000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 3, Status: Full, Version: 2, CRC32: e1a484ac, Page address: 0x3000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 4, Status: Full, Version: 2, CRC32: d0bcb3db, Page address: 0x4000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 5, Status: Full, Version: 2, CRC32: 5199d6fc, Page address: 0x5000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 6, Status: Full, Version: 2, CRC32:  9877fd4, Page address: 0x6000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page Empty, Page address: 0x7000
 Entry state bitmap: ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff 
 000. Empty
 ...
 125. Empty

Page no. 0 	CRC32: OK
Page no. 1 	CRC32: OK
Page no. 2 	CRC32: OK
Page no. 3 	CRC32: OK
Page no. 4 	CRC32: OK
Page no. 5 	CRC32: OK
Page no. 6 	CRC32: OK
Page Empty
```

</details>

Pages 1 to 6 statuses is Full.

chegewara, 2025-01-25:
https://github.com/espressif/esp-idf-nvs-partition-gen/issues/12

KaeLL, 2025-01-25:
@adokitkat care to chime in?

adokitkat, 2025-01-26:
Hello, thank you both for the report. I will be taking a look at it this week. It doesn't seem to affect the NVS when it is loaded by ESP-IDF from what I know, that's why it was in lower priority.

chegewara, 2025-01-26:
Hi,
How is it possible it does not affect if full page can't be written anymore with new data? It is no longer useful. 
In such case most of nvs partition is useless.

igrr, 2025-01-26:
Just a note, the main use case for NVS partition generator is to create partitions with factory-programmed data or settings. Such factory settings partitions typically should never be written to, since if the partition gets corrupted, factory data may be lost. Therefore we weren't carefully testing this scenario when the generated NVS partition is further written to. And as @adokitkat mentioned, this bug didn't affect the case when the partition is read-only.

chegewara, 2025-01-26:
I understand that point of view, but factory settings may also be just a set of default values which can be changed by user.
In that case do we need separate nvs partition and to make code more complicated to select between default and user values?

Thanks

adokitkat, 2025-01-26:
No, this is a bug in Python implementation of NVS partition generator and will be fixed.

igrr, 2025-01-26:
> In that case do we need separate nvs partition and to make code more complicated to select between default and user values?

Regardless of fixing this bug I think this (keeping values changed by the user in a separate partition) would be much more robust. If the partition gets corrupted, the device can still be factory reset and will maintain at least the factory defaults. If you start modifying the factory-programmed data directly, you risk not being able to recover the defaults should the partition get corrupted.

chegewara, 2025-01-26:
And now i understand why there was not response in my issue for 2 months.
It is not a bug/issue, the problem is i dont know what espressif team thought designing it and i am using it wrong.

adokitkat, 2025-01-26:
I modified `nvsgen` example, filled about 200 entires and this is the output from `nvs_tool.py` - ESP-IDF actually does use those pages marked as full (btw see their bitmaps are all `ff`, meaning they're empty), it just skips the first page which is marked full as well (it should be marked as active). 

<details>

<summary>NVS tool output</summary>
```
> ./nvs_tool.py ~/esp/esp-idf/examples/storage/nvsgen/nvs3.img
Page no. 0, Status: Full, Version: 2, CRC32: b9ba2d84, Page address: 0x0
 Entry state bitmap: aa a8 aa fa ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
 000. Written, Namespace Index: 000, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32:  750a909 | storage : 1
 001. Written, Namespace Index: 001, Type: uint8_t   , Span: 001, Chunk Index: 255, CRC32: 4351d44d | u8_key : 255
 002. Written, Namespace Index: 001, Type: int8_t    , Span: 001, Chunk Index: 255, CRC32: ac3521f8 | i8_key : -128
 003. Written, Namespace Index: 001, Type: uint16_t  , Span: 001, Chunk Index: 255, CRC32: 92787c98 | u16_key : 65535
 004. Erased , Namespace Index: 001, Type: uint32_t  , Span: 001, Chunk Index: 255, CRC32: 9f1661b6 | u32_key : 4294967295
 005. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 2e39873b | i32_key : -2147483648
 006. Written, Namespace Index: 001, Type: string    , Span: 008, Chunk Index: 255, CRC32: 7c05630a | str_key : Size=222, CRC32=8f66953b
      0x000  4c 6f 72 65 6d 20 69 70 73 75 6d 20 64 6f 6c 6f  72 20 73 69 74 20 61 6d 65 74 2c 20 63 6f 6e 73  Lorem ipsum dolo r sit amet, cons
      0x020  65 63 74 65 74 75 72 20 61 64 69 70 69 73 63 69  6e 67 20 65 6c 69 74 2e 0a 46 75 73 63 65 20 71  ectetur adipisci ng elit..Fusce q
      0x040  75 69 73 20 72 69 73 75 73 20 6a 75 73 74 6f 2e  0a 53 75 73 70 65 6e 64 69 73 73 65 20 65 67 65  uis risus justo. .Suspendisse ege
      0x060  73 74 61 73 20 69 6e 20 6e 69 73 69 20 73 69 74  20 61 6d 65 74 20 61 75 63 74 6f 72 2e 0a 50 65  stas in nisi sit  amet auctor..Pe
      0x080  6c 6c 65 6e 74 65 73 71 75 65 20 72 68 6f 6e 63  75 73 20 64 69 63 74 75 6d 20 73 6f 64 61 6c 65  llentesque rhonc us dictum sodale
      0x0a0  73 2e 0a 49 6e 20 6a 75 73 74 6f 20 65 72 61 74  2c 20 76 69 76 65 72 72 61 20 61 74 20 69 6e 74  s..In justo erat , viverra at int
      0x0c0  65 72 64 75 6d 20 65 67 65 74 2c 20 69 6e 74 65  72 64 75 6d 20 76 65 6c 20 64 75 69 2e 00 ff ff  erdum eget, inte rdum vel dui..ÿÿ
 014. Empty
 ...
 125. Empty

Page no. 6, Status: Active, Version: 2, CRC32:  9877fd4, Page address: 0x1000
 Entry state bitmap: aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa aa ff ff ff ff ff ff ff ff ff ff ff ff ff
 000. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: ab7a8780 | new_key_124 : 42
 001. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 367566f6 | new_key_125 : 42
 002. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 4a14432d | new_key_126 : 42
 003. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: d71ba25b | new_key_127 : 42
 004. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 818f16ad | new_key_128 : 42
 005. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 1c80f7db | new_key_129 : 42
 006. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32:  b2dd373 | new_key_130 : 42
 007. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 96223205 | new_key_131 : 42
 008. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: ea4317de | new_key_132 : 42
 009. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 774cf6a8 | new_key_133 : 42
 010. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 12815c68 | new_key_134 : 42
 011. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 8f8ebd1e | new_key_135 : 42
 012. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: f3ef98c5 | new_key_136 : 42
 013. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 6ee079b3 | new_key_137 : 42
 014. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 3874cd45 | new_key_138 : 42
 015. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: a57b2c33 | new_key_139 : 42
 016. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 902cdc69 | new_key_140 : 42
 017. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32:  d233d1f | new_key_141 : 42
 018. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 714218c4 | new_key_142 : 42
 019. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: ec4df9b2 | new_key_143 : 42
 020. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 89805372 | new_key_144 : 42
 021. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 148fb204 | new_key_145 : 42
 022. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 68ee97df | new_key_146 : 42
[... cut ...]
 121. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32:  dfd9da2 | new_key_119 : 42
 122. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: b2d6089b | new_key_120 : 42
 123. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 2fd9e9ed | new_key_121 : 42
 124. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: 53b8cc36 | new_key_122 : 42
 125. Written, Namespace Index: 001, Type: int32_t   , Span: 001, Chunk Index: 255, CRC32: ceb72d40 | new_key_123 : 42
```

</details>

KaeLL, 2025-01-27:
@adokitkat glad that got cleared up. But I didn't get the question about the minimum size. Was it rhetorical?

@igrr Interesting. My use case is both to provide defaults as well as factory data, all on the same partition to avoid waste, without much concern for reliability beyond the physical one of the flash itself, which the wear leveling is supposed to mitigate the risks of. This leads me to several questions: how effective is the wear leveling? And now that your brought it up, what else could cause NVS partitions to get corrupted? If possible, how to avoid it? And must partitions be erased when `nvs_flash_init` returns ESP_ERR_NVS_{NEW_VERSION_FOUND,NO_FREE_PAGES}? Why?

adokitkat, 2025-01-27:
Well it used to be in the docs, but I cannot find it there now. I suppose it still is 4kiB, meaning 4 pages. However recommended size is at least 6kiB. I was also wondering if there was a problem that didn't allow you to go under 12kiB?

adokitkat, 2025-01-27:
Sorry, I made a mistake about the size and it isn't 4kiB. It really is 12kiB meaning 0x3000 in hexadecimal, which means 3 pages.

rrtandler, 2025-01-27:
@KaeLL - the minimum NVS partition size answer is driven by the states of NVS pages on initialisation. In regular case, there is typically one page "active", several pages "full" and several pages "free". For correct initialisation, there has to be at least one free page available in the pool of pages. If the power is interrupted during the moment, when the NVS library writes the data and has to "compress" some of full pages, there may be one page temporarily marked as "freeing". This process might be interrupted i.e. by power-loss event. Then, during subsequent initialisation, there would be 2 pages present in states 'active' and 'freeing' and as there is a requirement to have at least one 'free' page, the minimal number of NVS pages is 3.

taliesin, 2025-02-03:
I just opened a feature request to allow for 4kB readonly NVS partitions for factory data, see https://github.com/espressif/esp-idf/issues/15317

NVS is a low effort method passing f.i. factory calibration data, because it allows for most data types (float and double are missing, one can cast an u32 into float and u64 into a double if needed). Alternatively you could write a JSON string to the partition and parse it using cJSON at the expense of a higher memory footprint.

closed 2025-02-05 by espressif-bot

adokitkat, 2025-02-05:
Please try the new version, should be fixed now.

KaeLL, 2025-02-05:
@adokitkat While we're at it, here's a question not worth filing another issue for: why aren't the nvs tools exposed through export.sh?