espressif/esp-idf #15280: ESP32-P4 v1.0 Silicon idf.py flash issue.

csalinasonline, 2025-01-24:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have new ESP32-P4 v1.0 Silicon.
Before I had v0.1 Silicon. working on a prototype that mirrors most of the hardware on ESP32-P4-eval board.

But since changing to ESP32-P4 v1.0 Silicon i have issues programming this new chip.

I get this error "A fatal error occurred: bootloader/bootloader.bin requires chip revision in range [v0.1 - v0.99] (this chip is revision v1.0). Use --force to flash anyway."

Please me my attached log file and sdkconfig below.

[idf_py_flash_issue_012425.txt](https://github.com/user-attachments/files/18542913/idf_py_flash_issue_012425.txt)
[sdkconfig.txt](https://github.com/user-attachments/files/18542917/sdkconfig.txt)

chegewara, 2025-01-25:
Hi,
you may want to switch to master branch (at least for now), there is version 1.0+ supported.

[image]

csalinasonline, 2025-01-25:
Thanks for update. I will try master branch and let u know status soon!

csalinasonline, 2025-01-25:
I have pulled latest https://github.com/espressif/esp-idf.git and checked out the master branch.

I set up env in root esp-idf directory.

- ./install.sh
- . ./export.sh

Then navigated and opened 'examples/get-started/blink' project and build flash and monitor.

- idf.py set-target esp32p4
- idf.py build
- -idf.py flash monitor

But now I get this issue.

[image]

Please see log files here.
[idf_py_flash_issue_012525.txt](https://github.com/user-attachments/files/18547938/idf_py_flash_issue_012525.txt)

Might my bootselect pins be an issue?
I have 
  - GPIO_35 pulled to 3.3V High
  - GPIO_36 pulled to 3.3V High

suda-morris, 2025-01-26:
HI @csalinasonline Have you configured the console output to choose the "USB Serial JTAG"?

[image]

csalinasonline, 2025-01-26:
@suda-morris your Awesome!
It works.
Now I have ESP32-p4 v1.0 working on esp-idf-master!

Might there be a release to v5.5?
Somehow IC in debug logs master is v5.5 already?

[image]

suda-morris, 2025-02-05:
@csalinasonline There will be a release v5.5 in this year: https://github.com/espressif/esp-idf/blob/master/ROADMAP.md#esp-idf-minor-releases

I think esp32-p4 chip revision v1.0 is already supported on release/v5.3/5.4 and master branch.

closed 2025-02-05 by suda-morris