espressif/esp-idf #15282: Nimble Static Random Address is not persisted after reboot

vandy, 2025-01-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm using the Nimble stack and several functions in `nimble_callback_on_sync` but get an unexpected result for me.

I want to get the same Static Random Address once I've set it up, but after each restart it's updated to a different one. My code is something like this:
```c
// on_sync
int rc = ble_hs_id_copy_addr(BLE_ADDR_RANDOM, address, NULL);
if (rc) {
    ESP_LOGI(TAG, "Random address is not set, rc: %d", rc); // rc == 21 (BLE_HS_ENOADDR)
    esp_ble_addr_t addr;
    rc = esp_ble_hw_get_static_addr(&addr);
    if (rc) {
        ESP_LOGI(TAG, "Error in esp_ble_hw_get_static_addr, rc: %d", rc); // always -1
    }
}
rc = ble_hs_util_ensure_addr(1);
```
What I expect is that on the first start there is no Static Random Address set. I call `ble_hs_util_ensure_addr(1)` which generates Static Random Address and set it via `ble_hs_id_set_rnd` (which sends an HCI command to the Controller (`ble_hs_hci_cmd_tx(BLE_HCI_OP(BLE_HCI_OGF_LE, BLE_HCI_OCF_LE_SET_RAND_ADDR), &cmd, sizeof(cmd), NULL, 0)`). From that point on I think that Controller should store the set address and give it back on the next restart, when `ble_hs_util_ensure_addr` loads it with `ble_hs_util_load_rand_addr -> esp_ble_hw_get_static_addr`. But it's not the case and a new address is generated each time.

Is it expected behavior? Why?

Shreeyash-17, 2025-01-29:
@vandy  As per the Bluetooth Core Specification v5.3, Vol 6, Part B, Section 1 (General Description), Subsection 1.3.2.1 (Static Device Address):
A device may choose to initialize its static address to a new value after each power cycle. This is expected behavior because the specification mandates it. Accordingly, nimble stack has been implemented to align with this requirement.
 
[image]

It is also noted in the specification that if the static address of a device is changed, the address stored in peer devices will no longer be valid, and the ability to reconnect using the old address will be lost. The NimBLE stack has been implemented with this behavior.
 
1. Additionally, I see you have pointed APIs that are used to set a "Random" address. But your question is about RPA Public ?  If so, then you can refer to bleprph/main/main.c example file and in bleprph_advertise function before passing own_address_type to ble_gap_adv_start set own_address_type to RPA_PUBLIC_DEFAULT(type 2).  
 
[image]

Which chip are you using for this? 
 
2. If you want to create a random address and set it to a specific address of your choice, you can make the below-highlighted changes in bleprph/main/main.c. This will ensure that the address remains consistent and does not change after each reboot.
 

[image]

vandy, 2025-01-29:
@Shreeyash-17, thank you for your answer. It seems my understanding of the Bluetooth spec and nimble implementation is not correct.

I meant that there are different types of addresses per spec.

1. Public Address (which is already quite random, because it's generated for each chip by manufacturer and burned into the chip at production) - it stays the same during each boot (but it's random). Could be used as Identity Address for RPA (also could be used without RPA, just `BLE_OWN_ADDR_PUBLIC` in `ble_gap_adv_start`).
2. Static Random Address - could be generated by developer (should follow rules mentioned above 11xx...). Could be used as Identity Address for RPA (also could be used without RPA, just `BLE_OWN_ADDR_RANDOM` in `ble_gap_adv_start`).

Resolvable Private Random Address (RPA) should use one of these as Identity Address (public or static).

So what I meant is that I want to create my own random address (in this case I have only one option - to use Static Random Address as a storage), use `ble_hs_id_set_rnd` (which sends it to the Controller via HCI command) and use this generated address in an advertising (can use just as `BLE_OWN_ADDR_RANDOM` or as RPA `BLE_OWN_ADDR_RPA_RANDOM_DEFAULT`). The problem arises when I reboot the chip, because the Static Random Address is gone. And if I used RPA than bonded information is not valid any more (because it uses the Identity Address).

I thought that after I set a Static Random Address into the Controller it's persisted until I set a new one. In this case a developer could change the SRA in each reboot (as per spec) if he wishes, but also could use the same SRA once generated. But seems it doesn't work this way?

Should I store generated SRA somewhere (flash or define, etc.) to be able to use the same one on each reboot? Is it a weird use case I described?

Shreeyash-17, 2025-01-30:
@vandy No, this will not work because NimBLE follows what is mentioned in the specification, i.e. generate a new address each time after a reboot.
However, I feel, you are free to store the address manually and implement a scheme of yours to use that address, for your use case. 

Also, Address resolution will ensure the address is always resolved, so RPA works correctly with bonding

vandy, 2025-05-30:
@Shreeyash-17 
Re-reading the thread I've realized there is yet one unanswered question.

For sure Nimble stack starts with an empty Static Random Address (SRA):
```c
// ble_hs_id.c
...
static uint8_t ble_hs_id_pub[6];
static uint8_t ble_hs_id_rnd[6]; // <- this is storage for SRA
static const uint8_t ble_hs_misc_null_addr[6];
```
Only `ble_hs_id_pub` is set during the startup.

But there is a function in `addr.c` which ensures that SRA is set. It could be called on start using `int ble_hs_util_ensure_addr(int prefer_random == 1)`.
```c
static int
ble_hs_util_ensure_rand_addr(void)
{
    ble_addr_t addr;
    int rc;

    /* If we already have a random address, then we are done. */
    rc = ble_hs_id_copy_addr(BLE_ADDR_RANDOM, NULL, NULL);
    if (rc == 0) {
        return 0;
    }

    /* Otherwise, try to load a random address. */
    rc = ble_hs_util_load_rand_addr(&addr);

    if (rc != 0) {
        /* If it didn't work, generate random address */
        rc = ble_hs_id_gen_rnd(0, &addr);
        if (rc != 0) {
            return rc;
        }
    }

    /* Configure nimble to use the random address. */
    rc = ble_hs_id_set_rnd(addr.val);
    if (rc != 0) {
        return rc;
    }

    return 0;
}
```
If `ble_hs_id_rnd` is empty (which is true on startup) we go to `ble_hs_util_load_rand_addr`:
```c
static int
ble_hs_util_load_rand_addr(ble_addr_t *addr)
{
    int rc = BLE_HS_ENOADDR;

#if MYNEWT_VAL(BLE_HCI_VS)
    struct ble_hci_vs_rd_static_addr_rp rsp;

    rc = ble_hs_hci_cmd_tx(BLE_HCI_OP(BLE_HCI_OGF_VENDOR,
                                      BLE_HCI_OCF_VS_RD_STATIC_ADDR),
                           NULL, 0, &rsp, sizeof(rsp));
    if (rc == 0) {
        addr->type = BLE_ADDR_RANDOM;
        memcpy(addr->val, rsp.addr, sizeof(addr->val));
        return 0;
    }
#endif

#if SOC_ESP_NIMBLE_CONTROLLER && CONFIG_BT_CONTROLLER_ENABLED
    rc = esp_ble_hw_get_static_addr((esp_ble_addr_t *)addr);
    if (rc == 0) {
        return 0;
    }
#endif
    return rc;
}
```
This function uses vendor specific HCI command to get SRA or `esp_ble_hw_get_static_addr` function. If we've never set SRA it should return nothing. So for the first time Nimble will generate SRA using `ble_hs_id_gen_rnd` and set SRA with `ble_hs_id_set_rnd`, which does:
```c
// excerpt
// ...
struct ble_hci_le_set_rand_addr_cp cmd;
memcpy(cmd.addr, addr, BLE_DEV_ADDR_LEN);

return ble_hs_hci_cmd_tx(BLE_HCI_OP(BLE_HCI_OGF_LE,
                                        BLE_HCI_OCF_LE_SET_RAND_ADDR),
                             &cmd, sizeof(cmd), NULL, 0);
// ...
memcpy(ble_hs_id_rnd, rnd_addr, 6);
```

Therefore after we generated SRA for the first time we send it with HCI command to the Controller and set variable in memory with its value. Of course this variable will be empty after reboot. But what I thought is the next time (after reboot) we try to `ble_hs_util_ensure_rand_addr` it gets SRA (previously set before reboot) from the Controller with `ble_hs_util_load_rand_addr` (because this function get SRA from the Controller via HCI command). But it doesn't. `ble_hs_util_load_rand_addr` always returns nothing after reboot. Seems like the Controller doesn't remember SRA set by HCI command after reboot. And the only way to persist SRA (in case one uses RPA with SRA as Identity) is to persist it manually (storing it in non-volatile memory or hardcoding its value).

If some future reader knows the answer, please share. Should a conforming Controller persist SRA which was set with HCI command between reboots or not? Per Bluetooth Core Specification v5.3, Vol 6, Part B, Section 1 (General Description), Subsection 1.3.2.1 (Static Device Address) it _may_ choose to initialize with a new value, so it seems for me not a Nimble issue. Nimble is trying to get SRA from the Controller, but the Controller returns nothing.