espressif/esp-idf #15283: Wifi won't connect - WIFI_REASON_AUTH_EXPIRE

Moddingear, 2025-01-25:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am using an esp32s3 (https://www.waveshare.com/esp32-s3-zero.htm) with espidf in platformio (no arduino), and anytime I want to connect to WiFi, I can't get connected and have WIFI_REASON_AUTH_EXPIRE. I tried the examples from https://github.com/espressif/esp-idf/tree/master/examples/wifi, notably getting_started/station, fast_scan and WPS. None work, with my home wifi and with my phone as a hotspot.

(code 205 corresponds to WIFI_REASON_CONNECTION_FAIL)

Below is the log of the attempts of connection from a fast scan.
```
I (393) wifi:wifi driver task: 3fcb40d0, prio:23, stack:6656, core=0
I (403) wifi:wifi firmware version: 0caa81945
I (403) wifi:wifi certification version: v7.0
I (403) wifi:config NVS flash: enabled
I (403) wifi:config nano formating: disabled
I (403) wifi:Init data frame dynamic rx buffer num: 32
I (413) wifi:Init static rx mgmt buffer num: 5
I (413) wifi:Init management short buffer num: 32
I (413) wifi:Init dynamic tx buffer num: 32
I (423) wifi:Init static tx FG buffer num: 2
I (423) wifi:Init static rx buffer size: 1600
I (433) wifi:Init static rx buffer num: 10
I (433) wifi:Init dynamic rx buffer num: 32
I (443) wifi_init: rx ba win: 6
I (443) wifi_init: accept mbox: 6
I (443) wifi_init: tcpip mbox: 32
I (443) wifi_init: udp mbox: 6
I (453) wifi_init: tcp mbox: 6
I (453) wifi_init: tcp tx win: 5760
I (453) wifi_init: tcp rx win: 5760
I (463) wifi_init: tcp mss: 1440
I (463) wifi_init: WiFi IRAM OP enabled
I (463) wifi_init: WiFi RX IRAM OP enabled
I (503) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
I (533) wifi:mode : sta (34:b7:da:53:1c:a8)
I (533) wifi:enable tsf
I (533) wifi: WIFI_EVENT_STA_START
I (543) wifi: wifi_init_sta finished.
I (2023) wifi:new:<9,0>, old:<1,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (2023) wifi:state: init -> auth (0xb0)
I (3023) wifi:state: auth -> init (0x200)
I (3023) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (3033) wifi: WIFI_EVENT_STA_DISCONNECTED
I (3033) wifi: connect to the AP fail, reason 2, rssi -50
I (5443) wifi: WIFI_EVENT_STA_DISCONNECTED
I (5443) wifi: connect to the AP fail, reason 205, rssi -128
I (5533) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (5533) wifi:state: init -> auth (0xb0)
I (6533) wifi:state: auth -> init (0x200)
I (6533) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (6533) wifi: WIFI_EVENT_STA_DISCONNECTED
I (6533) wifi: connect to the AP fail, reason 2, rssi -48
I (8943) wifi: WIFI_EVENT_STA_DISCONNECTED
I (8943) wifi: connect to the AP fail, reason 205, rssi -128
I (9013) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (9013) wifi:state: init -> auth (0xb0)
I (10013) wifi:state: auth -> init (0x200)
I (10013) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (10013) wifi: WIFI_EVENT_STA_DISCONNECTED
I (10023) wifi: connect to the AP fail, reason 2, rssi -50
I (12423) wifi: WIFI_EVENT_STA_DISCONNECTED
I (12423) wifi: connect to the AP fail, reason 205, rssi -128
I (12493) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (12493) wifi:state: init -> auth (0xb0)
I (13493) wifi:state: auth -> init (0x200)
I (13493) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (13493) wifi: WIFI_EVENT_STA_DISCONNECTED
I (13503) wifi: connect to the AP fail, reason 2, rssi -50
I (15903) wifi: WIFI_EVENT_STA_DISCONNECTED
I (15903) wifi: connect to the AP fail, reason 205, rssi -128
I (15973) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (15973) wifi:state: init -> auth (0xb0)
I (16973) wifi:state: auth -> init (0x200)
I (16973) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (16973) wifi: WIFI_EVENT_STA_DISCONNECTED
I (16983) wifi: connect to the AP fail, reason 2, rssi -48
I (19393) wifi: WIFI_EVENT_STA_DISCONNECTED
I (19393) wifi: connect to the AP fail, reason 205, rssi -128
I (19453) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (19453) wifi:state: init -> auth (0xb0)
I (20453) wifi:state: auth -> init (0x200)
I (20463) wifi:new:<9,0>, old:<9,0>, ap:<255,255>, sta:<9,0>, prof:1, snd_ch_cfg:0x0
I (20463) wifi: WIFI_EVENT_STA_DISCONNECTED
I (20463) wifi: Failed to connect!
I (20463) wifi: connect to the AP fail, reason 2, rssi -48
I (20473) wifi: Failed to connect```

mhdong, 2025-02-19:
hi @Moddingear 
Hello, may I ask what encryption mode your router is using? Could you configure the log level through menuconfig.
idf.py menconfig:
1. Component config -> Log -> Log Level -> Default log verbosity -> Debug
2. Component config->Wi-Fi->Print debug message from WPA Supplicant
If possible, could you provide a packet capture? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Alvin1Zhang, 2025-03-27:
Thanks for reporting, will close due to inactivity, feel free to reopen if any updates.

closed 2025-03-27 by Alvin1Zhang

mhdong, 2025-09-08:
hi @xiangxin1025 
Hello, may I ask what encryption mode your router is using? Could you configure the log level through menuconfig.
idf.py menconfig:

    Component config -> Log -> Log Level -> Default log verbosity -> Debug
    Component config->Wi-Fi->Print debug message from WPA Supplicant
    If possible, could you provide a packet capture? https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/wireshark-user-guide.html

Alvin1Zhang, 2025-11-21:
Thanks for reporting, feel free to reopen if the issue still happens.

sQueezy1332, 2026-04-24:
same issue, i hide macs
`I (515) wifi:mode : sta ()
I (515) wifi:enable tsf
D (515) wifi:filter: set rx policy=1
D (515) wifi:connect status 0 -> 0
W (515) WIFI_EVENT: event_id 43
D (515) STA: START
D (515) wifi:Start wifi connect
D (515) wifi:connect status 0 -> 0
D (515) wifi:connect chan=0
D (515) wifi:first chan=1
D (515) wifi:connect status 0 -> 1
D (515) wifi:filter: set rx policy=3
D (515) wifi:clear scan ap list
D (515) wifi:start scan: type=0x50f, priority=2, cb=0x42056cf4, arg=0x0, ss_state=0x1, time=519054, index=0   
--- 0x42056cf4: cnx_start_handoff_cb at ??:?
D (515) wifi:perform scan: ss_state=0x9, chan<1,0>, dur<0,120>
D (515) wifi_init_default: wifi_start esp-netif:0x3fcaaff8 event-id2
D (515) wifi_init_default: WIFI mac address: 00 00 00 00 00 00
D (515) esp_netif_handlers: esp_netif action has started with netif0x3fcaaff8 from event_id=2
I (515) MAIN: WiFi started, waiting for connection
D (635) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (635) wifi:perform scan: ss_state=0x9, chan<2,0>, dur<0,120>
D (755) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (755) wifi:perform scan: ss_state=0x9, chan<3,0>, dur<0,120>
D (875) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (875) wifi:perform scan: ss_state=0x9, chan<4,0>, dur<0,120>
D (995) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1005) wifi:perform scan: ss_state=0x9, chan<5,0>, dur<0,120>
D (1125) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1125) wifi:perform scan: ss_state=0x9, chan<6,0>, dur<0,120>
D (1245) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1245) wifi:perform scan: ss_state=0x9, chan<7,0>, dur<0,120>
D (1365) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1365) wifi:perform scan: ss_state=0x9, chan<8,0>, dur<0,120>
D (1485) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1485) wifi:perform scan: ss_state=0x9, chan<9,0>, dur<0,120>
D (1605) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1605) wifi:perform scan: ss_state=0x9, chan<10,0>, dur<0,120>
D (1735) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (1735) wifi:perform scan: ss_state=0x9, chan<11,0>, dur<0,120>
D (1735) wifi:rsn valid: gcipher=1 ucipher=3 akm=5 mac=00:00:00:00:00:00
D (1735) wifi:profile match: ss_state=0x7
D (1735) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x7
D (1745) wifi:find first mathched ssid, scan done
D (1745) wifi:filter: set rx policy=4
D (1745) wifi:first chan=1
D (1745) wifi:handoff_cb: status=0
D (1745) wifi:ap found, mac=00:00:00:00:00:00
D (1855) wifi:going for connection with bssid=00:00:00:00:00:00
D (1855) wifi:new_bss=0x3fc98810, cur_bss=0x0, new_chan=<11,0>, cur_chan=1
D (1855) wifi:filter: set rx policy=5
I (1855) wifi:new:<11,0>, old:<1,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
D (1855) wifi:connect_op: status=0, auth=5, cipher=3
D (1855) wifi:auth mode is not none
D (1855) wifi:connect_bss: auth=1, reconnect=0
I (1855) wifi:state: init -> auth (0xb0)
D (1855) wifi:start 1000ms AUTH timer
D (1855) wifi:clear scan ap list
W (1855) WIFI_EVENT: event_id 43
D (2855) wifi:auth timeout
I (2855) wifi:state: auth -> init (0x200)
D (2855) wifi:stop beacon and connect timers
D (2855) wifi:connect status 1 -> 4
D (2855) wifi:reason: auth expire(2)
D (2855) wifi:add bssid 00:00:00:00:00:00 to blacklist, cnt=0
D (2855) wifi:stop CSA timer
D (2855) wifi:remove 00:00:00:00:00:00 from rc list
D (2855) wifi:connect status 4 -> 0
I (2855) wifi:new:<11,0>, old:<11,0>, ap:<255,255>, sta:<11,0>, prof:1, snd_ch_cfg:0x0
[... 43 lines cut ...]
D (4555) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (4565) wifi:perform scan: ss_state=0x9, chan<13,0>, dur<360,360>
D (4925) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (4925) wifi:perform scan: ss_state=0x9, chan<14,0>, dur<360,360>
D (5285) wifi:scan end: arg=0x3fc99bec, status=0, ss_state=0x3
D (5285) wifi:filter: set rx policy=4
D (5285) wifi:first chan=1
D (5285) wifi:handoff_cb: status=0
D (5285) wifi:clear blacklist
D (5285) wifi:clear rc list
D (5285) wifi:clear blacklist
D (5285) wifi:connect status 1 -> 3
D (5285) wifi:Send disconnect event, reason=205
D (5285) wifi:disable connect timer
D (5285) wifi:clear scan ap list
W (5285) STA: DISCONNECTED reason 205 retry 1
D (5285) wifi:Start wifi connect
D (5285) wifi:connect status 3 -> 0
D (5285) wifi:connect chan=0
D (5285) wifi:first chan=11
D (5285) wifi:connect status 0 -> 1
D (5285) wifi:filter: set rx policy=3
D (5285) wifi:clear scan ap list
D (5285) wifi:start scan: type=0x50f, priority=2, cb=0x42056cf4, arg=0x0, ss_state=0x1, time=5293107, index=0 
--- 0x42056cf4: cnx_start_handoff_cb at ??:?`