espressif/esp-idf #15285: bool std::filesystem::exists(const std::filesystem::path &p) throws Invalid Argument Exception

Zhu-jiatong, 2025-01-26:
### Answers checklist.

- [x] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [x] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [x] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When calling the function `std::filesystem::exists("/mountpoint/path/to/file")`, it throws an exception: `filesystem error: status: Invalid argument [/mountpoint/path/to/file]`

However, calling `std::filesystem::exists("/mountpoint")` would return `true` as expected.

igrr, 2025-01-26:
Hi @Zhu-jiatong,

- which IDF version are you using? (`git describe` can be used to check)
- which filesystem is mounted at /mountpoint?
- does the `/path/to/file` exist?

Zhu-jiatong, 2025-01-26:
I am using 5.4.0 version.
SDSPI (via FatFS)
The existence of the "/path/to/file" should be irrelevant in this case, as the function should not throw in either case.

igrr, 2025-01-26:
Could you please try to narrow down the issue to a small project you could share?

I wasn't able to reproduce the issue with IDF v5.4, FATFS and the following snippet. We also have a few related [test cases](https://github.com/espressif/esp-idf/tree/master/tools/test_apps/storage/std_filesystem) which are passing, so I guess there is something specific about your case which isn't covered by our tests.

<details>
<summary>main.cpp</summary>

```c++
#include <filesystem>
#include "esp_err.h"
#include "esp_vfs_fat.h"
#include "wear_levelling.h"

extern "C" void app_main(void)
{
    esp_vfs_fat_mount_config_t mount_config = {};
    mount_config.max_files = 1;
    mount_config.format_if_mount_failed = true;
    wl_handle_t wl_handle;
    ESP_ERROR_CHECK(esp_vfs_fat_spiflash_mount_rw_wl("/mountpoint", "storage", &mount_config, &wl_handle));

    try {
        bool exists = std::filesystem::exists("/mountpoint/path/to/file");
        if (exists) {
            printf("File exists\n");
        } else {
            printf("File does not exist\n");
        }
    } catch (std::exception& e) {
        printf("Error: %s\n", e.what());
    }
}
```
</details>

<details>
<summary>sdkconfig.defaults</summary>

```
CONFIG_CXX_EXCEPTIONS=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_FATFS_LFN_HEAP=y
```
</details>

<details>
<summary>partitions.csv</summary>

```
factory,  app,  factory, 0x10000, 1M,
storage,  data, fat,     ,        528k,
```
</details>

<details>
<summary>main/CMakeLists.txt</summary>

```cmake
idf_component_register(SRCS "main.cpp"
                    INCLUDE_DIRS "."
                    PRIV_REQUIRES vfs fatfs)
```
</details>

<details>
<summary>CMakeLists.txt</summary>

```cmake
cmake_minimum_required(VERSION 3.16)

set(COMPONENTS main)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)
project(idfgh_14521)
```
</details>

<details>
<summary>Output:</summary>

```
W (1917) vfs_fat_spiflash: f_mount failed (13)
I (1917) vfs_fat_spiflash: Formatting FATFS partition, allocation unit size=4096
I (2067) vfs_fat_spiflash: Mounting again
File does not exist
```
</summary>

Thank you.

Zhu-jiatong, 2025-01-27:
Apologies, it is mainly due to an issue on my end. I noticed that for FS functionalities to function reliably at all times, the long filename option must be enabled to deal with file names that does not fit the 8.3 convention. I did not enable this option initially. Thanks.

main.cpp

sdkconfig.defaults

partitions.csv

main/CMakeLists.txt

CMakeLists.txt

Output:

Thank you.

igrr, 2025-01-27:
I see, yes, that would lead to an "Invalid argument" exception since the path component isn't a valid 8.3 name.

closed 2025-01-27 by igrr